google-deepmind/mujoco #2073: After loading the soft body (using flexcomp and with .msh file), the simulator crashed

sIHURs, 2024-09-21:
hi,

I have a model from a BioTac, and I've converted the mesh into a .msh file. However, when I try to open the simulator using ./simulate and drag the XML file into the simulator, it shows a long loading time, and eventually, there's no response.

Here is the XML code:
```xml
<mujoco model="soft">
  <option timestep="0.01" integrator="implicitfast" solver="CG"/>

  <worldbody>
    <light diffuse=".5 .5 .5" pos="0 0 5" directional="true" dir="0 0 -1" />
	  <geom name="floor" type="plane" size="10 10 10" pos="0 0 0" rgba="1 1 1 1" />
    <flexcomp name="softbody" type="gmsh" dim="3" rgba=".8 .2 .2 1" radius="0.001" pos="0 0 .3" mass="10" scale="1 1 1" file="int_ext_skin_combined_ftw.msh">
        <edge equality="true"/>
    </flexcomp>
  </worldbody>
</mujoco>
```

The .msh file contains 4,246 vertices and 15,410 faces. Could the issue be caused by the data size being too large? Has anyone encountered the same problem? How much memory would be required for this? I was able to load it successfully on Colab, but calculating certain scenarios, such as falling, takes an extremely long time.

.msh file:
[int_ext_skin_combined_ftw.zip](https://github.com/user-attachments/files/17084718/int_ext_skin_combined_ftw.zip)

yifan

yuvaltassa, 2024-09-21:
Please don't use the old MSH file format. We will remove it soon.

Use [OBJ](https://mujoco.readthedocs.io/en/latest/python.html#msh2obj-py), let us know if this also fails and how, if yes.

sIHURs, 2024-09-21:
> Please don't use the old MSH file format. We will remove it soon.
> 
> Use [OBJ](https://mujoco.readthedocs.io/en/latest/python.html#msh2obj-py), let us know if this also fails and how, if yes.

Thank you for your reply. I converted the file loaded in the `flexcomp` to .obj and set the `type` to mesh, with `dim` set to 2 (it shows that dim can only be 2 when using mesh).

Here is the code:
```xml
<mujoco model="soft">
  <option timestep="0.01" integrator="implicitfast" solver="CG"/>

  <worldbody>
    <light diffuse=".5 .5 .5" pos="0 0 5" directional="true" dir="0 0 -1" />
	  <geom name="floor" type="plane" size="10 10 10" pos="0 0 0" rgba="1 1 1 1" />
    <flexcomp name="softbody" type="mesh" dim="2" rgba=".8 .2 .2 1" radius="0.001" pos="0 0 .3" mass="10" scale="1 1 1" file="int_ext_skin_combined_ftw.obj">
        <edge equality="true"/>
    </flexcomp>
  </worldbody>
</mujoco>
```

However, it still crashes at the end. 
[image]
[image]

Could you please suggest any more effective methods to simulate a soft body, especially where some of its faces need to be fixed to a rigid body?

here is the mesh file: 
[skin.zip](https://github.com/user-attachments/files/17085433/skin.zip)

sIHURs, 2024-09-21:
error from the terminal:
```bash
yifan@yifan-Ubuntu ~/.m/m/bin> ./simulate                                           (base) 
MuJoCo version 3.1.6
Plugins registered by library 'libsensor.so':
    mujoco.sensor.touch_grid
Plugins registered by library 'libsdf.so':
    mujoco.sdf.bolt
    mujoco.sdf.bowl
    mujoco.sdf.gear
    mujoco.sdf.nut
    mujoco.sdf.torus
    mujoco.sdf.sdflib
Plugins registered by library 'libelasticity.so':
    mujoco.elasticity.cable
    mujoco.elasticity.membrane
    mujoco.elasticity.shell
    mujoco.elasticity.solid
Plugins registered by library 'libactuator.so':
    mujoco.pid
ERROR: mj_stackAllocInt: requested size is too large.
```

yuvaltassa, 2024-09-22:
Did you try increasing the memory size? 

https://mujoco.readthedocs.io/en/latest/XMLreference.html#size-memory

sIHURs, 2024-09-22:
> Did you try increasing the memory size?
> 
> https://mujoco.readthedocs.io/en/latest/XMLreference.html#size-memory

I'm not sure how much memory size I have to set. When I tried with setring it to 8G, the program can display, but I can't perform any operations, and it still crashes. How much memory is needed for this size of file? My computer only has 16G of memory.

Alternatively, is there another method? This project was previously running on Isaac Gym.

[image]
after this, the programm still crashes.

Is this due to hardware limitations, or is there another reason? What solutions might be available?

quagla, 2024-09-23:
Do you still get the issue if you set

```<contact selfcollide="none" internal="false"/>```

I suspect that the solver might be trying to allocate a huge number of self- and internal collisions.

sIHURs, 2024-09-23:
It's this issue that has been resolved; I can open it now. Thank you very much.

I have a new questio, actually is that I have added the rigid body to the simulation (as described in https://github.com/google-deepmind/mujoco/issues/2077#issuecomment-2368062610), but I have not yet pinned the area where the soft body contacts the rigid body. However, the soft body (the red part, which looks like a balloon) is expanding. Why is this happening? 
[image]
[image]

Also, when it is first loaded, the soft body does not appear to be the correct size, as I have not applied any scaling. The correct situation is shown in the picture.
[image]

xml:
```xml
<mujoco model="biotac-sim">
  <!-- <compiler autolimits="true"/> -->
  <option solver="CG"/>
  <size memory="8G"/>
  
  <extension>
    <!-- add extension -->
  </extension>

  <asset>
    <mesh name="top_side_core_bc" file="top_side_core_bc.obj"/>
    <!-- <mesh name="skin" file="int_ext_skin_combined_ftw.msh"/> -->
  </asset>

  <visual>
      <!-- <headlight ambient=".7 .7 .7" diffuse=".2 .2 .2" specular="0.1 0.1 0.1"/>
      <map znear="0.01"/> -->
      <!-- <scale contactwidth=".02" contactheight=".5"/> -->
  </visual>

  <default>
    <!-- <geom friction="0.4" solimp="0 0.95 0.02"/> -->
  </default>

  <extension>
		<plugin plugin="mujoco.elasticity.solid" />
	</extension>

  <statistic center="0 0 1" extent="1" meansize=".1"/>

  <worldbody>
    <light pos="1 0 .3" dir="-1 0 -.3"/>
    <light pos="-1 0 .3" dir="1 0 -.3"/>

    <geom name="floor" pos="0 0 0" type="plane" size="1 1 .01"/>

    <body name="biotac" pos="0 0 0.1">
      <!-- <freejoint/> -->
      <geom name="core_bc" type="mesh" mesh="top_side_core_bc" euler="90 0 0" contype="1" pos="0 0 0" conaffinity="1" rgba=".5 .5 1 1"/>

      <flexcomp name="softbody" type="gmsh" dim="3" rgba=".8 .2 .2 0.3" radius=".005" euler="90 0 0" pos="0 0 0" mass="5" file="int_ext_skin_combined_ftw.msh">
        <edge equality="true" solref="0.01 1" solimp=".95 .99 .0001" damping="3"/>
        <contact selfcollide="none" conaffinity="1" internal="false"/>
        <plugin plugin="mujoco.elasticity.solid">
            <config key="young" value="8e4" />
            <config key="poisson" value="0.2" />
            <config key="damping" value=".0001" />
        </plugin>
    </flexcomp>
    </body> 

  </worldbody>
</mujoco>
```
(Should I create a new issue since the original problem has been resolved?）

sIHURs, 2024-09-23:
> Do you still get the issue if you set
> 
> `<contact selfcollide="none" internal="false"/>`
> 
> I suspect that the solver might be trying to allocate a huge number of self- and internal collisions.

It's this issue that has been resolved; I can open it now. Thank you very much.

I have a new questio, actually is that I have added the rigid body to the simulation (as described in https://github.com/google-deepmind/mujoco/issues/2077#issuecomment-2368062610), but I have not yet pinned the area where the soft body contacts the rigid body. However, the soft body (the red part, which looks like a balloon) is expanding. Why is this happening? 
[image]
[image]

Also, when it is first loaded, the soft body does not appear to be the correct size, as I have not applied any scaling. The correct situation is shown in the picture.
[image]

xml:
```xml
<mujoco model="biotac-sim">
  <!-- <compiler autolimits="true"/> -->
  <option solver="CG"/>
  <size memory="8G"/>
  
  <extension>
    <!-- add extension -->
  </extension>

  <asset>
    <mesh name="top_side_core_bc" file="top_side_core_bc.obj"/>
    <!-- <mesh name="skin" file="int_ext_skin_combined_ftw.msh"/> -->
  </asset>

  <visual>
      <!-- <headlight ambient=".7 .7 .7" diffuse=".2 .2 .2" specular="0.1 0.1 0.1"/>
      <map znear="0.01"/> -->
      <!-- <scale contactwidth=".02" contactheight=".5"/> -->
  </visual>

  <default>
    <!-- <geom friction="0.4" solimp="0 0.95 0.02"/> -->
  </default>

  <extension>
		<plugin plugin="mujoco.elasticity.solid" />
	</extension>

  <statistic center="0 0 1" extent="1" meansize=".1"/>

  <worldbody>
    <light pos="1 0 .3" dir="-1 0 -.3"/>
    <light pos="-1 0 .3" dir="1 0 -.3"/>

    <geom name="floor" pos="0 0 0" type="plane" size="1 1 .01"/>

    <body name="biotac" pos="0 0 0.1">
      <!-- <freejoint/> -->
      <geom name="core_bc" type="mesh" mesh="top_side_core_bc" euler="90 0 0" contype="1" pos="0 0 0" conaffinity="1" rgba=".5 .5 1 1"/>

      <flexcomp name="softbody" type="gmsh" dim="3" rgba=".8 .2 .2 0.3" radius=".005" euler="90 0 0" pos="0 0 0" mass="5" file="int_ext_skin_combined_ftw.msh">
        <edge equality="true" solref="0.01 1" solimp=".95 .99 .0001" damping="3"/>
        <contact selfcollide="none" conaffinity="1" internal="false"/>
        <plugin plugin="mujoco.elasticity.solid">
            <config key="young" value="8e4" />
            <config key="poisson" value="0.2" />
            <config key="damping" value=".0001" />
        </plugin>
    </flexcomp>
    </body> 

  </worldbody>
</mujoco>
```
(Should I create a new issue since the original problem has been resolved?）

lyd405121, 2024-09-24:
## Backgroud
- I am making a synthetic tool for robot training, the physics engine is mujoco 

https://github.com/user-attachments/assets/b79713ef-c30f-4993-87f4-574750d9c969

- Render engine using self-developed realtime raytracing(same tech from issac)
- It was great when the whole scene is built with rigid body
- But when it cames to soft body，the performance can not be realtime and has some bug
- Here is the real scene what I want to achieve

https://github.com/user-attachments/assets/75608369-7a9c-4b94-8edb-2307c5ce64ce

## Sample Scene
- To demonstrate it I wrote a simple soft scene with only one tetrahedron below

https://github.com/user-attachments/assets/48790ef5-f2aa-456e-9116-f3a0059ef2b1

- And I use fTetWild to tedrahetralize the tiger mesh as [the official doc told](https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-flexcomp-element)

https://github.com/user-attachments/assets/9fd4e6be-97d9-483b-90ea-71761bfb9f26

- The video shows the performance is not acceptable, and the tiger mesh never falls down to the groud
- Here is the origin scene data - [softbody.zip](https://github.com/user-attachments/files/17096499/softbody.zip)

## Question
- ① Am I doing right for building such kind of tedrahetra mesh to simulate?
- ② How to make it faster?
- ③ Is there any plan to optimize performance for soft body simulation?
- ④ If I using other soft body algorithm to replace mujoco's, is this feasible?

lyd405121, 2024-09-24:
## Thanks
- I have try the method as @quagla said in #2079 
- It works!

## Solution
- To solve the low performance, use this 
```html
<contact selfcollide="none" internal="false"/>
```
- To make mesh drop, choose small time step
```html
<option solver="CG" tolerance="1e-6" integrator="implicitfast" timestep="0.001"/>
```

## Final result
- Here is the [mjcf](https://github.com/user-attachments/files/17108871/softbody.zip)

https://github.com/user-attachments/assets/2a62a750-45d6-4d95-a007-50d6f96b9db8

quagla, 2024-09-24:
> > Do you still get the issue if you set
> > `<contact selfcollide="none" internal="false"/>`
> > I suspect that the solver might be trying to allocate a huge number of self- and internal collisions.
> 
> It's this issue that has been resolved; I can open it now. Thank you very much.
> 
> The current issue is that the soft body seems to be too soft, as it deforms excessively and sags significantly, and it also tends to break apart. I have adjusted the parameters shown in the XML below, but there has been no noticeable change. Which parameter should I focus on?

You should either use edge `equality="true"` or `elasticity.solid`, not both at the same time. If you want something very stiff, I'd recommend the former, if you want something softer, the latter (since it may require a very small time step when you increase the parameters). But your solref and solimp look already quite stiff, so I'm surprised you are getting a soft solution? Maybe your boundary conditions are not what you would like them to be, i.e. you either have to pin some vertices or add equality constraints.

quagla, 2024-09-24:
> ## Thanks
> * I have try the method as @quagla said in [[Soft Body]Simulate performance drops when the particle reach hundred #2079](https://github.com/google-deepmind/mujoco/issues/2079)
> * It works!

Great to hear! Sorry I deleted your other comment, but the original link is present in your response anyway :)

lyd405121, 2024-09-27:
## Another Question
- If I want to add a detailed & texutred mesh driven by the simplified tetrahedron mesh 
- What should I do?
## Potential Solution
- I noticed mujoco has a skin node
- If I make a skin animation between particles and detailed mesh using weights just like [Skeletal Animation](https://en.wikipedia.org/wiki/Skeletal_animation)
- Is that possible?

quagla, 2024-09-30:
Yes you can use a skin referencing the bodies that you want to use as bones by name.

closed 2024-09-30 by quagla