google-deepmind/mujoco #2077: How to attach a soft body on a rigid body?

sIHURs, 2024-09-22:
hi~

I would like to ask how to attach a soft body to a rigid body, similar to how rubber is fixed to plastic in the real world. For example, in a simple scenario, if there are two cubes in MuJoCo, how should this be implemented?

a example like this:
```xml
<mujoco model="soft_rigid">
    <include file="scene.xml"/>

    <compiler autolimits="true"/>

    <option solver="Newton" tolerance="1e-6" timestep=".001" integrator="implicitfast"/>

    <!-- <size memory="10M"/> -->

    <visual>
        <map stiffness="100"/>
    </visual>

    <extension>
            <plugin plugin="mujoco.elasticity.solid" />
    </extension>

    <worldbody>
        <body name="soft_rigid">
            <freejoint/>
            <flexcomp name="soft" type="grid" count="4 4 4" spacing=".2 .2 .2" pos="0 0 1.645"
                    radius=".005" rgba="0 .7 .7 1" dim="3" mass="5">
                <contact condim="3" solref="0.01 1" solimp=".95 .99 .0001" selfcollide="none"/>
                <edge damping="1"/>
                <plugin plugin="mujoco.elasticity.solid">
                    <config key="young" value="8e4" />
                    <config key="poisson" value="0.2" />
                    <config key="damping" value=".0001" />
                </plugin>
            </flexcomp>
                <!-- <joint name="soft" type="slide" axis="0 0 1" damping="500"/> -->
            <geom name="rigid" type="box" size=".35 .35 0.05" pos=".1 .1 2" density="300"/>
<!-- 
            <equality>
                <connect name="right_boundary" body1="B_last" anchor="0.07 0 0"/>
            </equality> -->
        </body>

    </worldbody>
<!-- 
    <actuator>
        <motor name="soft" joint="soft" gear="-1 0 0 0 0 0" ctrlrange="-150 300"/>
        <motor name="stiff" joint="stiff" gear="-1 0 0 0 0 0" ctrlrange="-150 1600"/>
    </actuator> -->
</mujoco>
```

This is a modified example from the model/plugin/..., where one is a rigid cube and the other is a soft cube. How can I fix them together at the surfaces in contact?

Thanks for any tips or suggestions!

yifan

quagla, 2024-09-23:
Hi, you don't need to use a constraint, you can just define the flexcomp as a child of the rigid body you would like to attach it to. A flexcomp is nothing but a sequence of bodies with 3 slider joints in each one of them.

sIHURs, 2024-09-23:
> Hi, you don't need to use a constraint, you can just define the flexcomp as a child of the rigid body you would like to attach it to. A flexcomp is nothing but a sequence of bodies with 3 slider joints in each one of them.

hello，I tried in several ways, but I haven't been successful. Could you please provide a mini example? Thank you very much! I'm very new to this.

quagla, 2024-09-23:
What issue are you getting with your example above?

sIHURs, 2024-09-23:
> What issue are you getting with your example above?

I want the rigid body (created with geom) and the soft body (created with flexcomp) to be fixed at the contact surface, just like in reality where rubber is glued to a piece of plastic. As shown in the image below, the bottom surface of the rigid body should be glued to the top surface of the soft body.
[image]

However, in the code I wrote, the two objects are separating.
[image]
[image]

This is a simple example I envisioned. The real problem I need to solve involves more complex shapes of the objects, as shown in the image below.
[image]
The blue part represents the rubber section, while the gray part represents the rigid body. The area where they make contact needs to be fixed. The rubber section is hollow, and inside it, there is another rigid body that has a fixed contact area with the rubber part.

thanks very much!

quagla, 2024-09-23:
Okay I see, the issue is that you have to pin the soft body to the rigid body because a flexcomp is always a soft body will no boundary constraints. See https://github.com/google-deepmind/mujoco/blob/main/model/flex/trampoline.xml for an example of 4 pinned vertices (note that this file has changed folder since the release if you can't find it in your local folder). In your case, you want to pin the entire surface, this part of the docs will help you https://mujoco.readthedocs.io/en/latest/XMLreference.html#flexcomp-pin let me know how you get on

sIHURs, 2024-09-23:
It fixed them together, thank you very much! However, there are still some questions.

I set a pin, and indeed there is a point connecting the soft body to the rigid body in the simulation.
```xml
<flexcomp name="soft" type="grid" count="4 4 4" spacing=".2 .2 .2" pos="0 0 1.645"
                    radius=".005" rgba="0 .7 .7 1" dim="3" mass="5">
                <contact condim="3" solref="0.01 1" solimp=".95 .99 .0001" selfcollide="none"/>
                <edge damping="1"/>
                <plugin plugin="mujoco.elasticity.solid">
                    <config key="young" value="8e4" />
                    <config key="poisson" value="0.2" />
                    <config key="damping" value=".0001" />
                </plugin>
                <pin grid="0 0 0"/>
            </flexcomp>
```
[image]
[image]
[image]

1.However, after looking at the documentation, I am unclear how to locate the ID of this point on the object, as it is not the label of the body. Additionally, if I want to select multiple points at once, and they are not arranged in order (im not sure, like it is loaded from a .msh file), how should I use the gridrange parameter to handle this? Do I have to write multiple gridrange entries to select all the points?

2.
Using pin will only pin the soft body to its parent body, right? In the example you provided, since the soft body is created in the worldbody, the four points are pinned directly in space.

quagla, 2024-09-23:
The way to think about a pin is that it removes degrees of freedom, rather than defining a constraint. Remember that the flex is a collection of bodies at the mesh vertices with 3 sliders each. A pin just removes the sliders, so the vertex will remain at the same position where it is initialized in the parent body.

If you have a general mesh with so special ordering, you probably have to take note manually of the vertex numbers you want to pin. If you can load the mesh in python, maybe do a loop and print a the indices that are above/below a certain coordinate of interest.

sIHURs, 2024-09-23:
> The way to think about a pin is that it removes degrees of freedom, rather than defining a constraint. Remember that the flex is a collection of bodies at the mesh vertices with 3 sliders each. A [...]

I will try these, thank you very much!

closed 2024-09-23 by sIHURs