google-deepmind/mujoco #2079: [Soft Body]Simulate performance drops when the particle reach hundred

lyd405121, 2024-09-23:
### The feature, motivation and pitch

## Backgroud
- I am making a synthetic tool for robot training, the physics engine is mujoco 

https://github.com/user-attachments/assets/b79713ef-c30f-4993-87f4-574750d9c969

- Render engine using self-developed realtime raytracing(same tech from issac)
- It was great when the whole scene is built with rigid body
- But when it cames to soft body，the performance can not be realtime and has some bug
- Here is the real scene what I want to achieve

https://github.com/user-attachments/assets/75608369-7a9c-4b94-8edb-2307c5ce64ce

## Sample Scene
- To demonstrate it I wrote a simple soft scene with only one tetrahedron below

https://github.com/user-attachments/assets/48790ef5-f2aa-456e-9116-f3a0059ef2b1

- And I use fTetWild to tedrahetralize the tiger mesh as [the official doc told](https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-flexcomp-element)

https://github.com/user-attachments/assets/9fd4e6be-97d9-483b-90ea-71761bfb9f26

- The video shows the performance is not acceptable
- Here is the origin scene data - [softbody.zip](https://github.com/user-attachments/files/17096499/softbody.zip)

## Question
- ① Am I doing right for building such kind of tedrahetra mesh to simulate?
- ② How to make it faster?
- ③ Is there any plan to optimize performance for soft body simulation?
- ④ If I using other soft body algorithm to replace mujoco's， is this feasible?
https://github.com/user-attachments/assets/d13627c8-2a0b-4d7b-bfbe-52ecdbbf57ce

quagla, 2024-09-23:
Hi, the soft body has nearly 800 vertices, so I would expect the performance to be comparable to a scene with 800 rigid bodies with 3 sliders each. Can you set `selfcollide` (https://mujoco.readthedocs.io/en/latest/XMLreference.html#flex-contact) to `none` and run https://github.com/google-deepmind/mujoco/blob/main/sample/testspeed.cc to get a breakdown of the timing?

I do have plans for much faster soft bodies but I'm not sure when I'll have time to finish it (I started it but it's WIP).

quagla, 2024-09-23:
I loaded your model. The issue is that you have thousands of internal contact. Try with 

```<contact selfcollide="none" internal="false"/>```

I am not sure if this is caused by your choice of `radius`, you could play with it and see. Note that you also have to reduce the time step to prevent instabilities.

sIHURs, 2024-09-23:
hi,

I would like to ask, if the mesh file I need contains more than 4000 points and over 15000 faces, how would MuJoCo handle this? Approximately how much memory would it require? Are there any similar examples? I encountered issues when loading it, and either the simulator crashes or becomes extremely slow and unresponsive. Addressed by https://github.com/google-deepmind/mujoco/issues/2073#issue-2540375215

The specific task requires to get deformation data (coordinates of points) or force calculations for the soft body.

Thanks!

yifan

quagla, 2024-09-23:
Do you get any information in the log when it crashes?

sIHURs, 2024-09-23:
If i use .msh file:
```xml
<flexcomp name="softbody" type="gmsh" dim="3" rgba=".8 .2 .2 1" radius="0.001" pos="0 0 .3" mass="10" scale="1 1 1" file="int_ext_skin_combined_ftw.msh">
```
it will be rendered at first like in the picture, but soon it just shows not responding on GUI.
[image]

if use .obj file maybe like:
```xml
<flexcomp name="softbody" type="mesh" dim="2" rgba=".8 .2 .2 1" radius="0.001" pos="0 0 .3" mass="10" scale="1 1 1" file="int_ext_skin_combined_ftw.obj">
```
it has log:
```bash
MuJoCo version 3.1.6
Plugins registered by library 'libsensor.so':
    mujoco.sensor.touch_grid
Plugins registered by library 'libsdf.so':
    mujoco.sdf.bolt
    mujoco.sdf.bowl
    mujoco.sdf.gear
    mujoco.sdf.nut
    mujoco.sdf.torus
    mujoco.sdf.sdflib
Plugins registered by library 'libelasticity.so':
    mujoco.elasticity.cable
    mujoco.elasticity.membrane
    mujoco.elasticity.shell
    mujoco.elasticity.solid
Plugins registered by library 'libactuator.so':
    mujoco.pid
ERROR: mj_stackAllocInt: requested size is too large.

Press Enter to exit ...
```

xml file:
```xml
<mujoco model="soft">
  <option solver="CG"/>
  <size memory="8G"/>
  <worldbody>
    <light diffuse=".5 .5 .5" pos="0 0 5" directional="true" dir="0 0 -1" />
	  <geom name="floor" type="plane" size="10 10 10" pos="0 0 0" rgba="1 1 1 1" />
    <flexcomp name="softbody" type="gmsh" dim="3" rgba=".8 .2 .2 1" radius="0.001" pos="0 0 .3" mass="10" scale="1 1 1" file="int_ext_skin_combined_ftw.msh">
        <edge equality="true"/>
    </flexcomp>
  </worldbody>
</mujoco>
```
I had set memory size of 8G.

msh file:
[skin.zip](https://github.com/user-attachments/files/17100461/skin.zip)

quagla, 2024-09-23:
I think you should post this in the other discussion. I will close this one since the original issue was solved.

closed 2024-09-23 by quagla