google-deepmind/mujoco #2082: zero drag problem

kassasin, 2024-09-24:
I have a question. When I do a zero-force drag problem on a robotic arm, after compensating for gravity, although there is a slight error, the robotic arm can still be stationary due to the influence of static friction. How should I achieve this in mujoco? Because as far as I know there is no Coulomb friction in mujoco.

kassasin, 2024-09-24:
frictionloss seems can't achieve this goal

quagla, 2024-09-24:
https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-noslip-iterations

kassasin, 2024-09-24:
Thanks very much, it works. But is also leads to small motion,can i  eliminate it completely?

quagla, 2024-09-24:
The best solution I can offer is playing with the parameters and see if you can improve the behavior. I'm not sure if you can really remove the motion completely since everything in MuJoCo has some small compliance by design.

kassasin, 2024-09-24:
Thank you very much .

closed 2024-09-24 by kassasin