google-deepmind/mujoco #2084: MuJoCo's elasticity plugin yields unphysical/unstable/inaccurate results for soft-gripper simulation

kurtenkera, 2024-09-24:
### Intro

Hi!

I am an Australian PhD researcher at the University of Queensland, and I use MuJoCo for my robotics research.

### My setup

I currently use MuJoCo via the Python bindings.

### What's happening? What did you expect?

I'm trying to model the soft-gripper from the paper "Optimal Design of a Soft Robotic Gripper for Grasping Unknown Objects" (Liu et. Al, 2018) using MuJoCo. I am just modelling the moving platform, fixed platform and two compliant fingers seen in the diagram below, and ignoring all additional features.

[image]

I have carefully selected the lengths and dimensions of my platforms and fingers to match those in the paper. Ultimately, I want the gripper to be able to grasp and lift a range of different test objects. However, I'm experiencing 3 concerning problems, which I suspect might be due to MuJoCo's elasticity plugin.

These are my three main problems: **(1) my simulation diverges when I set Young's Modulus (YM) and Poisson's Ratio (PR) values equal to that in the paper**; **(2) when I use _arbitrary_ YM and PR values that don't diverge, my two platforms erroneously penetrate one another**; **(3) introducing a slide joint to enable the entire gripper system to lift causes unphysical, unstable behaviour**. 

I will provide some screenshots/videos to illustrate the three problems clearly. The video below shows my gripper deforming nicely with arbitrarily selected values of YM = 5e4, and PR = 0.2. 

[nicedeform.webm](https://github.com/user-attachments/assets/fa6b18e6-80b9-42eb-ad69-3d72c6b311eb)

For problem (1), we see that when I set YM = 11e6 and PR = 0.45 I get the following divergent simulation below. This is problematic as these are the values used in the paper, and therefore MuJoCo doesn't allow me to simulate the values/soft material that I'm actually interested in.

[unstableYM&PR.webm](https://github.com/user-attachments/assets/169d55da-5d48-45a6-ad94-699c13c41ebf)

For problem (2), the video below shows that my two platforms can erroneously penetrate, which I absolutely want to prevent but am unsure how to.

[penetrating.webm](https://github.com/user-attachments/assets/96efb487-93cb-4025-aa30-264ff1773cbb)

For problem (3), the video below shows that when I introduce a slide joint for lifting the entire gripping system, I get unphysical oscillating platforms, despite applying no forces at the motors. This problem is particularly strange!

[oscillating.webm](https://github.com/user-attachments/assets/0002ce0c-2752-4d87-b215-c2d5e0c00f5a)

### Steps for reproduction

Do the following to reproduce problem (1):

1. Open `no_lift.xml` contained in the zip file below in a text editor and set `young = 11e6` and `poisson = 0.45` for **both** left and right soft fingers.
2. Load the model `no_lift.xml` contained in the zip file below into the GUI simulator and run.

To reproduce problem (2):

1. Open `no_lift.xml` contained in the zip file below in a text editor and set `young = 5e4` and `poisson = 0.2` for **both** left and right soft fingers.
2. Load the model `no_lift.xml` contained in the zip file below into the GUI simulator. Set the `close_gripper` control value to 1, and run the simulation for ~10 seconds.

To reproduce problem (3):

1. Load the model `lift.xml` contained in the zip file below into the GUI simulator. Run the simulation for ~10 seconds.

### Minimal model for reproduction

In the following zip file, I have included a 3D MSH file of my compliant finger being modelled (the same finger is used for the left and right fingers), and two relevant minimal XML files `lift.xml` and `no_lift.xml`: 
[softgripper.zip](https://github.com/user-attachments/files/17111928/softgripper.zip).

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

quagla, 2024-09-24:
Hi @kurtenkera I played with your model, I have two comments:
- The time step is big, I would reduce it by 5-10x in order to avoid the oscillations and preventing the explosions. Remember we are doing explicit integration.
- You have self-intersections in the starting configuration. If you press C, you will see them. I would suggest reducing radius to perhaps 1e-3 (see https://mujoco.readthedocs.io/en/latest/XMLreference.html#deformable-flex)

kurtenkera, 2024-09-24:
Thanks @quagla! Those two suggestions have helped resolve problem 3 i.e. the oscillating platform doesn't oscillate anymore. 

However, problem 1 and 2 still remain. As for problem 1, I can't run my `lift.xml` or `no_lift.xml` with `young = 11e6` and `poisson = 0.45` - the simulation still diverges. Is this issue resolvable via a simple XML file/modelling modification or is this a deeper issue to do with the elasticity plugin itself? 

I also still see my two platforms penetrate when I apply a sufficiently large enough motor force. How do I prevent this penetration of these geoms which are meant to be rigid bodies?

quagla, 2024-09-24:
Problem 1 is just an intrinsic feature of elasticity. The more you increase the stiffness, the more you have to reduce the timestep. The elasticity plugin does not support implicit integration yet.

For problem 2, you can try to make contact stiffer.  See https://mujoco.readthedocs.io/en/latest/modeling.html#csolver The contact in mujoco are always compliant, so for a large enough force you are going to get some penetration.

quagla, 2024-09-25:
I think what happened is that the change in the doc was released too soon. Setting those parameters in flexcomp won't have any effect until the next release, sorry for the confusion!

There won't be any new feature added when plugins are in the engine, but finally you won't have to specify the plugins in the XML, which sometimes is a bit cumbersome.

I wouldn't use elasticity and edge equality at the same time, since it's hard to predicts what to expect from their combined effect.

quagla, 2024-09-25:
If you need the edge constraints, you probably need *only* the edge constraints. The constraints are just a workaround for when the time step cannot solve the propagation of the elastic waves into a solid, causing an instability. With constraints, the propagation is instantaneous (quasistatic) and the system becomes stable. I think in the paper you are trying to reproduce they were not using them?

The behavior of elasticity with no constraints in your video looks wrong. It's like you have no elasticity at all. Are you using `mujoco.elasticity.solid` or `<elasticity/>`? If you are on the latest release you have to use the former, if you build from source you can use the latter.

kurtenkera, 2024-09-25:
I'm using `mujoco.elasticity.solid` in my videos with no edge constraints and I am on the latest release.

So you're saying that edge constraints are really only used for simulating stiff, deformable materials with a Young's Modulus on the order of 10e6 Pascals and above? I'd say that my soft fingers fall into this category. The fingers in the paper have a Young's Modulus of 12e6 Pascals (they are pretty much rubber in the paper). If that's the case, then it sounds like I just need to abandon using `mujoco.elasticity.solid` and simply use edge constraints instead. 

However, I wanted to take advantage of the Saint-Venant Kirchhoff material model that you've implemented in MuJoCo to achieve a more physically accurate simulation. But since simulations diverge for large stiffnesses, or the timestep has to be reduced to a super low value that slows the simulation down too much for my needs, I won't be able to do this? Can you see any way that I can make use of the elasticity plugin for my application, or should I only consider using it for simulating less stiff materials?

quagla, 2024-09-25:
It really depends what time step you are prepared to use.. However, with 1e6 I'm getting something much stiffer than in your videos, so there might be something wrong in your setup? Note that the reason why the time step is very small is that I am additionally using the plugin damping since I think it's more realistic than edge damping.

```
<mujoco>
	
  <option gravity = "0 0 -9.81" timestep="0.0005"/>
  
  <extension>
    <plugin plugin="mujoco.elasticity.solid" />
  </extension>
  
  <size memory="100M"/>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3"
      markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>
  
  <worldbody>
    <light pos="0 0 1.8" dir="0 0 -1" directional="true"/>
    <camera name="closeup" pos="0 -1.4 0.3" xyaxes = "1 0 0 0 0.1 1"/>
    <geom name="floor" size="0 0 0.05" type="plane" material = "groundplane"/>

    <!-- UPPER SLIDING PLATFORM -->
    <body name = "movingplatform" pos = "0 0 0.01">
        <joint name = "close_gripper" type = "slide" axis = "0 0 -1" damping = "5000"/>
        <geom name="movingplatform" type="box" size="0.03 0.02 0.01" pos="0 0 0.18" rgba="0.3 0.3 0.3 .9"
         density = "1000"/>

        <!-- LEFT SOFT-FINGER -->
        <flexcomp name="left" euler = "90 180 -44.5" type="gmsh" file="finger.msh" dim="3" pos="-0.092 -0.015 0.03" radius="0.001"
                  rgba="1 1 1 1" mass="5">
          <contact condim="3" solref="0.03 1" solimp=".95 .99 .0001" selfcollide="none"/>
          <plugin plugin="mujoco.elasticity.solid">
             <config key="poisson" value="0.2"/>
             <config key="young" value="5e5"/>
             <config key="damping" value=".003"/>
          </plugin>
        </flexcomp>

        <!-- RIGHT SOFT-FINGER -->
        <flexcomp name="right" euler = "90 0 -44.5" type="gmsh" file="finger.msh" dim="3" pos="0.092 0.015 0.03" radius="0.001"
                  rgba="1 1 1 1" mass="5">
          <contact condim="3" solref="0.03 1" solimp=".95 .99 .0001" selfcollide="none"/>
          <plugin plugin="mujoco.elasticity.solid">
             <config key="poisson" value="0.2"/>
             <config key="young" value="5e5"/>
             <config key="damping" value=".003"/>
          </plugin>
        </flexcomp>
    </body>
      
    <!-- FIXED LOWER PLATFORM -->
    <body name = "fixedplatform" pos = "0 0 0.106">
       <geom name="fixedplatform" type="box" size="0.03 0.02 0.01" pos="0 0 0" rgba="0.3 0.3 0.3 .9"/>
    </body>
    
    <!-- test object -->
    <body pos="0 0 0.02">
      <joint type = "free"/>
      <geom name="testobject" type="box" size="0.03 0.01 0.02" rgba="1 1 0 1"/>
    </body>
    
  </worldbody>
  
  <actuator>
    <motor name="close_gripper" joint="close_gripper" gear="100"/>
  </actuator>
  
  <equality>
    <!-- CONNECTING THE RIGHT FINGER TO UPPER PLATFORM -->
    <connect body1 = "movingplatform" body2="right_2" anchor = "0.03 0.012 0.19"/> 
    <connect body1 = "movingplatform" body2="right_2" anchor = "0.03 0.012 0.17"/> 
    <connect body1 = "movingplatform" body2="right_6" anchor = "0.03 -0.012 0.19"/> 
    <connect body1 = "movingplatform" body2="right_6" anchor = "0.03 -0.012 0.17"/> 
    <!-- CONNECTING THE RIGHT FINGER TO LOWER PLATFORM -->
    <connect body1 = "fixedplatform" body2="right_3" anchor = "0.03 0.012 0.01"/>
    <connect body1 = "fixedplatform" body2="right_3" anchor = "0.03 0.012 -0.01"/> 
    <connect body1 = "fixedplatform" body2="right_7" anchor = "0.03 -0.012 0.01"/> 
    <connect body1 = "fixedplatform" body2="right_7" anchor = "0.03 -0.012 -0.01"/> 
    
    <!-- CONNECTING THE LEFT FINGER TO UPPER PLATFORM -->
    <connect body1 = "movingplatform" body2="left_2" anchor = "-0.03 -0.012 0.19"/> 
    <connect body1 = "movingplatform" body2="left_2" anchor = "-0.03 -0.012 0.17"/> 
    <connect body1 = "movingplatform" body2="left_6" anchor = "-0.03 0.012 0.19"/> 
    <connect body1 = "movingplatform" body2="left_6" anchor = "-0.03 0.012 0.17"/> 
    <!-- CONNECTING THE RIGHT FINGER TO LOWER PLATFORM -->
    <connect body1 = "fixedplatform" body2="left_3" anchor = "-0.03 -0.012 0.01"/> 
    <connect body1 = "fixedplatform" body2="left_3" anchor = "-0.03 -0.012 -0.01"/> 
    <connect body1 = "fixedplatform" body2="left_7" anchor = "-0.03 0.012 0.01"/> 
    <connect body1 = "fixedplatform" body2="left_7" anchor = "-0.03 0.012 -0.01"/> 
  </equality>
  
</mujoco>
```

commit 1eb6fc22 (2024-09-25): Disallow using elasticity and edge constraints at the same time.

Fixes #2084.

PiperOrigin-RevId: 678614880
Change-Id: I44eead3dc4edf82d725a3063b927a421c660721a
closed 2024-09-27 by quagla