google-deepmind/mujoco #2087: How to actuate a free joint?

omarrayyann, 2024-09-24:
I am looking for a way to actuate a free joint using position and quaternion data. While I am aware that mocaps exist and can be applied to bodies, I want to achieve this through actuators so that collisions and other physical interactions still occur. I attempted to approximate this by using six joints (three sliders and three hinges) instead of a free joint, but encountered limitations when it comes to continuously tracking a rotations using an Euler-based controller. Is there an alternative way to actuate a free joint using actuators  to be able to use pos and quat?

yuvaltassa, 2024-09-24:
Yes, this is possible. Copy what is happening in this [reference model](https://github.com/google-deepmind/mujoco/blob/main/test/engine/testdata/actuation/refsite.xml).

- The translation actuators are exactly what you want.
- The rotation, not exactly. I won't go into the reason but we can't use quaternions as inputs to `ctrl` and joint angles are a problem for the same reason your Euler angles don't work well. You'll see if you make the rotation ranges in the example model bigger that you get instabilities around pi. So your best bet is to implement the rotation controller yourself. Instead of `<position>` use `<motor>` for the rotations and then write the logic yourself that goes from quat to torque. And don't forget to add damping!

closed 2024-09-24 by yuvaltassa

omarrayyann, 2024-09-24:
Thanks a lot, will try adding a rotation controller. Also you're referring to the gimbal lock issue I'm assuming right?

yuvaltassa, 2024-09-24:
It's not really about gimbal lock in this case, is a topological thing. Consider a simple angle x represented as [cos(x), sin(x)] (analogous to 3D rotation and quaternion, for our purposes). If the current angle is 0 (= [1, 0]) and the desired angle is pi (= [-1, 0]), what torque should a "position" controller apply? Should it rotate clockwise? anti-clockwise? no torque at all? 

As you can see there is no satisfactory answer. So instead of dealing with this I will let you discover for yourself that quaternion-denominated position controllers are a bad idea, unless you can guarantee that you are always close to the target (maybe you can do that).

omarrayyann, 2024-09-25:
Thanks for the explanation, that makes a lot of sense. I got it to work now.