google-deepmind/mujoco #2089: Error in creating 2 digit by 2 digit rangefinder array using <replicate/>

yzwfromk, 2024-09-24:
### Intro

Hi!

I am a graduate student at Georgia Tech, I use MuJoCo for my research on bipedal robots and simulating LiDAR and Depth camera.

### My setup

Ubuntu 20.04.6 LTS, 64 bit
Mujoco Python 3.2.0

### What's happening? What did you expect?

Recently I tried to mimic LiDAR using replicate for rangefinder sensors. However, I found a weird issue while using replicate as below:
```
<mujoco model='replicate error'>
  <compiler angle='degree' eulerseq='xyz'/>
  
  <worldbody>
    <geom name='ground' type='plane' size='0 0 1'/>
    <light diffuse="1 1 1" pos="0 0 3" dir="0 0 -1"/>
    
    <body name='base' pos='0 0 1' euler='0 0 0'>
      <geom type='cylinder' size='0.05 0.0325' pos='0 0 0' euler='-0 0 -0'/>
      <replicate count="91" offset="0 0 0" euler="0 0 1">
        <replicate count="20" offset="0 0 0" euler="0 1 0">
          <site name="rangefinder_array" pos="0 0 0" zaxis="1 0 -1"/>
        </replicate>
      </replicate>
    </body>
  </worldbody>

  <sensor>
    <rangefinder  name='LiDAR' site='rangefinder_array'/>
  </sensor>

</mujoco>

```

When load the XML file, it gives me error saying:
`
XML Error: Error: repeated name 'LiDAR0000' in sensor
Element 'replicate', line 11
`

However, If I change the number '20' to '9' or '111' or whatever number other than 2 digits number, it works without any error.
So the thing is, when I create 2 digits number-by-1-digit number (or 3 digit number), it works. However, for 2 digits number-by-2 digits number, doesn't work.

Could you help me with this?

### Steps for reproduction

You can see the error message by running
`python3 -m mujoco.viewer`
and loading the above XML file.

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2024-09-24:
Does adding a separator using the `sep` [attribute](https://mujoco.readthedocs.io/en/latest/XMLreference.html#replicate-r) in one or both `<replicate>` resolve the issue?

(@quagla: even if that does resolve the OP's specific issue, there might still be a bug, since the sensor `LiDAR0000` should only be created once, right? why is it created twice?)

yzwfromk, 2024-09-24:
Yes it works if I add `sep` to one of them (both of them doesn't work), Thanks!
Let me know if there are any updates or better solution available.

yuvaltassa, 2024-09-24:
Good to know.

> Yes it works if I add sep to one of them (both of them doesn't work),

Can you please spell out what works and what doesn't work (and what's the error) for these three cases?

yzwfromk, 2024-09-24:
```
      <replicate count="91" offset="0 0 0" euler="0 0 1">
        <replicate count="20" sep='sep_' offset="0 0 0" euler="0 1 0">
          <site name="rangefinder_array" pos="0 0 0" zaxis="1 0 -1"/>
        </replicate>
      </replicate>
```
, or
```
      <replicate count="91" sep='sep' offset="0 0 0" euler="0 0 1">
        <replicate count="20" offset="0 0 0" euler="0 1 0">
          <site name="rangefinder_array" pos="0 0 0" zaxis="1 0 -1"/>
        </replicate>
      </replicate>
```
, or
```
      <replicate count="91" sep='sep' offset="0 0 0" euler="0 0 1">
        <replicate count="20" sep='sep_'  offset="0 0 0" euler="0 1 0">
          <site name="rangefinder_array" pos="0 0 0" zaxis="1 0 -1"/>
        </replicate>
      </replicate>
```
works. However,
```
      <replicate count="91" sep='sep' offset="0 0 0" euler="0 0 1">
        <replicate count="20" sep='sep'  offset="0 0 0" euler="0 1 0">
          <site name="rangefinder_array" pos="0 0 0" zaxis="1 0 -1"/>
        </replicate>
      </replicate>
```
which is using the same string for both `sep` doesn't work and gives an error saying `LiDARsep00sep00` is repeated

yuvaltassa, 2024-09-24:
Thanks! That's very helpful

quagla, 2024-09-25:
I am not able to reproduce the error, the model loads fine for me..

[image]

quagla, 2024-09-25:
I think 6928e09f362aa4ba1d335e6f060dda72b1476941 fixed the issue. It looks like you are using an older version so the solution is to update mujoco :) I will mark this as closed.

closed 2024-09-25 by quagla