google-deepmind/mujoco #2093: How to firmly attach other bodies to the last joint of the cable?

avaa999, 2024-09-26:
Hello! I want to start by expressing my gratitude for the fantastic work and for sharing it with all of us!

In your example " mujoco/model/plugin/elasticity/cable.xml", the last joint of the cable is connected to the ball. 

<equality> 
    <connect name="right_boundary" body1="B_last" body2="slider" anchor=".025 0 0"/>
  </equality>

However, when I connect the last joint to a box, I notice that the box remains relatively stationary in position with respect to the last joint, but there is a relative orientation difference between them. I also tried using "weld," 
<weld name="relpose" body1="B_last" body2="slider" relpose="0 0 0 1 -.3 0 0"/>

but the same issue persists. Is there a good solution to ensure that the box and joint are firmly connected without any relative pose difference?

avaa999, 2024-09-26:
Uploading 木块错误.mp4

avaa999, 2024-09-26:
Uploading 夹持错误_2.mp4…

avaa999, 2024-09-26:
[image]
[image]
I tried to use  the following codes to bind the last joint of the cable to the box:
<equality>
    
    <weld name="relpose" body1="slider" body2="B_last" relpose="0 0 0 1 0 0 0"/>
 </equality>
You can see that there is always a relative orientation difference between them during the translation of the box. Could you please give me some advice on how to avoid such a phenomenon? Thanks a lot!

yuvaltassa, 2024-09-26:
Duplicate of #1243. Save and load the XML then make the body a child of the last element.

closed 2024-09-26 by yuvaltassa

avaa999, 2024-09-27:
Thanks @yuvaltassa ! My problem has been solved. Thank-you very much for these informative and quick responses.

avaa999, 2024-11-22:
Hello! Here’s my current solution, but I’m afraid it still encounters the same issue you described. In my implementation, box3 and the static_box at the end of the rope are connected using a weld constraint, as shown below:
`
<mujoco model="Cable">
  <include file="scene.xml"/>

   
  <extension>
        <plugin plugin="mujoco.elasticity.cable">
        <instance name="composite">
            <config key="twist" value="1e5"/>
            <config key="bend" value="1e5"/>
            <config key="vmax" value="0"/>
        </instance>
        </plugin>
  </extension>    

  <custom>
        <text name="composite_" data="rope_"/>
  </custom>

  <statistic center="0 0 .5" extent="1"/>
  <visual>
    <global elevation="-30"/>
  </visual>

  <compiler autolimits="true"/>

  <size memory="2M"/>

  <worldbody>    
      <body name="slider" pos="0 0 0.1"> 
            <geom size=".01" rgba="0 0 1 1"/>
            <site type="box" pos="0 0 0" size=".012 .012 .012" rgba="1 0 0 1"/>
            <joint name="slide_x" type="slide" axis="1 0 0" damping="5"/>
            <joint name="slide_y" type="slide" axis="0 1 0" damping="5"/>
            <joint name="slide_z" type="slide" axis="0 0 1" damping="5"/>
            <body name="rope_" pos="0 0 0" euler="0 0 0">                                         
                <include file="/home/liuaohua/Documents/Shape_servoing/tactile_manipulation/envs/assets/tactile/MyCalble_box.xml"></include>
            </body> 
      </body>  
      <body name="box3" pos="0.3 0 0.1">
        <geom type="box" size="0.01 0.01 0.01" rgba=".5 .7 .5 .3"/>
        <joint name="slide_x2" type="slide" axis="1 0 0" damping="5"/>
        <joint name="slide_y2" type="slide" axis="0 1 0" damping="5"/>
        <joint name="slide_z2" type="slide" axis="0 0 1" damping="5"/>
      </body>
  </worldbody>

  <contact>
        <exclude body1="B_first" body2="B_1"/>
        <exclude body1="B_1" body2="B_2"/>
        <exclude body1="B_2" body2="B_3"/>
        <exclude body1="B_3" body2="B_4"/>
        <exclude body1="B_4" body2="B_5"/>
        <exclude body1="B_5" body2="B_6"/>
        <exclude body1="B_6" body2="B_7"/>
        <exclude body1="B_7" body2="B_8"/>
        <exclude body1="B_8" body2="B_9"/>
        <exclude body1="B_9" body2="B_10"/>
        <exclude body1="B_10" body2="B_11"/>
        <exclude body1="B_11" body2="B_12"/>
        <exclude body1="B_12" body2="B_13"/>
        <exclude body1="B_13" body2="B_14"/>
        <exclude body1="B_14" body2="B_15"/>
        <exclude body1="B_15" body2="B_16"/>
        <exclude body1="B_16" body2="B_17"/>
        <exclude body1="B_17" body2="B_18"/>
        <exclude body1="B_18" body2="B_19"/>
        <exclude body1="B_19" body2="B_20"/>
        <exclude body1="B_20" body2="B_21"/>
[... 253 lines cut ...]
                                        </body>
                                      </body>
                                    </body>
                                  </body>
                                </body>
                              </body>
                            </body>
                          </body>
                        </body>
                      </body>
                    </body>
                  </body>
                </body>
              </body>
            </body>
          </body>
        </body>
      </body>
    </body>
  </body>
</mujoco>
`
However, when I try to actively move the right end of the rope to change its shape, these two objects fail to remain relatively stationary. I’m currently considering using a robotic gripper to directly hold box3 to add additional constraints. Unfortunately, I haven’t been able to resolve the issue completely yet.

If you’ve made any progress or found a solution, I’d greatly appreciate it if you could share it! I hope my attempt can also provide you with some inspiration. Thank you!

avaa999, 2024-11-22:
Hello! Here’s my current solution, but I’m afraid it still encounters the same issue you described.
In my implementation, box3 and the static_box at the end of the rope are connected using a weld constraint, as shown below:
`
<mujoco model="Cable">
  <include file="scene.xml"/>

   
  <extension>
        <plugin plugin="mujoco.elasticity.cable">
        <instance name="composite">
            <config key="twist" value="1e5"/>
            <config key="bend" value="1e5"/>
            <config key="vmax" value="0"/>
        </instance>
        </plugin>
  </extension>    

  <custom>
        <text name="composite_" data="rope_"/>
  </custom>

  <statistic center="0 0 .5" extent="1"/>
  <visual>
    <global elevation="-30"/>
  </visual>

  <compiler autolimits="true"/>

  <size memory="2M"/>

  <worldbody>    
      <body name="slider" pos="0 0 0.1"> 
            <geom size=".01" rgba="0 0 1 1"/>
            <site type="box" pos="0 0 0" size=".012 .012 .012" rgba="1 0 0 1"/>
            <joint name="slide_x" type="slide" axis="1 0 0" damping="5"/>
            <joint name="slide_y" type="slide" axis="0 1 0" damping="5"/>
            <joint name="slide_z" type="slide" axis="0 0 1" damping="5"/>
            <body name="rope_" pos="0 0 0" euler="0 0 0">                                         
                <include file="/home/liuaohua/Documents/Shape_servoing/tactile_manipulation/envs/assets/tactile/MyCalble_box.xml"></include>
            </body> 
      </body>  
      <body name="box3" pos="0.3 0 0.1">
        <geom type="box" size="0.01 0.01 0.01" rgba=".5 .7 .5 .3"/>
        <joint name="slide_x2" type="slide" axis="1 0 0" damping="5"/>
        <joint name="slide_y2" type="slide" axis="0 1 0" damping="5"/>
        <joint name="slide_z2" type="slide" axis="0 0 1" damping="5"/>
      </body>
  </worldbody>

  <contact>
        <exclude body1="B_first" body2="B_1"/>
        <exclude body1="B_1" body2="B_2"/>
        <exclude body1="B_2" body2="B_3"/>
        <exclude body1="B_3" body2="B_4"/>
        <exclude body1="B_4" body2="B_5"/>
        <exclude body1="B_5" body2="B_6"/>
        <exclude body1="B_6" body2="B_7"/>
        <exclude body1="B_7" body2="B_8"/>
        <exclude body1="B_8" body2="B_9"/>
        <exclude body1="B_9" body2="B_10"/>
        <exclude body1="B_10" body2="B_11"/>
        <exclude body1="B_11" body2="B_12"/>
        <exclude body1="B_12" body2="B_13"/>
        <exclude body1="B_13" body2="B_14"/>
        <exclude body1="B_14" body2="B_15"/>
        <exclude body1="B_15" body2="B_16"/>
        <exclude body1="B_16" body2="B_17"/>
        <exclude body1="B_17" body2="B_18"/>
        <exclude body1="B_18" body2="B_19"/>
        <exclude body1="B_19" body2="B_20"/>
        <exclude body1="B_20" body2="B_21"/>
        <exclude body1="B_21" body2="B_22"/>
        <exclude body1="B_22" body2="B_23"/>
        <exclude body1="B_23" body2="B_24"/>
        <exclude body1="B_24" body2="B_25"/>
        <exclude body1="B_25" body2="B_26"/>
        <exclude body1="B_26" body2="B_27"/>
        <exclude body1="B_27" body2="B_28"/>
        <exclude body1="B_28" body2="B_29"/>
        <exclude body1="B_29" body2="B_30"/>
        <exclude body1="B_30" body2="B_31"/>
        <exclude body1="B_31" body2="B_32"/>
        <exclude body1="B_32" body2="B_33"/>
        <exclude body1="B_33" body2="B_34"/>
        <exclude body1="B_34" body2="B_35"/>
        <exclude body1="B_35" body2="B_36"/>
        <exclude body1="B_36" body2="B_37"/>
        <exclude body1="B_37" body2="B_38"/>
        <exclude body1="B_38" body2="B_39"/>
        <exclude body1="B_39" body2="B_40"/>
        <exclude body1="B_40" body2="B_41"/>
        <exclude body1="B_41" body2="B_42"/>
        <exclude body1="B_42" body2="B_last"/>
  </contact>
  <equality>
    
    <weld name="relpose"   body1="box3" body2="static_box" relpose="0 0 0 1 0 0 0" />

  </equality>

  <actuator> 

      <position name="control_x" joint="slide_x" ctrlrange="-1 1" kp="100"/>
      <position name="control_y" joint="slide_y" ctrlrange="-1 1" kp="100"/>
      <position name="control_z" joint="slide_z" ctrlrange="-1 1" kp="100"/>

      <position name="control_x2" joint="slide_x2" ctrlrange="-1 1" kp="100"/>
      <position name="control_y2" joint="slide_y2" ctrlrange="-1 1" kp="100"/>
      <position name="control_z2" joint="slide_z2" ctrlrange="-1 1" kp="100"/>
  </actuator>

</mujoco>
 
`

avaa999, 2024-11-22:
@louisdehaybe **Hello! Here’s my current solution, but I’m afraid it still encounters the same issue you described. In my implementation, box3 and the static_box at the end of the rope are connected using a weld constraint, as shown below:**
### main code
```
<mujoco model="Cable">
  <include file="scene.xml"/>

   
  <extension>
        <plugin plugin="mujoco.elasticity.cable">
        <instance name="composite">
            <config key="twist" value="1e5"/>
            <config key="bend" value="1e5"/>
            <config key="vmax" value="0"/>
        </instance>
        </plugin>
  </extension>    

  <custom>
        <text name="composite_" data="rope_"/>
  </custom>

  <statistic center="0 0 .5" extent="1"/>
  <visual>
    <global elevation="-30"/>
  </visual>

  <compiler autolimits="true"/>

  <size memory="2M"/>

  <worldbody>    
      <body name="slider" pos="0 0 0.1"> 
            <geom size=".01" rgba="0 0 1 1"/>
            <site type="box" pos="0 0 0" size=".012 .012 .012" rgba="1 0 0 1"/>
            <joint name="slide_x" type="slide" axis="1 0 0" damping="5"/>
            <joint name="slide_y" type="slide" axis="0 1 0" damping="5"/>
            <joint name="slide_z" type="slide" axis="0 0 1" damping="5"/>
            <body name="rope_" pos="0 0 0" euler="0 0 0">                                         
                <include file="/home/liuaohua/Documents/Shape_servoing/tactile_manipulation/envs/assets/tactile/MyCalble_box.xml"></include>
            </body> 
      </body>  
      <body name="box3" pos="0.3 0 0.1">
        <geom type="box" size="0.01 0.01 0.01" rgba=".5 .7 .5 .3"/>
        <joint name="slide_x2" type="slide" axis="1 0 0" damping="5"/>
        <joint name="slide_y2" type="slide" axis="0 1 0" damping="5"/>
        <joint name="slide_z2" type="slide" axis="0 0 1" damping="5"/>
      </body>
  </worldbody>

  <contact>
        <exclude body1="B_first" body2="B_1"/>
        <exclude body1="B_1" body2="B_2"/>
        <exclude body1="B_2" body2="B_3"/>
        <exclude body1="B_3" body2="B_4"/>
        <exclude body1="B_4" body2="B_5"/>
        <exclude body1="B_5" body2="B_6"/>
        <exclude body1="B_6" body2="B_7"/>
        <exclude body1="B_7" body2="B_8"/>
        <exclude body1="B_8" body2="B_9"/>
        <exclude body1="B_9" body2="B_10"/>
        <exclude body1="B_10" body2="B_11"/>
        <exclude body1="B_11" body2="B_12"/>
        <exclude body1="B_12" body2="B_13"/>
        <exclude body1="B_13" body2="B_14"/>
        <exclude body1="B_14" body2="B_15"/>
        <exclude body1="B_15" body2="B_16"/>
        <exclude body1="B_16" body2="B_17"/>
        <exclude body1="B_17" body2="B_18"/>
        <exclude body1="B_18" body2="B_19"/>
        <exclude body1="B_19" body2="B_20"/>
[... 253 lines cut ...]
                                            </body>
                                          </body>
                                        </body>
                                      </body>
                                    </body>
                                  </body>
                                </body>
                              </body>
                            </body>
                          </body>
                        </body>
                      </body>
                    </body>
                  </body>
                </body>
              </body>
            </body>
          </body>
        </body>
      </body>
    </body>
  </body>
</mujoco>
```
However, when I try to actively move the right end of the rope to change its shape, these two objects fail to remain relatively stationary. I’m currently considering **using a robotic gripper to directly hold box3** to add additional constraints. Unfortunately, I haven’t been able to resolve the issue completely yet. If you’ve made any progress or found a solution, I’d greatly appreciate it if you could share it! I hope my attempt can also provide you with some inspiration. Thank you!

avaa999, 2024-11-22:
@louisdehaybe  Hello! Here’s my current solution, but I’m afraid it still encounters the same issue you described. In my implementation, box3 and the static_box at the end of the rope are connected using a weld constraint, as shown below:
<details>
<summary>My scene including cable</summary>

```xml
<mujoco model="Cable">
  <include file="scene.xml"/>

  <extension>
    <plugin plugin="mujoco.elasticity.cable">
      <instance name="composite">
        <config key="twist" value="1e5"/>
        <config key="bend" value="1e5"/>
        <config key="vmax" value="0"/>
      </instance>
    </plugin>
  </extension>

  <custom>
    <text name="composite_" data="rope_"/>
  </custom>

  <statistic center="0 0 .5" extent="1"/>
  <visual>
    <global elevation="-30"/>
  </visual>

  <compiler autolimits="true"/>

  <size memory="2M"/>

  <worldbody>    
      <body name="slider" pos="0 0 0.1"> 
            <geom size=".01" rgba="0 0 1 1"/>
            <site type="box" pos="0 0 0" size=".012 .012 .012" rgba="1 0 0 1"/>
            <joint name="slide_x" type="slide" axis="1 0 0" damping="5"/>
            <joint name="slide_y" type="slide" axis="0 1 0" damping="5"/>
            <joint name="slide_z" type="slide" axis="0 0 1" damping="5"/>
            <body name="rope_" pos="0 0 0" euler="0 0 0">                                         
                <include file="/home/Documents/manipulation/envs/assets/MyCalble_box.xml"></include>
            </body> 
      </body>  
      <body name="box3" pos="0.3 0 0.1">
        <geom type="box" size="0.01 0.01 0.01" rgba=".5 .7 .5 .3"/>
        <joint name="slide_x2" type="slide" axis="1 0 0" damping="5"/>
        <joint name="slide_y2" type="slide" axis="0 1 0" damping="5"/>
        <joint name="slide_z2" type="slide" axis="0 0 1" damping="5"/>
      </body>
  </worldbody>

  <contact>
    <exclude body1="B_first" body2="B_1"/>
    <!-- more contact exclusions -->
    <exclude body1="B_42" body2="B_last"/>
  </contact>
  <equality>
    <weld name="relpose"   body1="box3" body2="static_box" relpose="0 0 0 1 0 0 0" />
  </equality>

  <actuator> 
    <position name="control_x" joint="slide_x" ctrlrange="-1 1" kp="100"/>
    <position name="control_y" joint="slide_y" ctrlrange="-1 1" kp="100"/>
    <position name="control_z" joint="slide_z" ctrlrange="-1 1" kp="100"/>
    <position name="control_x2" joint="slide_x2" ctrlrange="-1 1" kp="100"/>
    <position name="control_y2" joint="slide_y2" ctrlrange="-1 1" kp="100"/>
    <position name="control_z2" joint="slide_z2" ctrlrange="-1 1" kp="100"/>
  </actuator>

</mujoco>
```
[... 215 lines cut ...]
                                      </body>
                                    </body>
                                  </body>
                                </body>
                              </body>
                            </body>
                          </body>
                        </body>
                      </body>
                    </body>
                  </body>
                </body>
              </body>
            </body>
          </body>
        </body>
      </body>
    </body>
  </body>
</mujoco>
```
</details>
However, when I try to actively move the right end of the rope to change its shape, these two objects fail to remain relatively stationary. I’m currently considering using **a robotic gripper to directly hold box3 to add additional constraints**. Unfortunately, I haven’t been able to resolve the issue completely yet.

If you’ve made any progress or found a solution, I’d greatly appreciate it if you could share it! I hope my attempt can also provide you with some inspiration. Thank you!

louisdehaybe, 2024-11-30:
Hello @avaa999, I have found some workaround, you can give me some contact details if you wish to discuss this

avaa999, 2024-11-30:
> Hello @avaa999, I have found some workaround, you can give me some contact details if you wish to discuss this

Hello @louisdehaybe , I’m glad to hear that you’ve found a workaround! Here is my contact detail: liu_aohua@163.com.