google-deepmind/mujoco #2099: `island`-flag affects physics despite documentation stating otherwise

hartikainen, 2024-09-27:
Toggling the [`island`-flag](https://mujoco.readthedocs.io/en/stable/XMLreference.html#option-flag-island) affects this behavior, despite the documentation explicitly stating that the islands should have no effect on the physics:
> The flag currently has no effect on the physics pipeline, but enabling it allows for [island visualization](https://youtu.be/Vc1tq0fFvQA).

_Originally posted by @hartikainen in https://github.com/google-deepmind/mujoco/discussions/2098#discussioncomment-10774316_

yuvaltassa, 2024-09-27:
The documentation is wrong, but in a very very subtle way. I'll shortly clarify a bit more in the docs but let me expand here first.

In one sense this is true, the constraint solver (CG, until we add island support to Newton and PGS -- coming soon!) is solving exactly the same convex optimization problem, so in that sense yes, "the physics is the same".

However, "the same physics " is only true if the optimization problem is **actually solved**. If you do reach the minimum, the minimum is the same whether islands are on or off. However if you terminate early by reducing `iterations`, you don't actually reach the minimum and the solution will be different if islands are on or off. This is particularly egregious with CG, which—until the minimum is reached—will mix accelerations from unrelated dofs. By turning islands on, you are preventing this mixing since the separate per-island CG solvers can't interfere with each other. In this sense islands should always be turned on for CG.

I'd love to tell you to do that if you use CG you should always turn them on, but unfortunately I can't, since we need to refactor the island code. They way it currently works is not memory-local and adds some indirection, so even though it's supposed to be a strict improvement is sometimes isn't. (tl;dr we need to sort the constraints and the dofs so that each island is contiguous in memory).

I recommend turning on simulate's profiler (F3), to see this in action.

In terms of upcoming featureset for MuJoCo C, the order is roughly:
1. Fix the over-allocation in the Newton solver to allow it to handle very large models.
2. Refactor islands to be memory-local.
3. Add island support to the Newton solver.
4. Add island support to the PGS solver.
5. Allow multithreading over islands.
6. Allow sleeping islands.

This work package was delayed by about a year but is now finally making progress, you'll be happy to hear @thowell is very close to completing 1, which is a longstanding thorn in our side :slightly_smiling_face: 

cc @bd-mdelasa

hartikainen, 2024-09-28:
Thanks for the super detailed answer. Guaranteed Yuval quality hehe. Much appreciated!

bd-mdelasa, 2024-09-30:
@yuvaltassa thanks for the detailed answer. This info and the breakdown of next steps are very useful.

yuvaltassa, 2025-11-07:
sleeping is now at HEAD: https://mujoco.readthedocs.io/en/latest/computation/index.html#sleeping

So the only unfinished items in the list above are 4 and 5. We will probably do 5 first (threading), then 4 (PGS support)