google-deepmind/mujoco #2110: Unreliable/unphysical contact forces for box geom type versus mesh geom type

kurtenkera, 2024-10-01:
### Intro

Hi!

I am an Australian PhD researcher at the University of Queensland, and I use MuJoCo for my robotics research.

### My setup

I currently use MuJoCo via the Python bindings.

### What's happening? What did you expect?

I have developed a simple model of a robotic gripper that has two rigid fingers for gripping. When I use the `box` geom type to represent both fingers, I get a sensible simulation (i.e., normal contact forces and 'smooth' gripping of the test object) as seen below.

[boxgeoms.webm](https://github.com/user-attachments/assets/8ae1a461-2da6-40aa-9ac4-891bc25b8991)

However, when I use the `mesh` geom type to represent both fingers, the resulting contact forces on the fingers are unreasonable, and my gripper fails to grasp the test object as seen below. 

[meshgeoms.webm](https://github.com/user-attachments/assets/7856ea9f-b6e8-46b5-91e2-468f53116cff)

Why is there this disparity? I want to understand this disparity more deeply so I can accurately model contact forces on more exotic meshes of 3D solids. I would expect the two simulations to be identical, since I crafted my mesh to match the dimensions of the `box` used in the first simulation?

### Steps for reproduction

1. Load the model `boxgeom.xml` contained in the zip file below in the MuJoCo simulation GUI, and max out the `close_gripper` control value followed by the `lift_gripper` control as seen in the videos above. This model uses the `box` geom type for both the rigid fingers.
2. Load the model `meshgeom.xml` contained in the zip file below in the MuJoCo simulation GUI, and repeat the above steps. In this model, an STL file titled `finger.stl` is used to model both rigid fingers. The STL file is a simple rectangular prism with identical dimensions to the `box` geom type used in `boxgeom.xml`. The STL file was produced using [cadquery](https://cadquery.readthedocs.io/en/latest/).

### Minimal model for reproduction

In the following zip file, I have included an STL file of my rigid finger being modelled (the same finger is used for the left and right fingers), and two relevant minimal XML files `boxgeom.xml` and `meshgeom.xml`: 
[rigidgripper.zip](https://github.com/user-attachments/files/17199757/rigidgripper.zip).

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

nimrod-gileadi, 2024-10-07:
MuJoCo uses different collision functions for different kinds of geoms.
One major difference between the box-box collision function and the mesh-mesh collision function is that by default, the mesh-mesh collider returns only one contact point. When the meshes share faces that lay flat on top of each other, that contact point can jump around as you see here.

One fix for this issue is to use [multiccd](https://mujoco.readthedocs.io/en/stable/XMLreference.html#option-flag-multiccd) in your model to let the mesh-mesh collider output multiple points per contact:

```
<option multiccd="enable"/>
```

Note, this will slow down collision detection.

You may also want to try `nativeccd`, which is a rewrite of the collision detection code that may give better results.

yuvaltassa, 2024-10-08:
What Nimrod said but also, you ticked a box that said you searched the documentation, but did you... really? 😉

https://mujoco.readthedocs.io/en/stable/modeling.html#preventing-slip

closed 2024-10-08 by yuvaltassa