google-deepmind/mujoco #2113: Flex element gives `range[0] should be smaller than range[1] in joint`

ajoardar007, 2024-10-02:
### Intro

Hi!

I am a graduate student at the University of Waterloo, and I am using MuJoCo for validating my control-scripts before running them on the actual robot.

### My setup

Based upon the requirements (by running [this](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/python/tutorial.ipynb#scrollTo=Xqo7pyX-n72M&line=41&uniqifier=1) cell while displaying all outputs):

1. MuJoCo Version: 3.2.3,
2. Python: 3.8.10 64-bit,
3. `absl-py`: 2.1.0,
4. `etils`: 1.3.0 (cannot get it to upgrade it any further),
5. `numpy`: 1.24.4

Environment: Python,
Architecture: x86_64,
OS: Ubuntu 20.04.6 LTS

### What's happening? What did you expect?

Objective: Place flex-material pads on the fingers of the end-effector.
Result: 
```
/bin/python3 /home/.../src/flexible_bodies.py                                                                  ──(Wed,Oct02)─┘
Traceback (most recent call last):
  File "/home/.../src/flexible_bodies.py", line 189, in <module>
    model = mujoco.MjModel.from_xml_string(xml)
ValueError: Error: range[0] should be smaller than range[1] in joint
Element name '', id 154, line 84
```
Expected:
```
/bin/python3 /home/.../src/flexible_bodies.py                                                              1 ↵ ──(Wed,Oct02)─┘
"Invalid name ''. Valid names: ['cylinder_0/col', 'cylinder_0/viz', 'ground', 'smt/col/base_link/box_body', 'smt/col/base_link/cylinder_lidar', 'smt/col/whl/LF_cylinder', 'smt/col/whl/LR_cylinder', 'smt/col/whl/RF_cylinder', 'smt/col/whl/RR_cylinder', 'smt/viz/base_link', 'smt/viz/whl/LF', 'smt/viz/whl/LR', 'smt/viz/whl/RF', 'smt/viz/whl/RR']"
Hey!
<IPython.core.display.HTML object>
```
Observation: Element `softbody_1` causes no issues in its implementation (I want it to be attached to its parent-body). However, `softbody_2` is causing errors, even-though it has been placed significantly far-away to prevent it from initially colliding with any other body.

### Steps for reproduction

1. Run the code below in python (run it on any cell after running cell-1 in the [tutorial-notebook](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/python/tutorial.ipynb)).
2. You will get:
```
/bin/python3 /home/.../src/flexible_bodies.py                                                                  ──(Wed,Oct02)─┘
Traceback (most recent call last):
  File "/home/.../src/flexible_bodies.py", line 189, in <module>
    model = mujoco.MjModel.from_xml_string(xml)
ValueError: Error: range[0] should be smaller than range[1] in joint
Element name '', id 154, line 84
```
3. Comment out `softbody_2` element (`flexcomp` element starting at line 173 to 181), and run again.
4. You will get:
```
/bin/python3 /home/.../src/flexible_bodies.py                                                              1 ↵ ──(Wed,Oct02)─┘
"Invalid name ''. Valid names: ['cylinder_0/col', 'cylinder_0/viz', 'ground', 'smt/col/base_link/box_body', 'smt/col/base_link/cylinder_lidar', 'smt/col/whl/LF_cylinder', 'smt/col/whl/LR_cylinder', 'smt/col/whl/RF_cylinder', 'smt/col/whl/RR_cylinder', 'smt/viz/base_link', 'smt/viz/whl/LF', 'smt/viz/whl/LR', 'smt/viz/whl/RF', 'smt/viz/whl/RR']"
Hey!
<IPython.core.display.HTML object>
```
(successful compilation)

### Minimal model for reproduction

No mesh-files needed.
The MJCF is included in the script for compactness.

### Code required for reproduction

[... 196 lines cut ...]
scene_option = mujoco.MjvOption()
scene_option.flags[mujoco.mjtVisFlag.mjVIS_JOINT] = True

duration = 3.8  # (seconds)
framerate = 60  # (Hz)

# Simulate and display video.
frames = []
mujoco.mj_resetData(model, data)
with mujoco.Renderer(model) as renderer:
  while data.time < duration:
    mujoco.mj_step(model, data)
    if len(frames) < data.time * framerate:
      renderer.update_scene(data, scene_option=scene_option)
      pixels = renderer.render()
      frames.append(pixels)

print("Hey!")
media.show_video(frames, fps=framerate)
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

quagla, 2024-10-03:
The reason is that the bodies in flexcomp inherit the `wam` defaults. Can you set the hinge joint directly in the body instead of using the default?

closed 2024-10-03 by quagla

ajoardar007, 2024-10-03:
Yup. That seemed to resolve the issue. I guess, even if a joint was not defined for the body (no range), the other attributes that were defined as default were absorbed.
Maybe, later down the line, a rule, similar to how joint-behavior is defined for `body`, could be ported to `flexcomp`; like how a `body` with no joint-range does not throw the same error as for `flexcomp`.