google-deepmind/mujoco #2114: `mjContact.vert` bugs for `flexcomp`-`geom` collisions

kurtenkera, 2024-10-03:
### Intro

Hi!

I am an Australian PhD researcher at the University of Queensland, and I use MuJoCo for my robotics research.

### My setup

MuJoCo version: 3.2.3
API: Python
Architecture: x86_64
OS: Ubuntu 22.04.1 LTS

### What's happening? What did you expect?

I'm trying to use MuJoCo's elasticity plugin to accurately simulate contact forces on soft, deformable robots. However, I've noticed that **`mjContact.vert` returns incorrect flex vertex ids** when simulating a simple `flexcomp` soft cube colliding with a `plane` geom.

Note that the cube that I have considered in my tests was meshed in [gmsh](https://gmsh.info/). The picture below shows the vertex ids of the meshed cube in GMSH. In particular, note that the base of the cube has vertex ids 1, 2, 3, 4 and 9. From personal experiments, I've noticed that MuJoCo uses the same vertex id labelling scheme as GMSH, however all ids are based from 0 in MuJoCo (i.e., each id in GMSH is subtracted by 1 for zero-indexing). 

[image]

**When I set the value of `euler` of my `flexcomp` cube to non-trivial values, bugs arise.**

### Steps for reproduction

1. Download the contents of [softbox.zip](https://github.com/user-attachments/files/17243207/softbox.zip).
2. In the same directory create a new empty Python file called `minimal.py` and copy and paste the Python code below into your script. **Make sure you adjust the `xml_filepath` variable in line 4 accordingly!**
3. Set `euler` in line 25 of `softbox.xml` to the following values: `"euler = 1 0 0"`, `"euler = 2 0 0"`, `"euler = 45 0 0"`, `"euler = 0 1 0"`. For each case, run `minimal.py`. In each case, the box is dropped from a height and ultimately lands on flex vertex ids 0, 1, 2, 3 and 8 (or 1, 2, 3, 4 and 9 are the GMSH ids as mentioned earlier). However, if you inspect the terminal output, **there are some flex vertex ids that are repeated, and we get the incorrect flex vertex ids involved in the contact between the `flexcomp` and `plane` in each case**.
4. Confusingly, if you uncomment the `if` condition in line 14 of `minimal.py` and repeat step 3 for all cases, **we get the correct output/flex vertex ids for all cases**. What is causing this!?
5. Repeat step 3 for the following values: `"euler = 90 0 0"`, `"euler = 90 90 0"`, `"euler = 90 0 90"` and `"euler = 90 0 180"`. You will get the correct flex vertex ids **regardless of whether the if statement in line 14 of `minimal.py` is commented out or not**. So it seems 90-degree rotations are handled correctly, but anything else produces erroneous bugs?

### Minimal model for reproduction

The following zip folder contains the mesh of my soft cube - `softbox.msh` - and a minimal MJCF file called `softbox.xml`: 
[softbox.zip](https://github.com/user-attachments/files/17243220/softbox.zip).

Please run the Python code below following the instructions above to reproduce the bugs.

### Code required for reproduction

```
import mujoco
import os

xml_filepath = "path/to/softbox.xml"
model = mujoco.MjModel.from_xml_path(xml_filepath)  
data = mujoco.MjData(model)
duration = 3.0 
n_steps = int(duration/model.opt.timestep) #timestep = 0.0005

for i in range(n_steps):

    mujoco.mj_step(model, data)

    # if i == n_steps - 1:
    print("t = ", data.time, "\n")
    for con_id, con in enumerate(data.contact):
        print("Contact #", con_id+1)
        print("con.vert = ", con.vert, "\n")
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

kurtenkera, 2024-10-25:
Hi @quagla and @nimrod-gileadi, were you able to reproduce this bug? I am eager to resolve this issue because in my work, I need to accurately retrieve: (a) **contact points i.e. (x,y,z) coordinates** in the local frame of a simulated _soft object_ and (b) **contact forces with correct direction and magnitude** in the local frame of soft objects. I've been able to achieve this with **rigid bodies** in MuJoCo. For example, the video below shows a falling rigid dodecahedron, and a corresponding animation of the contact forces rotated into the local frame of the rigid object.

[rigidbody_contacts.webm](https://github.com/user-attachments/assets/84881a0e-83b7-4eb1-a319-199615311da2)

With rigid bodies, one only needs access to something like `xmat` or `geom_xmat` to rotate both contact points `con.pos` and contact forces into the local frame of the rigid body. But because something like `flexcomp_xmat` doesn't exist, I cannot do either of these things with a simulated soft object in MuJoCo. 

As a result, I've attempted to use `mjContact.vert` to retrieve the vertex ids of nodes in my soft flexcomp objects (and map these ids to their correct (x,y,z) coordinates in the local frame of my soft object), but as mentioned above, this approach currently fails due to bugs. Is there a solution to this issue that I'm not seeing?

quagla, 2024-10-28:
Hi have you checked that when the `vert` are repeated, the `elem` are also repeated? In principle, I don't see anything wrong with repeated vertices since the collisions are element-wise.

quagla, 2024-10-28:
Also, you don't need a flexcomp_mat, because each vertex is a body, so you can retrieve his Cartesian position and orientation in `xpos` and `xquat` using `flex_vertbodyid` from mjModel.

kurtenkera, 2024-10-29:
1. I've noticed that `elem` for each `flex-geom` collision is `[-1,-1]` **regardless of whether the `vert` are repeated or not**. I believe this is a bug, as we're expecting a `-1` for geoms and the element id for flexes right? I haven't been able to print out element ids that are not `-1` for this simple soft box example for any of the above mentioned `euler` settings. Also, the [documentation](https://mujoco.readthedocs.io/en/stable/APIreference/APItypes.html#mjcontact) for the relevant `mjContact` attributes is currently the wrong way around:

```
  int     elem[2];                 // element ids; -1 for geom or flex vertex
  int     vert[2];                 // vertex ids;  -1 for geom or flex element
```

i.e. `flex vertex` and `flex element` should be swapped around for the relevant attributes.

2. Even though each vertex is a body and its position and orientation **relative to its local body frame** can be obtained via `xpos` and `xquat` or `xmat`, I've noticed that the orientation of each flex's Cartesian coordinate system/body frame doesn't rotate in the way that I need. This can be best understood by comparing a falling rigid geom box to a falling soft flexcomp box. 

Below, we see that for a falling rigid geom box, the body frame/coordinate system rotates with the rotating body as expected. 

[falling_rigidbox.webm](https://github.com/user-attachments/assets/1777cd09-a15e-419e-985a-9eb15158e497)

Hence, it is possible to rotate a contact force measured in the contact frame, `f_C`, into the body frame via `R_B.T @ R_C @ f_C`, where `R_B.T` is the transpose of the relevant `body_xmat` and `R_C = contact.frame.reshape(3,3).T` (this formula can be confirmed in #1720). Clearly, from the video above, there is a single rotating coordinate system associated with the entire box geometry.

[rigid_forcerotation.webm](https://github.com/user-attachments/assets/5c2d5047-c2e3-453c-b005-9707e98147f9)

Below in the soft case, each coordinate system/body frame associated with each flex vertex **always stays aligned with the world frame**, despite the non-trivial rotation that the entire flexcomp experiences. Hence, there is no way that I can rotate the forces experienced by the entire flexcomp into the still frame of the flexcomp, as I can in the rigid case (see above). In my case, I need to treat the entire flexcomp as a single "geometry", despite the fact that under-the-hood MuJoCo associates a single body with each flex vertex. 

[falling_softbox.webm](https://github.com/user-attachments/assets/e8066d1e-9648-46f4-b0d8-913137fb213a)

Why do the body frames associated with each flex vertex always stay aligned with the world frame despite non-trivial rotations? Is there a way I can change this unwanted behaviour?

quagla, 2024-10-29:
I will look into 1. because I haven't written the flex collision code myself. But it could be that it's a special vertex-plane case, which is why the elements are -1. The documentation seems correct however?

As for 2., this is correct. A flex is a collection of sliders, so the bodies at the vertices cannot rotate (at least not with respect to their parent, which is the world in your case), and rotations are achieved as a combination of vertex displacements. There is no unique way to define a flex rotation - unless the flex is set to be rigid - so you could e.g. extract a best-fit rotation from the vertex displacements. Once you have that frame, you can rotate the contact forces using it.

kurtenkera, 2024-10-30:
> The documentation seems correct however?

The current documentation is:

```
  int     elem[2];                 // element ids; -1 for geom or flex vertex
  int     vert[2];                 // vertex ids;  -1 for geom or flex element
```

But I'm saying I think it should be the following (?):

```
  int     elem[2];                 // element ids; -1 for geom or flex **element**
  int     vert[2];                 // vertex ids;  -1 for geom or flex **vertex**
```

> As for 2., ...

Thank you, this is excellent information and very helpful!

Finally, I just want to make sure I understand what you said here: 

> have you checked that when the vert are repeated, the elem are also repeated? In principle, I don't see anything wrong with repeated vertices since the collisions are element-wise.

In the case of the soft box example, are you saying that since collisions involving flexes are element-wise, repeated `vert` should be fine so long as the `elem` for repeated vertices are different? If the `elem` were the same for a repeated `vert`, this would suggest that a collision has been wrongly double-counted, whereas if we have a unique `elem` for a repeated `vert`, then this is fine?

yuvaltassa, 2024-10-30:
I'll let @quagla answer the second part, but can explain the comments, which are correct. They mean

```c
  int     elem[2];      // element ids (>=0) if the collision is between two elems; if one of the objects is a geom or a vertex, use -1
  int     vert[2];      // vertex ids (>=0) if the collision is between two verts; if one of the objects is a geom or an element, use -1
```
Does this make sense?

kurtenkera, 2024-10-30:
This is crystal clear and aligns with my observations (barring the case of `flex-plane` collisions mentioned above, where I continue to observe `elem = [-1,-1]` for different simulations). Thank you for the clarification!

kurtenkera, 2024-10-31:
I have successfully rotated the forces on a soft, deformable box into its rest frame using the method @quagla described (see below).

[softbox_rotatedforces.webm](https://github.com/user-attachments/assets/297ef759-24e8-46ba-81ee-8b83027effc5)

I have tried to do the same thing for the soft-gripper model that was discussed in #2084 i.e. I've tried to rotate the contact forces into the rest frame of the left-finger. However, **I get erroneous bugs** which are related to that outlined in step 4. of the **Steps for reproduction** in my first comment in this thread (see comment quote below):

> 4. Confusingly, if you uncomment the if condition in line 14 of minimal.py and repeat step 3 for all cases, we get the correct output/flex vertex ids for all cases. What is causing this!?

In the following zip folder (
[minimal_gripper.zip](https://github.com/user-attachments/files/17580838/minimal_gripper.zip)) I have included `soft_gripper.xml` and `minimal_gripper.py`. If you run `minimal_gripper.py` (code attached below), a 25 second simulation video will be saved. It is evident that from ~5 seconds onwards, the gripper is grasping the `testobject` i.e. a yellow box geom. However, when I run the code below, `if con.geom[0] == 3 and con.flex[1] == 0` never evaluates to True? Note that by inspection, I've confirmed that the `testobject` has a geom id of 3, and the flexcomp corresponding to the left finger has a flex id of 0. Hence, this if statement in `minimal_gripper.py` should always trigger for all times greater than ~5s (but it never does). 

But if I uncomment `# if data.time > 6.0 and data.time < 6.0005`, then `if con.geom[0] == 3 and con.flex[1] == 0` evaluates to True as desired. Again, what is causing this inconsistency? Clearly, we would expect that such a simulation should behave the same, whether I include if statements or not. But the inclusion of if statements is clearly impacting the attributes of `mjContact`. 

I have tried storing `con.geom` and `con.flex` in separate variables, e.g.:

```
geom_copy = deepcopy(con.geom)
flex_copy = deepcopy(con.flex)
if con.geom[0] == 3 and con.flex[1] == 0:
     ...
```

But this does not solve the issue. I've also observed that sometimes (very rarely) the code below can run correctly, with `if con.geom[0] == 3 and con.flex[1] == 0` evaluating to true for all times greater than ~5 seconds. This is very confusing behaviour, but very important for me to resolve. Obviously, I need to include these if statements to isolate contact events involving the left-gripper and test object only, as I need to ignore internal `flex-flex` contacts and all other contacts for my purposes. But doing so via if statements seems to alter the attributes of `mjContact` class instances in an incorrect manner.

I will continue to play around with this, but I am considering switching to the C API rather than the Python bindings to see if this will solve these issues. I appreciate any feedback in advance!

```
import mujoco
import os
from mediapy import VideoWriter
from copy import deepcopy

xml_filepath = os.path.join(os.getcwd(), "soft_gripper.xml")

model = mujoco.MjModel.from_xml_path(xml_filepath)  
data = mujoco.MjData(model)
duration = 25
n_steps = int(duration/model.opt.timestep) #timestep = 0.0005

framerate = 60
video_shape = (480, 640)

# visualize contact forces, make body transparent
scene_option = mujoco.MjvOption()
mujoco.mjv_defaultOption(scene_option)
scene_option.flags[mujoco.mjtVisFlag.mjVIS_CONTACTPOINT] = True
scene_option.flags[mujoco.mjtVisFlag.mjVIS_CONTACTFORCE] = True
scene_option.flags[mujoco.mjtVisFlag.mjVIS_TRANSPARENT] = True

# tweak scales of contact visualization elements
model.vis.scale.contactwidth = 0.2
model.vis.scale.contactheight = 0.2
model.vis.scale.forcewidth = 0.05
model.vis.map.force = 0.02 

# Simulate and save video
frames = []

with mujoco.Renderer(model, video_shape[0], video_shape[1]) as renderer:
  for i in range(n_steps):
      
    # Print simulation time after each second has passed
    if abs(data.time % 1.0) < model.opt.timestep:
        print(f"Simulation time: {data.time:.3f} seconds")

    # Bang-bang binary control scheme
    if data.time > 0:
        data.ctrl[0] = 0.5  # Closing the gripper instantly/always
    if data.time > 14.0: 
        data.ctrl[0] = 0.45 # Relax the grip strength as we lift to avoid penetrating platforms
        data.ctrl[1] = 1    # Lifting gripper

    mujoco.mj_step(model, data)

    # if data.time > 6.0 and data.time < 6.0005:
    for con_id, con in enumerate(data.contact):
        #geom_copy = deepcopy(con.geom)
        #flex_copy = deepcopy(con.flex)
        if con.geom[0] == 3 and con.flex[1] == 0:
            print("Contact #", con_id+1)
            print("con.geom1 = ", con.geom1)
            print("con.geom2 = ", con.geom2)
            print("con.elem = ", con.elem)
            print("con.flex = ", con.flex)
            print("con.geom = ", con.geom)
            print("con.vert = ", con.vert, "\n")

    if len(frames) < data.time * framerate:
        renderer.update_scene(data, "closeup", scene_option=scene_option)
        pixels = renderer.render()
        frames.append(pixels)

renderer.close()

output_file_path = os.path.join(os.getcwd(), "minimal_softgripsim.mp4")
with VideoWriter(output_file_path, video_shape, fps = framerate) as writer:
    for video_frame in frames:
        writer.add_image(video_frame)
```

kurtenkera, 2024-11-01:
Update: I am convinced this issue is related to the **bug** identified in #1710, which I understand is still unresolved and would occur whether I use the Python bindings or the C API. I can demonstrate that this is the case with the simple minimal example below:

```python
import mujoco
import os
from copy import copy
import numpy as np

model = mujoco.MjModel.from_xml_path("soft_gripper.xml")
data = mujoco.MjData(model)
duration = 7
n_steps = int(duration/model.opt.timestep) #timestep = 0.0005

for i in range(n_steps):

    # Bang-bang binary control scheme
    if data.time > 0:
        data.ctrl[0] = 0.5  # Closing the gripper instantly/always
    if data.time > 14.0: 
        data.ctrl[0] = 0.45 # Relax the grip strength as we lift to avoid penetrating platforms
        data.ctrl[1] = 1    # Lifting gripper

    mujoco.mj_step(model, data)

    print("  Original data:")
    print("  n_contacts: ", data.ncon, "\n")
    print("  colliding geoms: ", data.contact.geom, "\n")
    print("  colliding flexes: ", data.contact.flex, "\n")

    con_geom = []
    con_flex = []
    for con_id, con in enumerate(data.contact):
        con_geom.append(copy(con.geom))
        con_flex.append(copy(con.flex))

    print("  Data accessed during for-loop over list of mjContact data structures: ")
    print("  n_contacts: ", data.ncon, "\n")
    print("  colliding geoms: ", np.array(con_geom), "\n")
    print("  colliding flexes: ", np.array(con_flex), "\n")
```

If you run the above code, it will print out all contacts at all time steps for a total time of 7s. The output below sampled from a particular timestep clearly shows I am getting the wrong contact information during the for-loop over all `mjContact` instances:

```
  Original data:
  n_contacts:  24 

  colliding geoms:  [[ 0  3]
 [ 0  3]
 [ 0  3]
 [ 0  3]
 [ 3 -1]
 [ 3 -1]
 [ 3 -1]
 [ 3 -1]
 [ 3 -1]
 [ 3 -1]
 [ 3 -1]
 [ 3 -1]
 [ 3 -1]
 [ 3 -1]
 [ 3 -1]
 [ 3 -1]
 [-1 -1]
 [-1 -1]
 [-1 -1]
 [-1 -1]
 [-1 -1]
 [-1 -1]
 [-1 -1]
 [-1 -1]] 

  colliding flexes:  [[-1 -1]
 [-1 -1]
 [-1 -1]
 [-1 -1]
 [-1  0]
 [-1  0]
 [-1  0]
 [-1  0]
 [-1  0]
 [-1  0]
 [-1  1]
 [-1  1]
 [-1  1]
 [-1  1]
 [-1  1]
 [-1  1]
 [ 0  0]
 [ 0  0]
 [ 0  0]
 [ 0  0]
 [ 1  1]
 [ 1  1]
 [ 1  1]
 [ 1  1]] 

  Data accessed during for-loop over list of mjContact data structures: 
  n_contacts:  24 

  colliding geoms:  [[ 0.  3.]
 [ 0.  3.]
 [ 0.  3.]
 [ 0.  3.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]] 

  colliding flexes:  [[-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [-1. -1.]
 [ 0.  0.]
 [ 1.  1.]
 [ 0.  0.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 1.  1.]
 [ 0.  0.]
 [ 0.  0.]] 
```

kurtenkera, 2024-11-01:
**UPDATE**: I believe all the issues I have had with contact information being altered were fixed via #1710. I was accessing the Python bindings via version 3.1.4, but after updating to 3.2.4 my issues have been resolved!

closed 2024-11-02 by yuvaltassa