google-deepmind/mujoco #2115: Plugings segfault when built with mjUSESINGLE

PeterOGB, 2024-10-03:
### Intro

New mujoco user here, I'm a retired software engineer with 40 years C/unix/linux experience.

I'm trying to integrate mujoco into an existing application that runs on a Raspberry Pi5 and uses openGLES for rendering and gclm for maths.

### My setup

Raspberry Pi 5 running wayland on top of 64 bit PiOS (Debian 12).
cmake version 3.25.1
gcc (Debian 12.2.0-14) 12.2.0
MujoCo built from github with mjUSESINGLE defined and CCD built for single precision.

### What's happening? What did you expect?

Running simulate on model/plugin/elasticity/cable.xml produces a segfaut

Thread 3 "simulate" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe9562800 (LWP 47911)]
__memcpy_generic () at ../sysdeps/aarch64/multiarch/../memcpy.S:173
173	../sysdeps/aarch64/multiarch/../memcpy.S: No such file or directory.
(gdb) bt
#0  __memcpy_generic () at ../sysdeps/aarch64/multiarch/../memcpy.S:173
#1  0x00007ffff74d3ef4 in mujoco::plugin::elasticity::Cable::RegisterPlugin()::{lambda(mjModel_ const*, mjData_*, int)#2}::_FUN(mjModel_ const*, mjData_*, int) () from /usr/local/bin/mujoco_plugin/libelasticity.so
#2  0x00007ffff7d1ee24 in mj_initPlugin () from /usr/local/bin/../lib/libmujoco.so.3.2.3
#3  0x00007ffff7d25634 in mj_makeData () from /usr/local/bin/../lib/libmujoco.so.3.2.3
#4  0x00007ffff7db1ea8 in mjCModel::TryCompile(mjModel_*&, mjData_*&, mjVFS_ const*) ()
   from /usr/local/bin/../lib/libmujoco.so.3.2.3
#5  0x00007ffff7daead8 in mjCModel::Compile(mjVFS_ const*, mjModel_**) ()
   from /usr/local/bin/../lib/libmujoco.so.3.2.3
#6  0x00007ffff7df54b4 in mj_loadXML () from /usr/local/bin/../lib/libmujoco.so.3.2.3
#7  0x000055555555c798 in (anonymous namespace)::LoadModel(char const*, mujoco::Simulate&) ()
#8  0x0000555555561fc0 in PhysicsThread(mujoco::Simulate*, char const*) ()
#9  0x00007ffff792e9dc in ?? () from /lib/aarch64-linux-gnu/libstdc++.so.6
#10 0x00007ffff76dee58 in start_thread (arg=0x7fffffffecb7) at ./nptl/pthread_create.c:442
#11 0x00007ffff7747f9c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:79

### Steps for reproduction

Any demo with a plugin segfaults.

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2024-10-03:
I think none of the plugins were written with float32 in mind. Do you actually need them or just want to have a fully working build?

PeterOGB, 2024-10-03:
I would like to use mujoco.elasticity.cable to model a strip of paper tape as it is read by a vintage paper tape reader.  So yes I have a good reason for at least that one to work.

PeterOGB, 2024-10-03:
This is actually caused by the install process not updating the plugins in the installation directory. In fact it doesn't seem to install them at all !  At some point the "double" versions were put into /usr/local/bin/mujoco_plugins and they were not being replaced by the "float" versions I was actually building correctly.
Again I've ended up wasting several hours due to the mujoco build system not doing what you might reasonably expect it to do:-(

closed 2024-10-03 by PeterOGB

traversaro, 2024-10-03:
> This is actually caused by the install process not updating the plugins in the installation directory. In fact it doesn't seem to install them at all !  At some point the "double" versions were put [...]

Related PR: https://github.com/google-deepmind/mujoco/pull/1515 .