google-deepmind/mujoco #2126: MJX collision between Sphere-Ellipsoid

kinalmehta, 2024-10-08:
### Intro

Hi!

I am trying to use MuJoCo-MJX with CMUHumanoid from [dm-control](https://github.com/google-deepmind/dm_control/tree/main/dm_control/locomotion). However I'm getting the following error.

```sh
NotImplementedError: (mjtGeom.mjGEOM_SPHERE, mjtGeom.mjGEOM_ELLIPSOID) collisions not implemented.
```

However in the documentation it is mentioned that `ELLIPSOID and CYLINDER are implemented but only collide with other primitives, note that BOX is implemented as a mesh.`
I assume sphere is a primitive, so this error is unexpected.

### My setup

### Mujoco versions
- mujoco                       3.2.3
- mujoco-mjx                   3.2.3

### Python and OS version
- python 3.11.8
- ubuntu 22.04

### What's happening? What did you expect?

Setting `rgba="1 0 0 1"` renders as **green**, I expected it to render as **red**.

Here is a screen-shot showing a green sphere:

### Steps for reproduction

1. Load the model below.
2. Run the code below.
3. See green sphere (should be red).

### Minimal model for reproduction

If you encountered the issue in a complex model, please simplify it as much as possible (while still reproducing the issue).

minimal XML link: https://github.com/google-deepmind/dm_control/blob/main/dm_control/suite/humanoid_CMU.xml

### Code required for reproduction

```python
import mujoco
from mujoco import mjx

mj_model = mujoco.MjModel.from_xml_path(xml_path)
mj_data = mujoco.MjData(mj_model)
print(mj_data.qpos, type(mj_data.qpos))

mjx_model = mjx.put_model(mj_model) # error here
mjx_data = mjx.put_data(mj_model, mj_data)

print(mjx_data.qpos, type(mjx_data.qpos), mjx_data.qpos.devices())
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

kinalmehta, 2024-10-08:
I tried adding the following code in [collision_sdf.py](https://github.com/google-deepmind/mujoco/blob/main/mjx/mujoco/mjx/_src/collision_sdf.py)

```python
@collider(ncon=1)
def sphere_ellipsoid(s: GeomInfo, e: GeomInfo) -> Collision:
  """"Calculates contact between a sphere and an ellipsoid."""
  x0 = 0.5 * (s.pos + e.pos)
  return _optim(_sphere, _ellipsoid, s, e, x0)

```

Not sure if this works, I saw that there is an implementation of `_sphere` and `_ellipsoid` and wanted some feedback on this.

Thanks
Kinal

btaba, 2024-10-08:
@kinalmehta let us know if the recent push fixes the issue, thanks for reporting this!

closed 2024-10-08 by btaba

kinalmehta, 2024-10-09:
Thanks for the quick fix.