google-deepmind/mujoco #2128: Simulate Dependencies Deprecated on Latest macOS Sequoia (15.0+)

dtorre38, 2024-10-09:
### Intro

Hi!

I am a graduate student at the University of Illinois at Chicago, I use MuJoCo for my research on motion planning.

### My setup

Tested on 3.2.3 & 3.1.1
M2 Pro MacBook Pro
MacOS 15.0.1
C API

### What's happening? What did you expect?

I am making modifications to simulate, so I need to build via CMakeLists.txt. When I make, I obtain the errors shown in the screenshot. As the errors indicate, some functionality has been deprecated on the latest MacOS version. Can we expect an update to correct this issue for Mac users on 15.0+?

[image]

### Steps for reproduction

1. cd simulate
2. mkdir build && cd build
3. cmake ..
4. make

### Minimal model for reproduction

Utilize mujoco-3.2.3.

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2024-10-31:
@dtorre38 the error is telling you how to fix this, would you like to send a PR?

dtorre38, 2024-11-05:
@yuvaltassa Apologies, I won't be able to get to this soon. I reverted to an older version of mujoco (2.2.1) for the time being.

yuvaltassa, 2024-11-14:
Okay, but you do realize that this is only because we are telling the compiler to treat warnings as errors?

You can just delete `-Werror` and it will build.

yuvaltassa, 2024-11-14:
Delete [this line](https://github.com/google-deepmind/mujoco/blob/0e8cca93e185c0ca7dfdd15b52db591c591a9c14/cmake/MujocoOptions.cmake#L89)