google-deepmind/mujoco #2136: Not displaying flexcomp objects in scene nor raising errorr

Nick20500, 2024-10-11:
### Intro

Hi!

I am a undergrad student at ETH Zurich, I use MuJoCo for my research on Multimaterial deformable objects.

### My setup

I use MuJoCo 3.2.3 with python bindings.

### What's happening? What did you expect?

I am currently trying to run my simulation, which I ran successfully with mujoco, now with mjx.
The python script in the attached zip folder
[zip folder mjx_bug_files.zip](https://github.com/user-attachments/files/17347477/mjx_bug_files.zip)
 will run the simulation for 5 seconds and produce frames for a video of 24 fps. In the end, all the generated frames get stitched together to a mp4 file.

I expect the output to be a red bottle and a green sphere droping from the heigth of 5 units. (e.g. in the picture generated from running the simulation only with mujoco)
[image]

What I get, is only the green sphere (e.g. in the picture)
[image]

The only reasonable expleantion for me, is that flexcomps aren't yet supported in mjx. But if this is true, then there should be an error raised.

### Steps for reproduction

1. Download the zip folder
2. unzip the downloaded zip folder and open the folder
3. run the python script "simulate_object_drop_mjx_parallel.py"
3b. if necessary, download the used librarys with "pip install *"
4. see my results

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

quagla, 2024-10-14:
The second error will be fixed soon and it's the same as #2135 hopefully the error message will tell you what the issue is.

Nick20500, 2024-10-14:
> The first error is due to the fact that as you correctly anticipated, flex is not available in MJX.

@quagla Will flex be available in MJX in the near future?

quagla, 2024-10-14:
It's not a priority at the moment I'm afraid.

closed 2024-10-15 by quagla