google-deepmind/mujoco #2153: NotImplementedError: Elliptic cone force is not implemented yet.

kinalmehta, 2024-10-18:
### Intro

Hi!

I am a ([graduate / undergrad] student / professor / researcher) at XYZ, I use MuJoCo for my research on ABC.

### My setup

ase

### What's happening? What did you expect?

Setting `rgba="1 0 0 1"` renders as **green**, I expected it to render as **red**.

Here is a screen-shot showing a green sphere:

### Steps for reproduction

1. Load the model below.
2. Run the code below.
3. See green sphere (should be red).

### Minimal model for reproduction

If you encountered the issue in a complex model, please simplify it as much as possible (while still reproducing the issue).

<details>
<summary>minimal XML</summary>

```XML

<mujoco>
  <worldbody>
    <light pos="0 0 1"/>
    <geom type="sphere" size="1" rgba="1 0 0 1"/>
  </worldbody>
</mujoco>

```
</details>

### Code required for reproduction

```python
import mujoco
import mediapy as media
model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)

with mujoco.Renderer(model) as renderer:
  mujoco.mj_forward(model, data)
  renderer.update_scene(data)

  media.show_image(renderer.render())
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

uvikas, 2024-10-28:
Hi @kinalmehta, I am getting the same error for a different XML model. Were you able to resolve this?

kinalmehta, 2024-10-28:
I switched to using `pyramidal` cone till this is fixed
```xml
<option cone="pyramidal"/>
```

commit 55464106 (2024-10-31): Add support for computing elliptic cone contact forces to MJX. Fixes #2153.

PiperOrigin-RevId: 691829630
Change-Id: I862ac9c7db07d0a74b718d2a41e19d5ae636a260
closed 2024-10-31 by copybara-service[bot]