google-deepmind/mujoco #2155: MJX ray has no implementation for hfield geoms

nico-bohlinger, 2024-10-18:
### The feature, motivation and pitch

The `mjx.ray` function is great to get depth images in MJX and it is used in the recently added rangefinder, but the handling of `hfield` geoms is currently not supported. To add this, one would need to port the [mj_rayHfield](https://github.com/google-deepmind/mujoco/blob/dcf2816543dd6b1fda14faa2b6d242c3b9c60f7f/src/engine/engine_ray.c#L453) function to `mjx`.

This feature would be highly appreciated to enable depth rendering in hfield environments / terrains for learning robot locomotion.

### Alternatives

``

btaba, 2024-10-24:
Hi @nico-bohlinger thanks for the heads up on the missing feature. Would you like to implement it?

shreyans413, 2025-02-28:
Hi, can i work on this issue?