google-deepmind/mujoco #2172: closed loop crash or not accurate when simulation.

WeisonWEileen, 2024-10-24:
### Intro

Hi!

I am a undergrad student at sustech, I use MuJoCo for my research on sim2sim.

### My setup

ubuntu22.04  mujoco 3.2.3

### What's happening? What did you expect?

I use equality/connect to implement closed loop robot. However, it crash when moving. I am sure my relative coordinate is accurate because  I get it from solidworks. I expect the two hole site of the two link are aligned to each other.

https://github.com/user-attachments/assets/e9734706-b7ad-432b-ab8d-09ca11ccebc6

### Steps for reproduction

sw file too big to reproduce 

### Minimal model for reproduction

<mujoco model="robot">
    <compiler angle="radian" />
    <option timestep="0.001" gravity="0 0 -10" />
    <include file="./worlds.xml" />
    <asset>
        <mesh name="torso_link" file="torso_link.STL" />
        <mesh name="motor_r_Link" file="motor_r_Link.STL" />
        <mesh name="thigh_r_Link" file="thigh_r_Link.STL" />
        <mesh name="shank_r_Link" file="shank_r_Link.STL" />
        <mesh name="foot_r_Link" file="foot_r_Link.STL" />
        <mesh name="crank_r_Link" file="crank_r_Link.STL" />
        <mesh name="calf_r_Link" file="calf_r_Link.STL" />
        <mesh name="motor_l_Link" file="motor_l_Link.STL" />
        <mesh name="thigh_l_Link" file="thigh_l_Link.STL" />
        <mesh name="shank_l_Link" file="shank_l_Link.STL" />
        <mesh name="foot_l_Link" file="foot_l_Link.STL" />
        <mesh name="crank_l_Link" file="crank_l_Link.STL" />
        <mesh name="calf_l_Link" file="calf_l_Link.STL" />
    </asset>

    

    <worldbody>
        <light name="spotlight" mode="targetbodycom" target="torso_Link"
            diffuse=".8 .8 .8" specular="0.3 0.3 0.3" pos="0 -6 4" cutoff="30" />
        <camera name="topview" pos="0 0 200" xyaxes="1 0 0 0 1 0" target="torso_Link" />
        <geom name="ground" type="plane" size="0 0 1" pos="0.0 0 -1.02746" quat="1 0 0 0"
            material="matplane" condim="1" conaffinity='15' />
        <body name="ctbase">
            <inertial pos="0 0 0" mass="0" diaginertia="0 0 0" />
            <geom size="0.01 1.015 0.25" type="box" contype="0" conaffinity="0" group="1"
                rgba="0.5 0.5 0.5 0.2" />
            <body name="x_block" pos="-0.4 0 0">
                <inertial pos="0 0 0" quat="0.820473 0.339851 -0.17592 0.424708" mass="1e-07"
                    diaginertia="1e-10 1e-10 1e-10" />
                <geom size="0.01 0.015 0.02" type="box" contype="0" conaffinity="0" group="1"
                    rgba="1.0 0.0 0.0 0.15" />
                <body name="y_block" pos="-0.2 0 0">
                    <inertial pos="0 0 0" quat="0.820473 0.339851 -0.17592 0.424708" mass="1e-07"
                        diaginertia="1e-10 1e-10 1e-10" />
                    <joint name="x_prismatic_joint" pos="0 0 0" axis="1 0 0" type="slide"
                        limited="true" range="-100 100" damping="10" frictionloss="1" />
                    <geom size="0.02" contype="0" conaffinity="0" group="1" rgba="0.0 1.0 0.0 0.15" />
                    <body name="z_block" pos="-0.2 0 0">
                        <inertial pos="0 0 0" quat="0.820473 0.339851 -0.17592 0.424708"
                            mass="1e-07" diaginertia="1e-10 1e-10 1e-10" />
                        <joint name="y_prismatic_joint" pos="0 0 0" axis="0 1 0" type="slide"
                            limited="true" range="-100 100" damping="10" frictionloss="1" />
[... 322 lines cut ...]

    <equality>
        <connect name="connect_l" active="true" body1="calf_l_Link" body2="shank_l_Link"
            anchor="0.56 0 0.00885" />

        <connect name="connect_r" active="true" body1="calf_r_Link" body2="shank_r_Link"
            anchor="0.56 0 0.00885" />
    </equality>

</mujoco>

### Code required for reproduction

    <equality>
        <connect name="connect_l" active="true" body1="calf_l_Link" body2="shank_l_Link"
            anchor="0.56 0 0.00885" />

        <connect name="connect_r" active="true" body1="calf_r_Link" body2="shank_r_Link"
            anchor="0.56 0 0.00885" />
    </equality>

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2024-10-24:
1. The video looks fine, where is the crash?
2. Your model is not loadable without the assets, so we can't reproduce.

yuvaltassa, 2024-10-28:
Are you aware the if you click the 'E' button in `simulate`, you can see the two points of the equality constraint? That should  help you debug what is happening.

If that's doesn't help you solve it, and you still need us to look at your example, it would be much easier for us if you posted a file that only contains primitives and does not require meshes. You can easily ask MuJoCo to fit primitives in place of the meshes, then save the XML and delete all the mesh assets. This will give you a self-contained XML that you can paste here.

Regarding using sites to define the constraints, we just discovered several bugs thanks to @yinfanyi's #2179, so please don't use those until the fix has been submitted.

WeisonWEileen, 2024-10-28:
oh, After I click  'E' button, I see there are difference between the two points I want to connect. What does it implied(right leg is successful connected)? Does it means the two sites of left leg are coincident in position ?
[image]
original screenshot(without clicking E)
[image]

yuvaltassa, 2024-10-29:
This looks fine. These are just different kinds of [z-fighting](https://en.wikipedia.org/wiki/Z-fighting), the equality is satisfied in the default configuration.

Okay, I looked at your model posted above. I uncommented line 394 in `test.xml` and played with the mechanism, both constraints work fine. My assumption is that this is one of the bugs that was fixed in the change that fixed #2179. If you build from source everything will work as expected.

WeisonWEileen, 2024-11-01:
Thanks for replied! After install the latest version of mujoco(3.2.5), the problem metioned above was successful solved. However, the constrained is not tight when moving, which caused a lot of trouble for my Angle conversion in the simulation. It is a bug ?

https://github.com/user-attachments/assets/55756c62-20f7-44c9-ba1c-0865783952cf

yuvaltassa, 2024-11-04:
Not a bug. Default settings are fairly soft, make the equality constraints stiffer.