google-deepmind/mujoco #2174: Weld Constraint Specified with Sites Not Working in MJX

HaoruXue, 2024-10-25:
### Intro

Hi!

I am a PhD student at UC Berkeley, I use MuJoCo for my research on humanoid whole-body control.

### My setup

Mujoco 3.2.4, MJX 3.2.3, Python 3.10, Ubuntu 22.04

### What's happening? What did you expect?

Weld equality in Mujoco can be specified with either `body1` and `body2` or `site1` and `site2`. However, specifying with site is not working in MJX.

### Steps for reproduction

1. Load the model below.
2. Run the model with Mujoco `python3 -m mujoco.viewer --mjcf=test.xml`. Drag around the cube and the sphere will follow.
3. Run the model with MJX `mjx-viewer  --mjcf=test.xml`. The sphere falls to the ground although constrained by the weld joint.
4. Comment out the weld constraint specified with site and enable the weld constraint specified with body. Now it works properly in MJX.

### Minimal model for reproduction

```xml
<mujoco model="test">
  <statistic center="0 0 1" extent="1.8"/>

  <option iterations="2" ls_iterations="5" gravity="0.0 0.0 -9.81">
    <flag eulerdamp="disable" />
  </option>

  <visual>
    <headlight diffuse="0.6 0.6 0.6" ambient="0.3 0.3 0.3" specular="0 0 0"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <global azimuth="160" elevation="-20"/>
  </visual>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3"
      markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>

  <worldbody>
    <!-- Lighting and ground plane -->
    <light pos="0 0 3.5" dir="0 0 -1" directional="true"/>
    <geom name="floor" size="0 0 0.05" type="plane" material="groundplane"/>

    <body name="cube1" pos="0 0 1">
      <!-- Joints allowing movement in x, y and rotation around z -->
      <joint name="cube1_x" type="slide" axis="1 0 0" limited="false"/>
      <joint name="cube1_y" type="slide" axis="0 1 0" limited="false"/>
      <joint name="cube1_rot" type="hinge" axis="0 0 1" limited="false"/>
      <inertial pos="0 0 0" quat="1 0 0 0" mass="1" diaginertia="1 1 1"/>
      <geom name="cube1_geom" type="box" size="0.1 0.1 0.1" contype="0" conaffinity="0" rgba="0.9 0.9 0.9 0.5"/>
      <site name="cube1_site" pos="0 0 0"/>
    </body>

    <body name="sphere1" pos="0 0 1">
      <geom name="sphere1_geom" type="sphere" size="0.05" contype="0" conaffinity="1" rgba="0.9 0.0 0.0 0.5"/>
      <inertial pos="0 0 0" quat="1 0 0 0" mass="1" diaginertia="1 1 1"/>
      <site name="sphere1_site" pos="0 0 0"/>
      <freejoint/>
    </body>
  </worldbody>

  <!-- weld with body works but weld with site fails -->
  <equality>
    <weld torquescale="1" solref="0.1 1" solimp="0.1 0.1 0.1" site1="cube1_site" site2="sphere1_site"/>
    <!-- <weld torquescale="1" solref="0.1 1" solimp="0.1 0.1 0.1" body1="cube1" body2="sphere1"/> -->
  </equality>

</mujoco>
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

thowell, 2024-10-25:
Please try MJX version 3.2.4 for this feature.

HaoruXue, 2024-10-25:
Thanks! works now in 3.2.4.

closed 2024-10-25 by HaoruXue