google-deepmind/mujoco #2179: Simulation Instability After Adding Sites to Parallel Mechanism

yinfanyi, 2024-10-26:
### Intro

Hi!

I am a undergrad student, I use MuJoCo for my research on parallel mechanism modeling.

### My setup

MuJoCo version:3.2.4

### What's happening? What did you expect?

When no site has been added to the black ball ( body carrier), I can use the mouse to drag the mechanism in the Simulate app, and observe its movement. However, after adding a site to the body 'carrier', this functionality is no longer available. Logically, adding sites should not affect the movement of the mechanism.

[image]

### Steps for reproduction

this model can work.
```
<?xml version='1.0' encoding='utf-8'?>
<mujoco model="TT12_environment">
  <compiler autolimits="true" coordinate="local" angle="degree" inertiafromgeom="true" />
  <option timestep="1e-06" gravity="0 0 0" integrator="Euler">
    <flag override="disable" energy="enable" />
  </option>
  <visual>
    <headlight ambient="0.59999999999999998 0.59999999999999998 0.59999999999999998" diffuse="0.29999999999999999 0.29999999999999999 0.29999999999999999" specular="0.10000000000000001 0.10000000000000001 0.10000000000000001" />
  </visual>
  <asset>
    <texture name="skybox" type="skybox" builtin="gradient" rgb1="0.29999999999999999 0.5 0.69999999999999996" rgb2="0 0 0" width="512" height="3072" />
    <texture name="2d" type="2d" builtin="checker" rgb1="0.20000000000000001 0.29999999999999999 0.40000000000000002" rgb2="0.10000000000000001 0.20000000000000001 0.29999999999999999" width="300" height="300" />
    <material name="grid" texture="2d" texrepeat="5 5" reflectance="0.20000000000000001" />
  </asset>
  <worldbody>
    <geom name="floor" type="plane" condim="6" priority="1" size="0 0 0.125" material="grid" friction="0.5 0.2821614807335161 0.071577037975213545" solref="-5304.0750037206153 -487.91575316053519" solimp="0.58198118105385244 0.038070880959479185 0.088896595964777433 0.5 2" pos="0 0 0" />
    <light name="light_-2" pos="-2 -1 3" dir="2 1 -2" />
    <light name="light_2" pos="2 -1 3" dir="-2 1 -2" />
    <site name="world_site" pos="0 0 0" />
    <body name="carrier" pos="0 0 0.4">
        <freejoint name="carrier_joint" />
        <geom name="carrier_geo" type="sphere" size="0.1" rgba="0.1 0.1 0.1 0.5" />
        
        <body name="link_p1"> 
          <joint name="R_joint_rotor_link_p1" type="hinge" pos="0 0 0" axis="0 1 0" />
          <geom name="link_p1_geo" type="cylinder" size="0.003" fromto="0 0 0 0 0 -0.19" rgba="0.9 0.9 0.2 1"/>
          <body name="link_d1" pos="0 0 -0.19">
            <joint name="R_joint_link_p1_d1" type="hinge" pos="0 0 0" axis="0 0 1" />
            <geom name="link_d1_geo" type="cylinder" size="0.003" fromto="0 0 0 -0.13435028842544403 -0.13435028842544403 0.19" rgba="0 0 0.9 1"/>
            <site name="d1_in" size="0.004" rgba="0.1 0.5 0.7 0.1" pos="-0.13 -0.13 0.19" />
            <site name="d1_out" size="0.004" rgba="0.1 0.7 0.5 0.1" pos="-0.14 -0.14 0.19" />
          </body>
        </body>

        <body name="link_p2"> 
          <joint name="R_joint_rotor_link_p2" type="hinge" pos="0 0 0" axis="0 1 0" />
          <geom name="link_p2_geo" type="cylinder" size="0.003" fromto="0 0 0 0.19 0 0" rgba="0.5 0.8 0.2 1"/>
          <body name="link_d2" pos="0.19 0 0">
            <joint name="R_joint_link_p2_d2" type="hinge" pos="0 0 0" axis="1 0 0" />
            <geom name="link_d2_geo" type="cylinder" size="0.003" fromto="0 0 0 -0.19 0.13435028842544403 -0.13435028842544403" rgba="0 0 0.9 1"/>
            <site name="d2_in" size="0.004" rgba="0.1 0.5 0.7 0.1" pos="-0.19 0.13 -0.13"/>
            <site name="d2_out" size="0.004" rgba="0.1 0.7 0.5 0.1" pos="-0.19 0.14 -0.14"/>
          </body>
        </body>

        <body name="link_p3"> 
          <joint name="R_joint_rotor_link_p3" type="hinge" pos="0 0 0" axis="0 1 0" />
          <geom name="link_p3_geo" type="cylinder" size="0.003" fromto="0 0 0 -0.13435028842544403 0 0.13435028842544403" rgba="0.8 0.6 0.2 1"/>
          <body name="link_d3" pos="-0.13435028842544403 0 0.13435028842544403" euler="0 45 0">
            <joint name="R_joint_link_p3_d3" type="hinge" pos="0 0 0" axis="1 0 0" />
            <geom name="link_d3_geo" type="cylinder" size="0.003" fromto="0 0 0 0.19 0 0.19" rgba="0 0 0.9 1"/>
            <site name="d3_in" size="0.004" rgba="0.1 0.5 0.7 0.1" pos="0.19 0 0.18384776310850237"/>
            <site name="d3_out" size="0.004" rgba="0.1 0.7 0.5 0.1" pos="0.19 0 0.19798989873223333"/>
          </body>
        </body>
        <!-- <site name="carrier_x" type="box" size="0.005 0.005 0.005" rgba="1 0 0 1" pos="0.1 0 0" /> -->
        <!-- <site name="carrier_y" type="box" size="0.005 0.005 0.005" rgba="0 1 0 1" pos="0 0.1 0" /> -->
        <!-- <site name="carrier_z" type="box" size="0.005 0.005 0.005" rgba="0 0 1 1" pos="0 0 0.1" /> -->
    </body>

    <body name="ring" pos="0 0 0.4">
      <freejoint name="ring_joint" />
      <geom name="ring_geo" type="sphere" size="0.19" rgba="0.9 0.1 0.1 0.2" />
      <site name="d1_in_ring" size="0.002" rgba="0.1 0.1 0.5 1" pos="-0.13 -0.13 0" />
      <site name="d1_out_ring" size="0.002" rgba="0.1 0.5 0.1 1" pos="-0.14 -0.14 0" />

      <site name="d2_in_ring" size="0.002" rgba="0.1 0.1 0.5 1" pos="0 0.13 -0.13" />
      <site name="d2_out_ring" size="0.002" rgba="0.1 0.5 0.1 1" pos="0 0.14 -0.14" />

      <site name="d3_in_ring" size="0.002" rgba="0.1 0.1 0.5 1" pos="0.13 0 0.13" />
      <site name="d3_out_ring" size="0.002" rgba="0.1 0.5 0.1 1" pos="0.14 0 0.14" />

      <site name="ring_x" type="box" size="0.005 0.005 0.005" rgba="1 0 0 1" pos="0.19 0 0"/>
      <site name="ring_y" type="box" size="0.005 0.005 0.005" rgba="0 1 0 1" pos="0 0.19 0"/>
      <site name="ring_z" type="box" size="0.005 0.005 0.005" rgba="0 0 1 1" pos="0 0 0.19"/>
    </body>

  </worldbody>

  <contact>
    <exclude body1="carrier" body2="ring" />
    <exclude body1="ring" body2="link_p1" />
    <exclude body1="ring" body2="link_p2" />
    <exclude body1="ring" body2="link_p3" />

    <exclude body1="ring" body2="link_d1" />
    <exclude body1="ring" body2="link_d2" />
    <exclude body1="ring" body2="link_d3" />
    
    <exclude body1="link_p1" body2="link_p2" />
    <exclude body1="link_p1" body2="link_p3" />
    <exclude body1="link_p2" body2="link_p3" />

    <exclude body1="link_d1" body2="link_p2" />
    <exclude body1="link_d1" body2="link_p3" />
    <exclude body1="link_d2" body2="link_p3" />
    <exclude body1="link_d2" body2="link_p1" />
    <exclude body1="link_d3" body2="link_p1" />
    <exclude body1="link_d3" body2="link_p2" />

    <exclude body1="link_d1" body2="carrier" />
    <exclude body1="link_d2" body2="carrier" />
    <exclude body1="link_d3" body2="carrier" />

  </contact>
  <equality>
    <connect site1="d1_in" site2="d1_in_ring"/>
    <connect site1="d1_out" site2="d1_out_ring"/>

    <connect site1="d2_in" site2="d2_in_ring"/>
    <connect site1="d2_out" site2="d2_out_ring"/>

    <connect site1="d3_in" site2="d3_in_ring"/>
    <connect site1="d3_out" site2="d3_out_ring"/>
  </equality>
</mujoco>
```
If these three lines of code are uncommented, it no longer works:
```
        <!-- <site name="carrier_x" type="box" size="0.005 0.005 0.005" rgba="1 0 0 1" pos="0.1 0 0" /> -->
        <!-- <site name="carrier_y" type="box" size="0.005 0.005 0.005" rgba="0 1 0 1" pos="0 0.1 0" /> -->
        <!-- <site name="carrier_z" type="box" size="0.005 0.005 0.005" rgba="0 0 1 1" pos="0 0 0.1" /> -->
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2024-10-26:
Wow @yinfanyi, I took a look, this is a serious bug!

Fix incoming.

closed 2024-10-28 by copybara-service[bot]

yuvaltassa, 2024-10-28:
This is now fixed. Build from source or wait for the next release.

commit 864b805a (2024-10-28): Fix multiple bugs related to connect and weld constraints with site semantics. Fixes #2179

The introduction of site specification to connects and welds in 3.2.3 conditionally changed the semantics of `mjData.eq_obj1id` and `mjData.eq_obj2id`. These changes were not properly propagated in several places leading to incorrect computations of constraint inertia, readings of affected force/torque sensors and runtime enabling/disabling of such constraints.

PiperOrigin-RevId: 690670420
Change-Id: I55ee8a013cbee8457f8d6c7f33c2981aedafbab6