google-deepmind/mujoco #2181: MJX external force Bug

lucattyy, 2024-10-27:
### Intro

Hi!

I am a student and I am using MJX for my research on Robotics.

### My setup

I am using mujoco-mjx version is 3.1.3 on Ubuntu22.04

### What's happening? What did you expect?

I need to apply external force to robot and I discover that parametre 'xfrc_applied' can be checked in mjx, but something wrong happened when I try to modify them.

Traceback (most recent call last):
  File "/home/pyproject/mjx/scripts/env.py", line 66, in <module>
    state = jit_step(state, ctrl)
  File "/home/pyproject/mjx/scripts/../env/env.py", line 255, in step
    state.pipeline_state.xfrc_applied = state.pipeline_state.xfrc_applied.at[1, 2].set(75)
  File "<string>", line 4, in __setattr__
dataclasses.FrozenInstanceError: cannot assign to field 'xfrc_applied'

It seems that 'xfrc_applied ' has been frozen. Is there any method can help apply external force to robot in MJX?

### Steps for reproduction

1. Load the model below.
2. Run the code below.
3. See green sphere (should be red).

### Minimal model for reproduction

If you encountered the issue in a complex model, please simplify it as much as possible (while still reproducing the issue).

<details>
<summary>minimal XML</summary>

```XML

<mujoco>
  <worldbody>
    <light pos="0 0 1"/>
    <geom type="sphere" size="1" rgba="1 0 0 1"/>
  </worldbody>
</mujoco>

```
</details>

### Code required for reproduction

```python
import mujoco
import mediapy as media
model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)

with mujoco.Renderer(model) as renderer:
  mujoco.mj_forward(model, data)
  renderer.update_scene(data)

  media.show_image(renderer.render())
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

kevinzakka, 2024-10-27:
You’re trying to mutate a frozen field. You need to use the .at(idx).set(val) semantics.

closed 2024-10-27 by kevinzakka

lucattyy, 2024-10-28:
That's exactly what I did. The .at(idx).set(val) semantics can be used to modify value of MjData.ctrl, but still cannot solve the problem. 
state.pipeline_state.xfrc_applied = state.pipeline_state.xfrc_applied.at[1, 2].set(75)
dataclasses.FrozenInstanceError: cannot assign to field 'xfrc_applied'

Do you mean that MJX do not support external force applied through xfrc_applied currently?

kevinzakka, 2024-10-28:
Oh forgot to add that you also need to use the `replace` method of the dataclass:

```python

pipeline_state = pipeline_state.replace(xfrc_applied=xfrc_applied.at[1, 2].set(75))
```

lucattyy, 2024-10-30:
Thx! It really help a lot!