google-deepmind/mujoco #2186: Support for Building MuJoCo from Source Using Bazel

bd-jahn, 2024-10-28:
### The feature, motivation and pitch

Hello MuJoCo Team,

Our team has been integrating MuJoCo within a larger Bazel project. Currently, we fetch and link the prebuilt release binaries from this repository for C++ projects and download mujoco-py from PyPI for Python projects.

While this setup works well in most cases, we’d like to build the libraries directly using Bazel for several reasons, including hermeticity, reducing ODR issues from common dependencies, and improved integration with our existing infrastructure. I’ve developed an initial proof of concept that compiles `libmujoco.so` using Bazel targeting Ubuntu 22.04 / x86 configuration, and I’m open to contributing it to the community.

I am aware that the current CMake build system is more comprehensive, supporting a range of configurations and Python bindings, but I wanted to get your guidance before proceeding further.

Thank you!

yuvaltassa, 2024-10-29:
We would love to accept this contribution.
Please send us a PR, we look forward to it!
Closing for now, let's continue the conversation in the PR 🙂

closed 2024-10-29 by yuvaltassa

bd-jahn, 2024-11-11:
Hi @yuvaltassa 
I wanted to open a PR to the upstream repository, but my patches were based on v3.2.2. Since there’s no branch in the upstream repository associated with commit 5383d1c, I wasn’t able to create the PR there. Instead, I opened [a PR](https://github.com/bd-jahn/mujoco/pull/1) in my forked repository. Please let me know if there’s a better way to track this.

bd-jahn, 2024-11-12:
I managed to rebase the branch onto `main` and opened #2225. Let's continue the conversation in the PR!