google-deepmind/mujoco #2204: Manual loading of plugins libraries fails via python api, but works fine via c++ api

aaarne, 2024-11-03:
### Intro

Hi!

I am a graduate student at TU Munich and the German Aerospace Center (DLR); I use MuJoCo to research elastic robots with variable stiffness actuators.

### My setup

MuJoCo version: 2.3.4 
API: Both (Plugin Development: C++, Using the model: Python)
Architecture: Arch Linux on 6.11.5 kernel

### What's happening? What did you expect?

I have an actuator plugin compiled into a separate shared library (.so) and a model using that plugin. Different scenarios:

1. I use the model via the mujoco __C++__ API. I register the library via `mj_loadPluginLibrary`/`mj_loadAllPluginLibraries` or I use the `simulate` application and put the so where `simulate` wants. This works fine! (so model and plugin should be okay)
2. I use the __Python__ API and put the plugin .so into the mujoco plugin folder, where mujoco's `__init__.py` finds and loads it. This also works fine. (so it should generally work using the python API)
3. I use the __Python__ API, but keep my plugin .so in some folder where I want it. I use `mujoco.mj_loadPluginLibrary`/`mujoco.mj_loadAllPluginLibraries` to load the plugin manual via the python API. This does __not__ work. I see that the library registers the plugin. I have compiled a printout into the .so that prints a message before registering the plugin in the `mjPLUGIN_LIB_INIT`, and I see that message. However, it then fails on loading the XML model with `ValueError: XML Error: plugin vsa not found`.

### Steps for reproduction

1. Generate a plugin library compiled into a separate .so file, and a model referencing that plugin.
2. Load the library in python using the exposed functions `mj_loadPluginLibrary` or `mj_loadAllPluginLibraries`.
3. The library registers the plugins.
4. Load the XML model referencing the plugin. It will fail claiming it cannot find the plugin.

### Minimal model for reproduction

```XML

<mujoco>
  <extension>
    <plugin plugin="testplugin"/>
  </extension>
  <worldbody>
    <light pos="0 0 1"/>
    <geom type="sphere" size="1" rgba="1 0 0 1"/>
  </worldbody>
</mujoco>
```

### Code required for reproduction

```python
import mujoco

mujoco.mj_loadPluginLibrary("somepath/mypluginlibrary.so")  
# after this call mjPLUGIN_LIB_INIT in mypluginlibrary.so is executed, so it is somehow loaded.

model = mujoco.MjModel.from_xml_string("some_model_using_testplugin_in_mypluginlibrary.xml")
# This fails with ValueError: XML Error: plugin testplugin not found, but testplugin is registered by mypluginlibrary.so
```

Doing the above steps via the c++ API works on the same files. It also works via the python API when removing the `mj_loadPluginLibrary` call and placing `mypluginlibrary.so` into `mujoco/plugin`.

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

saran-t, 2025-01-03:
Can you try replacing `mujoco.mj_loadPluginLibrary` with `ctypes.CDLL` and see if that changes anything?

DavidPL1, 2025-07-08:
Using MuJoCo 3.3.3 I can confirm that both `mujoco.mj_loadPluginLibrary` and `ctypes.CDLL` do work