google-deepmind/mujoco #2212: [MJX] mjx.get_data error introduced in v3.2.5

driesmarzougui, 2024-11-05:
### Intro

Hi!

I am a phd student at Ghent University, I use MuJoCo for my research on bio-inspired robots.

### My setup

MJX v3.2.5

### What's happening? What did you expect?

`mjx.get_data` throws the following error `TypeError: mju_dense2sparse(): incompatible function arguments. The following argument types are supported:...`. Bug probably introduced in a51f346.

### Steps for reproduction

1. Load the model below.
2. Run the code below.

### Minimal model for reproduction

If you encountered the issue in a complex model, please simplify it as much as possible (while still reproducing the issue).

<details>
<summary>minimal XML</summary>

```XML

<mujoco>
  <compiler autolimits="true"/>
  <worldbody>
    <body name="box">
      <joint name="slide1" type="slide" axis="1 0 0" />
      <geom type="box" size=".05 .05 .05" mass="1"/>
    </body>
  </worldbody>
  <actuator>
    <motor joint="slide1"/>
  </actuator>
</mujoco>

```
</details>

### Code required for reproduction

```python
import mujoco
from mujoco import mjx

m = mujoco.MjModel.from_xml_string(xml)
d = mujoco.MjData(m)
dx = mjx.put_data(m, d)

mjx.get_data(m, dx)
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

asuzukii, 2024-11-22:
I apologize to comment on a closed thread, but how does [f9569cd](https://github.com/google-deepmind/mujoco/commit/f9569cdab0d3f56d37f7ede598563becf464c9e7) actually resolve the mismatch between the `mat` type (`float64`) and input type (`array([[0.]], dtype=float32)`) in the given bug report)?

btaba, 2024-12-04:
@asuzukii there was an issue with the int32 inputs, not the float inputs.  Please open a bug if you're seeing one!