google-deepmind/mujoco #2215: Stiffness and Damping Switched in Tendon .cs file

sriddle97, 2024-11-06:
### Intro

Hello,
I've been trying to debug my model of a soft robot constructed using composite cable and tendon elements. I made a separate post about the cable element moments of inertia but I also noticed that changing the tendon stiffnesses in the xml file did not result in a higher stiffness when the simulation runs. 

### My setup

Mujoco, python

### What's happening? What did you expect?

As part of the debug process I tried changing the stiffness parameter in the xml file and noticed that multiplying that by even a factor of 1000 did not appreciably change the stiffness experienced when the simulation runs. The force required to contract the tendon a prescribed amount for stiffness=50 is 5N and for stiffness=50,000 it is 5.8N.

I went through the Mujoco Github repository looking for potential bugs and stumbled across one that may be relevant. In the MjBaseTendon.cs file [https://github.com/google-deepmind/mujoco/blob/main/unity/Runtime/Components/Tendons/MjBaseTendon.cs](url) it appears that the stiffness is set equal to the damping and vice versa on lines 45 and 46. 

`Stiffness = mjcf.GetFloatAttribute("damping");`
`Damping = mjcf.GetFloatAttribute("stiffness");`

It looks correct later in lines 55 and 56 but I am not sure what purpose each of those components of the code serve as they are in different `protected override` sections.

Could this by why changing my tendon stiffness value is not actually resulting in increased stiffness at run time? 

### Steps for reproduction

1. Load the model below.
2. Run the code below.
3. See green sphere (should be red).

### Minimal model for reproduction

If you encountered the issue in a complex model, please simplify it as much as possible (while still reproducing the issue).

<details>
<summary>minimal XML</summary>

```XML

<mujoco>
  <worldbody>
    <light pos="0 0 1"/>
    <geom type="sphere" size="1" rgba="1 0 0 1"/>
  </worldbody>
</mujoco>

```
</details>

### Code required for reproduction

```python
import mujoco
import mediapy as media
model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)

with mujoco.Renderer(model) as renderer:
  mujoco.mj_forward(model, data)
  renderer.update_scene(data)

  media.show_image(renderer.render())
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2024-11-14:
@Balint-H this looks like an actual (easy to fix) bug. Can you confirm?

Balint-H, 2024-11-14:
Hello, yes this looks like a clear cut fix, will push it today.

sriddle97, 2024-11-18:
Thank you for taking care of this. Did this fix ever get pushed? I still see the original, incorrect lines in the .cs file.

Balint-H, 2024-11-18:
I sent the pull request for the fix, but it still needs to be approved by someone. @erez-tom could you help out?

closed 2025-02-03 by Balint-H