google-deepmind/mujoco #2221: `MjSpec` attachment messes `sol{ref,imp}` values

hartikainen, 2024-11-09:
### Intro

Hi!

I am a MuJoCo user working on manipulation.

### My setup

MuJoCo:
```sh
$ python -c "import mujoco; print(mujoco.__version__)"
3.2.5
```

### What's happening? What did you expect?

```python
import mujoco
import numpy as np
from robot_descriptions import fr3_mj_description
from robot_descriptions import shadow_dexee_mj_description

def main():
    hand_spec = mujoco.MjSpec.from_file(shadow_dexee_mj_description.MJCF_PATH)
    hand_model = hand_spec.compile()
    arm_spec = mujoco.MjSpec.from_file(fr3_mj_description.MJCF_PATH)
    arm_model = arm_spec.compile()

    combined_spec = arm_spec.copy()

    attachment_site = next(
        s for s in combined_spec.sites if s.name == "attachment_site"
    )
    attachment_site.attach(hand_spec.worldbody, "hand_base:", "")

    combined_model = combined_spec.compile()

    joint_names = {f"F{f}/J{j}" for f in range(3) for j in range(4)}
    joint_keys = {
        "M0",
        "armature",
        "axis",
        "damping",
        "frictionloss",
        "limited",
        "margin",
        "pos",
        "qpos0",
        "qpos_spring",
        "range",
        "simplenum",
        "solimp",
        "solref",
        "stiffness",
        "type",
    }
    for joint_name in joint_names:
        print(f"Checking joint {joint_name}")
        for key in joint_keys:
            hand_spec_value = getattr(hand_model.joint(joint_name), key)
            combined_spec_value = getattr(
                combined_model.joint(f"hand_base:{joint_name}"), key
            )

            if not np.allclose(hand_spec_value, combined_spec_value):
                print(
                    f"{key} does not match: {hand_spec_value} != {combined_spec_value}"
                )

if __name__ == "__main__":
    main()
```

Output:
```
Checking joint F1/J2
solref does not match: [[0.02 1.  ]] != [[1.   0.02]]
solimp does not match: [[9.5e-01 1.0e-03 5.0e-01 2.0e+00 9.0e-01]] != [[5.0e-01 2.0e+00 9.0e-01 9.5e-01 1.0e-03]]
Checking joint F0/J3
solref does not match: [[1.   0.02]] != [[0.02 1.  ]]
solimp does not match: [[5.0e-01 2.0e+00 9.0e-01 9.5e-01 1.0e-03]] != [[9.0e-01 9.5e-01 1.0e-03 5.0e-01 2.0e+00]]
Checking joint F2/J3
solref does not match: [[1.   0.02]] != [[0.02 1.  ]]
solimp does not match: [[9.5e-01 1.0e-03 5.0e-01 2.0e+00 9.0e-01]] != [[5.0e-01 2.0e+00 9.0e-01 9.5e-01 1.0e-03]]
Checking joint F2/J1
solref does not match: [[1.   0.02]] != [[0.02 1.  ]]
solimp does not match: [[2.0e+00 9.0e-01 9.5e-01 1.0e-03 5.0e-01]] != [[9.5e-01 1.0e-03 5.0e-01 2.0e+00 9.0e-01]]
Checking joint F0/J0
solref does not match: [[0.02 1.  ]] != [[1.   0.02]]
solimp does not match: [[9.0e-01 9.5e-01 1.0e-03 5.0e-01 2.0e+00]] != [[1.0e-03 5.0e-01 2.0e+00 9.0e-01 9.5e-01]]
Checking joint F2/J0
solref does not match: [[0.02 1.  ]] != [[1.   0.02]]
solimp does not match: [[5.0e-01 2.0e+00 9.0e-01 9.5e-01 1.0e-03]] != [[9.0e-01 9.5e-01 1.0e-03 5.0e-01 2.0e+00]]
Checking joint F1/J1
solref does not match: [[1.   0.02]] != [[0.02 1.  ]]
solimp does not match: [[9.0e-01 9.5e-01 1.0e-03 5.0e-01 2.0e+00]] != [[1.0e-03 5.0e-01 2.0e+00 9.0e-01 9.5e-01]]
Checking joint F0/J1
solref does not match: [[1.   0.02]] != [[0.02 1.  ]]
solimp does not match: [[9.5e-01 1.0e-03 5.0e-01 2.0e+00 9.0e-01]] != [[5.0e-01 2.0e+00 9.0e-01 9.5e-01 1.0e-03]]
Checking joint F1/J3
solref does not match: [[1.   0.02]] != [[0.02 1.  ]]
solimp does not match: [[1.0e-03 5.0e-01 2.0e+00 9.0e-01 9.5e-01]] != [[2.0e+00 9.0e-01 9.5e-01 1.0e-03 5.0e-01]]
Checking joint F1/J0
solref does not match: [[0.02 1.  ]] != [[1.   0.02]]
solimp does not match: [[2.0e+00 9.0e-01 9.5e-01 1.0e-03 5.0e-01]] != [[9.5e-01 1.0e-03 5.0e-01 2.0e+00 9.0e-01]]
Checking joint F0/J2
solref does not match: [[0.02 1.  ]] != [[1.   0.02]]
solimp does not match: [[1.0e-03 5.0e-01 2.0e+00 9.0e-01 9.5e-01]] != [[2.0e+00 9.0e-01 9.5e-01 1.0e-03 5.0e-01]]
Checking joint F2/J2
solref does not match: [[0.02 1.  ]] != [[1.   0.02]]
solimp does not match: [[9.0e-01 9.5e-01 1.0e-03 5.0e-01 2.0e+00]] != [[1.0e-03 5.0e-01 2.0e+00 9.0e-01 9.5e-01]]
```

I would expect the values to match but for some reason they seem to be ["rolled"](https://numpy.org/doc/stable/reference/generated/numpy.roll.html).

### Steps for reproduction

See above.

### Minimal model for reproduction

See above.

### Code required for reproduction

See above.

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

quagla, 2024-11-11:
I can't reproduce it in C so this looks like a binding issue (likely not even related to mjSpec). I'm having a deeper look. My MRE is 

```
  def test_solref(self):
    spec = mujoco.MjSpec.from_string("""
      <mujoco>
        <worldbody>
          <body>
            <geom size=".1"/>
            <joint axis="0 0 1"/>
            <body>
              <geom size=".1"/>
              <joint axis="0 0 1"/>
            </body>
          </body>
        </worldbody>
      </mujoco>
    """)

    model = spec.compile()
    for joint in range(model.njnt):
      solref = model.joint(joint).solref
      solimp = model.joint(joint).solimp
      np.testing.assert_array_equal(solref, [[0.02, 1]])
      np.testing.assert_array_equal(solimp, [[0.9, 0.95, 0.001, 0.5, 2.0]])
```

quagla, 2024-11-11:
In the meantime, could you please see if extracting the values with

```
solref = model.jnt_solref[i]
solimp = model.jnt_solimp[i]
```

fixes your issue?

hartikainen, 2024-11-11:
Indeed, extracting the values directly works as expected. Here's a bit simpler test case that passes:
```python
import mujoco
import numpy as np
from robot_descriptions import fr3_mj_description
from robot_descriptions import shadow_dexee_mj_description

def main():

    hand_spec = mujoco.MjSpec.from_file(shadow_dexee_mj_description.MJCF_PATH)
    hand_model = hand_spec.compile()
    arm_spec = mujoco.MjSpec.from_file(fr3_mj_description.MJCF_PATH)
    arm_model = arm_spec.compile()

    combined_spec = arm_spec.copy()

    attachment_site = next(s for s in combined_spec.sites if s.name == "attachment_site")
    attachment_site.attach(hand_spec.worldbody, "hand_base:", "")

    combined_model = combined_spec.compile()

    np.testing.assert_equal(
        hand_model.jnt_solref,
        combined_model.jnt_solref[None:hand_model.njnt, ...],
    )
    np.testing.assert_equal(
        arm_model.jnt_solref,
        combined_model.jnt_solref[hand_model.njnt:None, ...],
    )
    np.testing.assert_equal(
        hand_model.jnt_solimp,
        combined_model.jnt_solimp[None:hand_model.njnt, ...],
    )
    np.testing.assert_equal(
        arm_model.jnt_solimp,
        combined_model.jnt_solimp[hand_model.njnt:None, ...],
    )

if __name__ == "__main__":
    main()
```

commit f7cc1c2e (2024-11-11): Read solref from jnt_solref when the user calls model.joint.solref.

Same for solimp. Fixes #2221.

PiperOrigin-RevId: 695416121
Change-Id: I962dc12b13fa4d02dc349ecad1a1ddc7085565db
closed 2024-11-11 by copybara-service[bot]