google-deepmind/mujoco #2223: texrepeat attribute in material broken

snasiriany, 2024-11-12:
### Intro

Hi!

I'm part of the RoboSuite and [RoboCasa](https://github.com/robocasa/robocasa) team.

### My setup

I noticed that textures break if the texrepeat attributes are set less than 1. Eg. if `texrepeat = "0.1 1"` or `texrepeat = "1 0.1"` or `texrepeat = "0.1 0.1"`. All of these result in the texture collapsing in the dimensions that are set less than 1.

This wasn't an issue before MuJoCo 3.2.0, but in 3.2.{0,1,2,3,4,5} I'm noticing this happening.

Screenshots attached for v3.1.6 followed by v3.2.0:
[image]
[image]

### What's happening? What did you expect?

Setting `rgba="1 0 0 1"` renders as **green**, I expected it to render as **red**.

Here is a screen-shot showing a green sphere:

### Steps for reproduction

1. Download the attached model [simple_model.zip](https://github.com/user-attachments/files/17709519/simple_model.zip)
2. Run `python -m mujoco.viewer --mjcf <path-to-model.xml>`
3. Run once for MuJoCo 3.1.6 and once for MuJoCo 3.2.0.

### Minimal model for reproduction

If you encountered the issue in a complex model, please simplify it as much as possible (while still reproducing the issue).

<details>
<summary>minimal XML</summary>

```XML

<mujoco model="base">
  <asset>
    <texture type="skybox" builtin="flat" rgb1="1 1 1" rgb2="1 1 1" width="256" height="1536" />
    <texture type="2d" name="floor_room_wall" file="assets/textures/wood/gray_wood_planks.png" />
    <material name="floor_room_wall_mat" texture="floor_room_wall" texrepeat="0.1 0.1" texuniform="true" />
  </asset>
  <worldbody>
    <light pos="2 -2 2" dir="0.009999 0.009999 -0.9999" directional="true" castshadow="false" ambient="0.3 0.3 0.3" diffuse="0.3 0.3 0.3" />
    <body name="floor_room_main" pos="2.75 -1.5 -0.02" quat="0.707107 0 0 0.707107">
      <geom name="floor_room_g0" size="1.54 2.79 0.02" type="box" rgba="0.5 0 0 1" />
      <geom name="floor_room_g0_vis" size="1.54 2.79 0.02" type="box" contype="0" conaffinity="0" group="1" mass="0" material="floor_room_wall_mat" />
    </body>
  </worldbody>
</mujoco>

```
</details>

### Code required for reproduction

N/A

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2024-11-14:
What does `texrepeat` with a non-integer value mean anyway? 
I.e. what do you expect it to do?

Can you please do the following, in 3.1.6 simulate do Print Model. Do the same in 3.2.0. Do you see any difference in the MATERIAL section?

snasiriany, 2024-11-14:
Thanks for the pointer. Here's the diff: https://www.diffchecker.com/AHmx27GH/

There's indeed a difference in the MATERIAL section:
v3.1.6:
```
MATERIAL 0:
  name                  floor_room_wall_mat
  mat_texid             1 
  ...
```

v3.2.0:
```
MATERIAL 0:
  name                  floor_room_wall_mat
  mat_texid             1  -1  -1  -1  -1  -1 
  ...
```

I expected that with a smaller value the images are dilated / zoomed into. Eg I tired for a better illustration:
`texrepeat="1 1"`:
[image]

and `texrepeat="0.5 0.5"`:
[image]

In this case the dilation factor is 1/0.5 = 2x, it makes sense conceptually to me, even if it's not an integer.

Attaching the model with the better texture: [simple_model_new.zip](https://github.com/user-attachments/files/17754667/simple_model_new.zip)

closed 2024-11-19 by copybara-service[bot]

yuvaltassa, 2024-11-19:
There was a mistaken cast from float to int, now fixed.

commit 5c23ae11 (2024-11-19): Fix bug in material `texrepeat` attribute, fixes #2223.

Was mistakenly cast from float to int, bug introduced in 3.2.0

PiperOrigin-RevId: 697993181
Change-Id: I643a07e2b0866a4af56d7cbd3344e764a7caf56f