google-deepmind/mujoco #2232: Mujoco fails to install on python 3.13

smorad, 2024-11-15:
### Intro

Hi!

I am a professor at the University of Macau, and I use MuJoCo for my research on RL.

### My setup

```
python==3.13
ubuntu==22.04
```

### What's happening? What did you expect?

`mujoco` fails to install on the latest python, but succeeds on an older version of python

### Steps for reproduction

`pip install mujoco` for python 3.13.

### Minimal model for reproduction

If you encountered the issue in a complex model, please simplify it as much as possible (while still reproducing the issue).

<details>
<summary>minimal XML</summary>

```XML

<mujoco>
  <worldbody>
    <light pos="0 0 1"/>
    <geom type="sphere" size="1" rgba="1 0 0 1"/>
  </worldbody>
</mujoco>

```
</details>

### Code required for reproduction

The following works

```
conda create -n py312 python=3.12
pip install mujoco
```

Using the latest python

```
conda create -n py313 python
pip install mujoco
```

Results in
```
        File "/tmp/pip-build-env-y7lyori3/overlay/lib/python3.13/site-packages/setuptools/_distutils/dist.py", line 954, in run_commands
          self.run_command(cmd)
          ~~~~~~~~~~~~~~~~^^^^^
        File "/tmp/pip-build-env-y7lyori3/overlay/lib/python3.13/site-packages/setuptools/dist.py", line 994, in run_command
          super().run_command(command)
          ~~~~~~~~~~~~~~~~~~~^^^^^^^^^
        File "/tmp/pip-build-env-y7lyori3/overlay/lib/python3.13/site-packages/setuptools/_distutils/dist.py", line 973, in run_command
          cmd_obj.run()
          ~~~~~~~~~~~^^
        File "/tmp/pip-build-env-y7lyori3/overlay/lib/python3.13/site-packages/setuptools/command/bdist_wheel.py", line 383, in run
          self.run_command("build")
          ~~~~~~~~~~~~~~~~^^^^^^^^^
        File "/tmp/pip-build-env-y7lyori3/overlay/lib/python3.13/site-packages/setuptools/_distutils/cmd.py", line 316, in run_command
          self.distribution.run_command(command)
          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
        File "/tmp/pip-build-env-y7lyori3/overlay/lib/python3.13/site-packages/setuptools/dist.py", line 994, in run_command
          super().run_command(command)
          ~~~~~~~~~~~~~~~~~~~^^^^^^^^^
        File "/tmp/pip-build-env-y7lyori3/overlay/lib/python3.13/site-packages/setuptools/_distutils/dist.py", line 973, in run_command
          cmd_obj.run()
          ~~~~~~~~~~~^^
        File "/tmp/pip-build-env-y7lyori3/overlay/lib/python3.13/site-packages/setuptools/_distutils/command/build.py", line 135, in run
          self.run_command(cmd_name)
          ~~~~~~~~~~~~~~~~^^^^^^^^^^
        File "/tmp/pip-build-env-y7lyori3/overlay/lib/python3.13/site-packages/setuptools/_distutils/cmd.py", line 316, in run_command
          self.distribution.run_command(command)
          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
        File "/tmp/pip-build-env-y7lyori3/overlay/lib/python3.13/site-packages/setuptools/dist.py", line 994, in run_command
          super().run_command(command)
          ~~~~~~~~~~~~~~~~~~~^^^^^^^^^
        File "/tmp/pip-build-env-y7lyori3/overlay/lib/python3.13/site-packages/setuptools/_distutils/dist.py", line 973, in run_command
          cmd_obj.run()
          ~~~~~~~~~~~^^
        File "<string>", line 152, in run
        File "<string>", line 166, in _find_mujoco
      RuntimeError: MUJOCO_PATH environment variable is not set
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for mujoco
Failed to build mujoco
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (mujoco)
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2024-11-15:
I don't think we support 3.13 yet?
@nimrod-gileadi do you remember the status of this?

rhaschke, 2024-11-15:
I ran into the same issue. Using python 3.12 instead of 3.13 resolved the problem for me.
Would be helpful, if you could provide a more meaningful error message if a specific python version is not (yet) supported.

nimrod-gileadi, 2024-11-15:
Yeah, we don't release binary wheels for Python 3.13 yet, and the error you see comes from PyPI trying and failing to build the mujoco bindings from source.

saran-t, 2024-11-20:
We'll be uploading cp313 wheels starting from our next release. Keeping this issue open until then.

rhaschke, 2024-11-20:
This is good news. Can you also think about providing a better error message in case a binary release build is missing for a _future_ python version? Maybe, simply print a useful error message if all environment variables are unset.

saran-t, 2024-11-20:
It's pip's failure to build a wheel, I don't think there's anything we can do about that.

saran-t, 2024-11-20:
That error doesn't have anything to do with us not providing wheels for a specific Python version though. It is telling you exactly what to do to get things building from source: if you so wish you could set MUJOCO_PATH appropriately and pip should start building things for you.

saran-t, 2024-11-20:
Alternatively we could just not upload the sdist, then pip would instead complain about not finding an appropriate package rather than trying to build from source. Not 100% sure how informative that error is though (i.e. I don't think it would specifically say "your Python version is the problem" but rather just a generic "can't find the right file to download" message).

rhaschke, 2024-11-21:
What about changing the error to:
`f'{MUJOCO_PATH} environment variable is not set. When installing via pypi, this means that a binary release of mujoco is not available for your python version yet.')`

saran-t, 2024-12-03:
Python 3.13 wheels are now available for MuJoCo version 3.2.6. https://pypi.org/project/mujoco/3.2.6/#files

closed 2024-12-03 by saran-t

rhaschke, 2024-12-04:
@saran-t, what do you think of my proposal https://github.com/google-deepmind/mujoco/issues/2232#issuecomment-2490217854 to augment the error message? The problem in general persists for the next release.

traversaro, 2024-12-04:
> I ran into the same issue. Using python 3.12 instead of 3.13 resolved the problem for me. Would be helpful, if you could provide a more meaningful error message if a specific python version is not (yet) supported.

Quite interesting recap on the issue (that applies to many Python project): https://pypackaging-native.github.io/key-issues/unexpected_fromsource_builds/ .

saran-t, 2024-12-04:
I'm not a fan of adding PyPI-specific messaging to our errors since over time the build failure can migrate to a different place. The only workable solutions are to either delete all existing sdists and stop uploading new sdists, or make the sdist properly buildable from source.

The point in the article that @traversaro referred to around people getting broken sdist builds due to incorrect dependencies don't really affect MuJoCo all that much since we have few dependencies and can afford to set things up to build "hermetically". However, on machines for which a wheel isn't available, `pip install mujoco` will wind up taking a very long time to complete, without a great deal of user feedback, and therefore appearing as if the installation hangs.

@traversaro what would be your preferred solution?

traversaro, 2024-12-04:
> The point in the article that @traversaro referred to around people getting broken sdist builds due to incorrect dependencies don't really affect MuJoCo all that much since we have few dependencies and can afford to set things up to build "hermetically".

Not sure if this is true at the moment, as the MuJoCo sdist expects `MUJOCO_PATH` to be set on the user machine, and it fails otherwise. Even if we modified the build system to include all the C/C++ dependencies hermetically, we would still need the users to have a compiler installed in their system, so the sdist installation may still fail.

> @traversaro what would be your preferred solution?

I think that adding an error message on the possible source of the error may help users that accidentally build from source, without disturbing at all the users that actually want to build from source. I can prepare something in that direction.

saran-t, 2024-12-04:
But if we bundle the full MuJoCo source code with the sdist, and make `MUJOCO_PATH` optional (i.e. build from source if not specified) then the only remaining sources of build errors are either non-working or non-existent build tools, or compiler errors, right?