google-deepmind/mujoco #2237: NaN Error in Gradient Computation of MJX Simulation with Contacts

thomascbrs, 2024-11-20:
### Intro

Hello!

We’re experimenting with the differentiable simulation features in MJX (working with @Daniellayeghi from the University of Edinburgh). While testing, we encountered an issue where computing gradients results in NaN values.

### My setup

- OS: Ubuntu 22.04
- Python: 3.10.12
- MuJoCo: 3.2.5
- JAX: 0.4.33
- Cuda compilation tools, release 12.6, V12.6.77
Build cuda_12.6.r12.6/compiler.34841621_0

### What's happening? What did you expect?

## Description
We wrote a simple example to test the gradient behavior. We are optimizing to find the optimal impulse for a capsule to collide with a sphere so that the sphere moves to the origin of the scene (mocap target). We provide an initial guess for the impulse to a simple gradient-based optimizer. We define a loss function based on the position of the second object (sphere). We apply a single control force along the x-axis to the first object (capsule) on the first time step. To optimize, when we compute the gradient of the trajectory loss (cumulative running cost) with respect to the impulse by using `jax.jacrev`, we consistently observe NaN values. You should be able to visualize the initial guess resulting trajectory when running the example below.

We have tested with different geometries, including Capsule, Box, and Sphere. We specifically use Capsule-Sphere in our example because the majority of our own project uses this type of collision geometry. 

## Comments
- *dt = 0.01; Initial force guess u = 2*:  
  This guess induces a trajectory that creates a lot of contact, and optimizing over it results in NaN values during the optimization.

- *dt = 0.01; Initial force guess: u = 15*:  
  This guess induces a trajectory with what looks like a single and short contact. The optimizer is able to optimize it without NaN values.

- *dt = 0.001*:  
  Reducing `dt = 0.001` (and proportionally increasing the initial guess) results in immediate NaNs.

- *solimp, solref*:  
  We have tested different contact values (`solimp`, `solref`, `friction`) as well, which still gives NaN values. We noted that these influence and modify the appearance of the NaN values but still result in NaNs.

- *collision type*:  
  For sphere-sphere collision cases, the NaN values are less common but still appear, and it’s unclear at what point they arise.

- *JIT*:  
  Not JIT-ing any of our functions does not remove any errors.

Is there a reason for this behavior? Are we making mistakes in our setup?

### Steps for reproduction

Run the following python code.

### Minimal model for reproduction

See below.

### Code required for reproduction

## Minimal example
<details>
<summary>python code</summary>

```python
import jax
import jax.numpy as jnp
import numpy as np
import mujoco
from mujoco import mjx
import os
from jax import config

# Jax configs
config.update('jax_default_matmul_precision', 'high')
# config.update("jax_debug_nans", True)
# config.update("jax_enable_x64", True)
[... 157 lines cut ...]
    u0, batch = 2., 1 # Initial guess
    u0_jnp = jnp.array([u0])
    qpos = jnp.expand_dims(jnp.array(idata.qpos), axis=0)
    qpos = jnp.repeat(qpos,batch, axis = 0)

    # Visualise guess
    visu_u(u0)

    # Run gradient descent
    optimal_x = gradient_descent(qpos, u0_jnp, learning_rate=0.1)   

    # Check the gradient along the trajectory, debugging
    # print(compute_traj_grad_wrt_u(qpos, jnp.array([2.8]))) # Working
    # print(compute_traj_grad_wrt_u(qpos, jnp.array([2.9]))) # NaNs
```
</details>

###

Thank you !

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

btaba, 2024-12-04:
Hi @thomascbrs , it looks like you already tried higher precision and f64. Without running the repro my initial guess at possible culprits are:

[1] Making sure the sim is stable with 1 solver iteration (although I understand why you chose 1 iteration, and you already tried smaller timesteps). Did you mess with your XML in simulate and the MJX viewer to make sure it looks OK?
[2] Gradients over 100 sim steps may be causing the numerical instability (chain rule). Have you tried modifying the horizon length for your task, or modifying the task so that the horizon is smaller? To keep a longer horizon, you can maybe consider breaking up `u` into segments (10 steps, multiple impulses), and adding stop gradients between segments (although that changes the nature of the task)

Andrew-Luo1, 2024-12-04:
Thank you for the detailed post. Please give this a try: 
```python
import jax
import jax.numpy as jnp
import numpy as np
import mujoco
from mujoco import mjx
import os
from jax import config

# Jax configs
config.update('jax_default_matmul_precision', 'high')
# config.update("jax_debug_nans", True)
config.update("jax_enable_x64", True)

def running_cost(dx):
    return jnp.array([dx.qpos[7]**2 + 0.00001*dx.qfrc_applied[0]**2])

@jax.vmap
def simulate_trajectory_mjx(qpos_init, u):
    """ Simulate the impulse of the force. Return states and costs."""
    def step_scan_mjx(carry, _):
        dx = carry
        dx = mjx.step(mx, dx) # Dynamics function
        t = jnp.expand_dims(dx.time, axis=0)
        cost = running_cost(dx)
        dx = dx.replace(qfrc_applied=dx.qfrc_applied.at[:].set(jnp.zeros_like(dx.qfrc_applied)))
        return (dx), jnp.concatenate([dx.qpos, dx.qvel, cost, t])
    dx = data_init
    dx = dx.replace(qpos=dx.qpos.at[:].set(qpos_init))
    dx = dx.replace(qfrc_applied=dx.qfrc_applied.at[:].set(u))
    (dx), res = jax.lax.scan(step_scan_mjx, (dx), None, length=Nlength)
    res, cost, t = res[...,:-2], res[...,-2], res[...,-1]
    return res, cost, t

def compute_trajectory_costs(qpos_init, u):
    """ Wrapper function to compute the gradient wrt costs."""
    res, cost, t = simulate_trajectory_mjx(qpos_init, u)
    return cost, res, t

def visu_u(u0):
    """ Visualisation for a single force."""
    def visualise(qpos, qvel):
        import time
        from mujoco import viewer
        data = mujoco.MjData(model)
        data.qpos = idata.qpos

        with viewer.launch_passive(model, data) as viewer:
            for i in range(qpos.shape[0]):
                step_start = time.time()
                data.qpos[:] = qpos[i]
                data.qvel[:] = qvel[i]
                mujoco.mj_forward(model, data)
                viewer.sync()
                time_until_next_step = model.opt.timestep - (time.time() - step_start)
                if time_until_next_step > 0:
                    # time.sleep(time_until_next_step)
                    time.sleep(0.075)
    qpos = jnp.expand_dims(jnp.array(idata.qpos), axis=0)
    qpos = jnp.repeat(qpos,1, axis = 0)

    u = set_u(jnp.array([u0]))
    res, _, _ = simulate_trajectory_mjx(qpos, u)
    qpos_mjx, qvel_mjx = res[0,:,:model.nq], res[0,:,model.nq:]
    visualise(qpos_mjx, qvel_mjx)

@jax.jit
def compute_loss_grad(qpos_init, u):
    """ Compute gradient of loss wrt u"""
[... 118 lines cut ...]
# visu_u(u0)

# Run gradient descent
optimal_x = gradient_descent(qpos, u0_jnp, learning_rate=0.1)   

# Check the gradient along the trajectory, debugging
# print(compute_traj_grad_wrt_u(qpos, jnp.array([2.8]))) # Working
# print(compute_traj_grad_wrt_u(qpos, jnp.array([2.9]))) # NaNs
```

What made it work for me was really just switching computation to use float64's. The output I get is:

```
Iteration 0: x = [2.20235773], f(x) = 0.7359345571075301
Iteration 1: x = [2.25593161], f(x) = 0.7171138442868448
Iteration 2: x = [2.31733098], f(x) = 0.731491598833455
Iteration 3: x = [2.3476431], f(x) = 0.8316846146504808
...
Iteration 96: x = [-1.64549796], f(x) = 4.0000270766355115
Iteration 97: x = [-1.64549467], f(x) = 4.000027076527205
Iteration 98: x = [-1.64549138], f(x) = 4.000027076418899
Iteration 99: x = [-1.64548809], f(x) = 4.000027076310594
```

Good luck on your project!

Daniellayeghi, 2024-12-05:
Thank you both, @btaba and @Andrew-Luo1! I work with @thomascbrs on this project. To answer the first comment, we played with the viewer to ensure the simulation was stable for our contact parameters and the simulation timestep. Also, a shorter horizon didn't change the NaN behaviour, so we didn't think it had to do with the chain rule. However, it did look like the NaN gradients were extremely sensitive to small numerical changes, which made us believe it could be a precision issue. As mentioned, we did try to enable `jax.config.update("jax_enable_x64", True)` but it seems like this is not sufficient for enabling higher precision in mjx  due to preset 32 bit precision types used in `mjx.make_data` and causes type errors, for example in this case:

```python
* the input carry component carry.contact.geom1 has type int32[4] but the corresponding output carry component has type int64[4], so the dtypes do not match;
* the input carry component carry.contact.geom2 has type int32[4] but the corresponding output carry component has type int64[4], so the dtypes do not match;
* the input carry component carry.contact.geom has type int32[4,2] but the corresponding output carry component has type int64[4,2], so the dtypes do not match. 
```

However, thanks to @Andrew-Luo1, manually mapping all the nodes to the correct data type circumvents this. This brings up a broader question: could `mjx.make_data` dynamically respect the type set by `jax.config.update("jax_enable_x64", True)`?

We internally added a type check w.r.t the config inside `mjx.make_data` and set data types accordingly, for example:

```python
# Inside the mjx.make_data function ...
if jax.config.jax_enable_x64:
    int_dtype = jp.int64
    float_dtype = jp.float64
else:
    int_dtype = jp.int32
    float_dtype = jp.float32

# Example usage with dynamically set data types
dist = jp.zeros((ncon,), dtype=float_dtype)
geom1 = jp.full((ncon,), -1, dtype=int_dtype)
...
```

Would it make sense to consider integrating a mechanism like this into `mjx.make_data` to streamline precision handling?

btaba, 2024-12-05:
@Daniellayeghi floats get correctly converted to float64 in `mjx.Data`  with `jax_enable_x64`. The `contact.geom` ints however do not. We'll push out a fix!

commit c1d97d4a (2024-12-06): Fix #2237.

PiperOrigin-RevId: 703547316
Change-Id: I279876a299e56a16f8d5227fc1d0f64265477853
closed 2024-12-06 by copybara-service[bot]