google-deepmind/mujoco #2242: Control Inputs Not Affecting Robot Movement in MuJoCo MJX Simulation

wgu93, 2024-11-23:
### Intro

Hi!

I am a researcher using MuJoCo for my research on robotics.

### My setup

MuJoCo: 3.2.5
Python: 3.10
OS and arch: Apple M3 & Ubuntu 20.04 (x86 + Nvidia GPU)
Jax: 0.4.35

### What's happening? What did you expect?

Dear Development Team,

We are currently using MJX for simulating a mobile differential drive robot. However, we have encountered an issue where setting control inputs via `mjx_d.ctrl` (specifically the wheel rotation) does not appear to propagate to the robot’s state, particularly its position (`mjx_d.qpos[:3]`). 

We attempted to assign control inputs in two different ways, as shown in the attached code snippet. In both cases, we assigned a scalar control input of 1 (with control limits set to ±10). The results from the first attempt are shown below:

`[0.   0.   0.05 1.   0.   0.   0.   0.   0.   0.   0.   0.   0.   0.
 0.   0.   0.   0.  ] <class 'numpy.ndarray'>
[0.   0.   0.05 1.   0.   0.   0.   0.   0.   0.   0.   0.   0.   0.
 0.   0.   0.   0.  ] <class 'jaxlib.xla_extension.ArrayImpl'> {CpuDevice(id=0)}
mjx_d.qpos: [ 1.2437687e-11 -4.1769656e-12  5.1890183e-02]
mjx_d.ctrl: [0. 0. 1. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
mjx_d.qpos: [ 4.2460681e-11 -8.4539611e-11  5.5426836e-02]
mjx_d.ctrl: [0. 0. 1. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
mjx_d.qpos: [ 8.4154135e-11 -1.3694780e-10  5.8978669e-02]
mjx_d.ctrl: [0. 0. 1. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
mjx_d.qpos: [ 1.2515745e-10 -1.8497391e-10  6.1971959e-02]
mjx_d.ctrl: [0. 0. 1. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
mjx_d.qpos: [ 1.6715739e-10 -2.3120268e-10  6.4294137e-02]
mjx_d.ctrl: [0. 0. 1. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
mjx_d.qpos: [ 2.0751242e-10 -2.7656868e-10  6.6012442e-02]
mjx_d.ctrl: [0. 0. 1. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
mjx_d.qpos: [ 2.4429786e-10 -3.1017214e-10  6.7246348e-02]
mjx_d.ctrl: [0. 0. 1. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
mjx_d.qpos: [ 2.7387220e-10 -3.1476238e-10  6.8114631e-02]
mjx_d.ctrl: [0. 0. 1. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
mjx_d.qpos: [ 2.9142327e-10 -3.2383191e-10  6.8708785e-02]
mjx_d.ctrl: [0. 0. 1. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
mjx_d.qpos: [ 3.049012e-10 -3.304112e-10  6.910376e-02]
mjx_d.ctrl: [0. 0. 1. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]`

As shown above, the control input is successfully assigned to `mjx_d.ctrl`. However, the robot’s position remains nearly constant, with only minimal oscillations near the origin observed. We repeated the process using the second method of assigning control inputs and observed the same behavior.

Additionally, we increased the control input from 1 to higher values and saw that the robot's position (`mjx_d.qpos[:3]`) changed very slightly, with the magnitude of the change on the order of 10^{-7}, which suggests that the system is still not acting as expected. Moreover, further increasing the control input eventually led to NaN values in the simulation.

Could you please kindly provide insight into why the control inputs are not resulting in state propagation for the robot? Are there any settings, configurations, or additional parameters that we may have overlooked in the simulation code? We would greatly appreciate your comments on this issue. Thank you for your time and assistance!

### Steps for reproduction

1. Define a simple robot model (e.g., a differential drive robot with four wheels) in XML, including joint actuators on each wheel.
2. Replace the placeholder <robot_xml_file> with the actual directory containing the model, and execute the code below.

### Code required for reproduction

import mujoco
import mujoco.viewer
import os
import sys
import numpy as np
import jax
from mujoco import mjx

robot_xml_file = " "

xml_path = os.path.join(os.path.dirname(os.path.dirname(__file__)), robot_xml_file)
m = mujoco.MjModel.from_xml_path(xml_path)
d = mujoco.MjData(m)

mjx_m = mjx.put_model(m)
mjx_d = mjx.put_data(m, d)
print(d.qpos, type(d.qpos))
print(mjx_d.qpos, type(mjx_d.qpos), mjx_d.qpos.devices())

jit_step = jax.jit(mjx.step)
for i in range(10):
    # Attempt 1
    ctrl = mjx_d.ctrl.at[2].set(1)
    mjx_d = mjx_d.replace(ctrl=ctrl)

    # Attempt 2
    # d.ctrl = 1
    # mjx_d = mjx.put_data(m, d)

    mjx_d = jit_step(mjx_m, mjx_d)
    print(f"mjx_d.qpos: {mjx_d.qpos[:3]}")
    print(f"mjx_d.ctrl: {mjx_d.ctrl}")

    # For attempt 2 only
    # d = mjx.get_data(m, mjx_d)

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2024-11-23:
This is most likely a bug in your model, but we cannot help you since you did not provide it. Please don't open bugs unless you know there is a bug, and certainly don't post things like this without full reproducibility. Your "Steps for Reproduction" above is borderline insulting.

closed 2024-11-23 by yuvaltassa

wgu93, 2024-11-24:
Hi @yuvaltassa, thank you for your response! I apologize if the description in the "Steps to Reproduce" section came across as unclear or incomplete. My primary assumption was that the issue stemmed from the Python code rather than the robot model, which is why I did not include the differential drive robot model in my initial report. I will open a new issue with the XML file for a standard differential drive model included to provide a more comprehensive context. Thanks again for your comment!