google-deepmind/mujoco #2245: Mujoco Unity OverflowException when trying to load ALOHA 2 scene

fayezsalka, 2024-11-24:
### Intro

Hi!

### My setup

MuJoCo plugin for Unity

### What's happening? What did you expect?

With Mujoco Unity plug in, most [mujoco_menagerie](https://github.com/google-deepmind/mujoco_menagerie) imports seem to fail. With common error: 

OverflowException
Mujoco.StlMeshParser.ParseBinary (System.Byte[] stlFileContents, UnityEngine.Vector3 scale) (at ./Packages/mujoco/Editor/Importer/StlMeshParser.cs:76)
Mujoco.MjImporterWithAssets.CopyMeshAndRescale (System.String sourceFilePath, System.String targetFilePath, UnityEngine.Vector3 scale) (at ./Packages/mujoco/Editor/Importer/MjImporterWithAssets.cs:182)
Mujoco.MjImporterWithAssets.ParseMesh (System.Xml.XmlElement parentNode) (at ./Packages/mujoco/Editor/Importer/MjImporterWithAssets.cs:161)
Mujoco.MjImporterWithAssets.ParseAssets (System.Xml.XmlElement parentNode) (at ./Packages/mujoco/Editor/Importer/MjImporterWithAssets.cs:126)
Mujoco.MjImporterWithAssets.ParseRoot (UnityEngine.GameObject parentObject, System.Xml.XmlElement parentNode) (at ./Packages/mujoco/Editor/Importer/MjImporterWithAssets.cs:89)
Mujoco.MjcfImporter.ImportXml (System.Xml.XmlDocument mjcfXml, System.String name) (at ./Packages/mujoco/Runtime/Importer/MjcfImporter.cs:100)
UnityEngine.Debug:LogException(Exception)
Mujoco.MjcfImporter:ImportXml(XmlDocument, String) (at ./Packages/mujoco/Runtime/Importer/MjcfImporter.cs:104)
Mujoco.MjImporterWithAssets:ImportString(String, String, String) (at ./Packages/mujoco/Editor/Importer/MjImporterWithAssets.cs:75)
Mujoco.MjImporterWithAssets:ImportFile(String) (at ./Packages/mujoco/Editor/Importer/MjImporterWithAssets.cs:60)
Mujoco.MjImporterEditorWindow:Apply() (at ./Packages/mujoco/Editor/Importer/MjImporterEditorWindow.cs:28)

And robots that manage to load, explodes when enetring play mode in Unity, with all parts scattered and in random orientations in the scene. 

### Steps for reproduction

Import any scene from [mujoco_menagerie](https://github.com/google-deepmind/mujoco_menagerie) into Unity

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

fayezsalka, 2024-11-25:
The MJ Geom regression was intorduced in the very latest release of Mujoco: [3.2.5](https://github.com/google-deepmind/mujoco/releases/tag/3.2.5)

With [3.2.4](https://github.com/google-deepmind/mujoco/releases/tag/3.2.4) MJ Geom works correctly (minus the import issue).

yuvaltassa, 2024-11-25:
I don't think we ever supported ASCII STL files?

The last few messages were a bit confusing, could you please clarify/repreat, what the two issues are, which one is new and which one is old, and how to reproduce them?

Balint-H, 2024-11-25:
Hello! I did some early tests with importing mesh based with 3.2.5 and they seem to work (the [MPL hand ](https://www.roboti.us/forum/index.php?resources/modular-prosthetic-limb-shap-test-suites.19/) is my go-to test scene).

One possible reason the type of issue you are describing (meshes being in the wrong spot and orientation) can arise is if there's a mismatch between package version and mujoco binary version. Could you confirm that those match for you?