google-deepmind/mujoco #2252: mujoco mjx stepping is not reproducible for some models

KamatMayur, 2024-11-26:
### Intro

Hi!

I am a graduate, I use MuJoCo for my research on RL.

### My setup

mujoco version: 3.2.5
python api
64 bit
Ubuntu 24.04.1 LTS
RTX 2060 super, 8GB @ 2010 MHz

### What's happening? What did you expect?

I'm simulating a model for a number of steps multiple times after resetting it. The results are always consistent with the mujoco.mj_step(). But never consistent with mjx.step(). The following code can be used to reproduce the results. I have tested my own xml model but the same is applicable to the humanoid.xml model under the mjx/test_data/humanoid/ directory. The ouput of the mjx step is always different despite of restarting my ipynb notebook kernel. 

**NOTE: This is only happening to models simillar to what have been defined in mjx or brax. Not with the models under the mujoco/model/. However the mujoco step is always consistent no matter what the model is.**

### Steps for reproduction

Run the code given below with the humanoid.xml model that comes in mujoco.xml

### Minimal model for reproduction

Use the humanoid.xml model under the mjx/test_data/humanoid/ directory.

### Code required for reproduction

```python
import time
import mujoco
import mujoco.mjx as mjx
import jax
import os
from jax import lax

xla_flags = os.environ.get('XLA_FLAGS', '')
xla_flags += ' --xla_gpu_triton_gemm_any=True'
os.environ['XLA_FLAGS'] = xla_flags

path = "temp.xml"

jit_step = jax.jit(mjx.step)
model = mujoco.MjModel.from_xml_path(path)
data = mujoco.MjData(model)
model_mjx = mjx.put_model(model)
data_mjx = mjx.put_data(model, data)

_ = jit_step(model_mjx, data_mjx)

def timeit_decorator(func):
    def wrapper(*args, **kwargs):
        start_time = time.perf_counter()
        result = func(*args, **kwargs)
        end_time = time.perf_counter()
        elapsed_time = end_time - start_time
        print(f"Function {func.__name__} executed in {elapsed_time:.6f} seconds.")
        return result
    return wrapper

@timeit_decorator
def simulate_1(path):
    model = mujoco.MjModel.from_xml_path(path)
    data = mujoco.MjData(model)

    for _ in range(2000):
        mujoco.mj_step(model, data)

    return data

@timeit_decorator
def simulate_2(path):
    model = mujoco.MjModel.from_xml_path(path)
    data = mujoco.MjData(model)
    model_mjx = mjx.put_model(model)
    data_mjx = mjx.put_data(model, data)

    for _ in range(2000):
        data_mjx = jit_step(model_mjx, data_mjx)

    return data_mjx

print(simulate_1(path).qpos[:6])
print(simulate_1(path).qpos[:6])

print(simulate_2(path).qpos[:6])
print(simulate_2(path).qpos[:6])
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

KamatMayur, 2024-11-26:
Some additional followup regarding some more observations about this. I used jax.lax.scan() to simulate and surprisingly now the results were consistent. However the scan version is way slower than python  for loop which is again something I am unable to understand. 

```python
import time
import mujoco
import mujoco.mjx as mjx
import jax
import os
from jax import lax

xla_flags = os.environ.get('XLA_FLAGS', '')
xla_flags += ' --xla_gpu_triton_gemm_any=True'
os.environ['XLA_FLAGS'] = xla_flags

path = "temp.xml"

jit_step = jax.jit(mjx.step)
model = mujoco.MjModel.from_xml_path(path)
data = mujoco.MjData(model)
model_mjx = mjx.put_model(model)
data_mjx = mjx.put_data(model, data)

_ = jit_step(model_mjx, data_mjx)

def timeit_decorator(func):
    def wrapper(*args, **kwargs):
        start_time = time.perf_counter()
        result = func(*args, **kwargs)
        end_time = time.perf_counter()
        elapsed_time = end_time - start_time
        print(f"Function {func.__name__} executed in {elapsed_time:.6f} seconds.")
        return result
    return wrapper

@timeit_decorator
def simulate_1(path):
    model = mujoco.MjModel.from_xml_path(path)
    data = mujoco.MjData(model)

    for _ in range(2000):
        mujoco.mj_step(model, data)

    return data

@timeit_decorator
def simulate_2(path):
    model = mujoco.MjModel.from_xml_path(path)
    data = mujoco.MjData(model)
    model_mjx = mjx.put_model(model)
    data_mjx = mjx.put_data(model, data)

    for _ in range(2000):
        data_mjx = jit_step(model_mjx, data_mjx)

    return data_mjx

@timeit_decorator
def simulate_3(path):
    # Load the Mujoco model and data
    model = mujoco.MjModel.from_xml_path(path)
    data = mujoco.MjData(model)
    model_mjx = mjx.put_model(model)
    data_mjx = mjx.put_data(model, data)

    # Define a single step function for lax.scan
    def step_fn(data_mjx, _):
        data_mjx = jit_step(model_mjx, data_mjx)
        return data_mjx, None

    # Use lax.scan to perform 2000 steps
    final_data_mjx, _ = lax.scan(step_fn, data_mjx, None, length=2000)

    return final_data_mjx

print(simulate_1(path).qpos[:6])
print(simulate_1(path).qpos[:6])

print(simulate_2(path).qpos[:6])
print(simulate_2(path).qpos[:6])

print(simulate_3(path).qpos[:6])
print(simulate_3(path).qpos[:6])
```
the ouputs i get are as follows:
```
Function simulate_1 executed in 0.042364 seconds.
[ 9.31566088e-01  1.16730959e-11 -1.22705535e+00  2.71106795e-02
  3.14076541e-12  9.99632438e-01]
Function simulate_1 executed in 0.035706 seconds.
[ 9.31566088e-01  1.16730959e-11 -1.22705535e+00  2.71106795e-02
  3.14076541e-12  9.99632438e-01]
Function simulate_2 executed in 3.613978 seconds.
[ 0.9548208   0.03341392 -1.1994746   0.20769314 -0.01191126  0.9780658 ]
Function simulate_2 executed in 3.705666 seconds.
[ 1.0426785e+00 -3.5116475e-04 -1.1833041e+00  2.4396260e-01
  1.0928431e-04  9.6978462e-01]
Function simulate_3 executed in 7.411681 seconds.
[ 0.9929759   0.05589803 -1.2009562   0.20302454 -0.05311187  0.9774589 ]
Function simulate_3 executed in 7.390856 seconds.
[ 0.9929759   0.05589803 -1.2009562   0.20302454 -0.05311187  0.9774589 ]
```

btaba, 2025-04-07:
Hi @KamatMayur , the scan version should be jitted from the outside of the scan to get good performance! For 2000 steps, it's not unexpected that the MJ and MJX versions would differ (f32 vs f64, CPU vs GPU). In our unit tests, we check much shorter stepping horizons.

closed 2025-04-07 by btaba