google-deepmind/mujoco #2259: [MJX] jax.lax.while_loop in solver.py prevents computation of backward gradients

EGalahad, 2024-11-29:
### The feature, motivation and pitch

## Problem
The solver's `jax.lax.while_loop` implementation prevents gradient computation through the environment step during gradient based trajectory optimization. This occurs in the [solver implementation](https://github.com/google-deepmind/mujoco/blob/300450f8b8441e5f00abc1105089a23f50d7147c/mjx/mujoco/mjx/_src/solver.py#L570) when iterations > 1.

Error encountered with `jax.jit` compiled grad function:
```
ValueError: Reverse-mode differentiation does not work for lax.while_loop or lax.fori_loop with dynamic start/stop values.
```

Current workaround of using `opt.iteration=1` leads to potentially inaccurate simulation and gradients.

## Proposed Solution
Add an option to set a fixed iteration count (e.g., 4) that would be compatible with reverse-mode differentiation using either `lax.scan` or `lax.fori_loop` with static bounds.

erikfrey, 2024-12-03:
I like this suggestion and have labeled it as a good one for someone to take on externally.  If no one does, we'll eventually implement it ourselves.

jaraujo98, 2024-12-30:
@erikfrey are you still looking for a volunteer to tackle this? I'd like to give it a shot.

varshneydevansh, 2025-01-29:
Hi @erikfrey 

I previously did try to contribute to this project. But my PR is still pending(now can complete as I now have some better experience with a large codebase[LibreOffice])

### Problem understanding: 
**jax.lax.while_loop** in the solver prevents gradient computation during backpropagation, we need to replace the dynamic loop with a static one when a fixed iteration count is specified.

### Solution proposed:
All Im thinking of introducing a new boolean option in the Model.opt. based on that, the solver will use a static loop with a fixed number of iterations, enabling gradient computation.

Named something like - static_iterations

### Code modifications:

 **Check static_iterations Flag:** In the solver's solve function, we can use **_jax.lax.fori_loop_** instead of **_jax.lax.while_loop_** when _static_iterations_ is enabled.

**Run Fixed Iterations:** When _static_iterations_ is True, execute the solver loop exactly _m.opt.iterations_ times, bypassing the convergence checks.

Basically replacing **jax.lax.while_loop** with **jax.lax.scan** or with **jax.lax.fori_loop** when the flag is enabled?

I'm thinking of using **jax.lax.fori_loop**  seems more appropriate choice

yuvaltassa, 2025-01-29:
In C MuJoCo there is a trivial way to fix the number of iterations: set `mjModel.opt.tolerance = 0`.

But I'll let @erikfrey comment on the correct way to do this in JAX

varshneydevansh, 2025-01-30:
I tried to understand regarding setting **mjModel.opt.tolerance = 0**  i.e. this effectively disables the convergence check which means the solver will continue iterating until it reaches the maximum number of iterations specified, regardless of whether the solution has converged.

It might lead to slightly less accurate solutions because the solver may continue even after reaching a satisfactory solution.

But, In Jax we are using a while(dynamic) loop which stops when the tolerance criteria met and having a fixed number of iterations could provide a deterministic behavior just like C MuJoCo.

tj279, 2025-05-03:
can you please let me work on this

jeh15, 2025-07-09:
The goal is to make the solver reverse mode autodiff compatible when `m.opt.iterations > 1`. In this specific context, setting `m.opt.tolerance = 0` to achieve this behavior feels a bit strange.
I think the simplest solution would be best here and is what I am currently using as a work around:
```python
if m.opt.iterations == 1:
	ctx = body(ctx)
else:
	ctx = _while_loop_scan(cond, body, ctx, max_iter=m.opt.iterations)
```
With this change, if you want the solver to run and return the gradients for exactly `m.opt.iterations`, you can set `m.opt.tolerance = 0`, matching C MuJoCo's behavior.

erikfrey, 2025-07-10:
@jeh15 Yes this is a JAX constraint.  If you want support for reverse mode autodiff, you need to run a fixed number of solver iterations.  In practice if you want to differentiate and you don't want your JIT time to explode, you will want to keep the number of iterations low.

We have plenty of evidence that for forward-mode, using `jax.lax.while` helps performance significantly, so it's important to keep that as the default behavior.

I think using `_while_loop_scan` when `m.opt.tolerance` == 0 is a decent compromise - would someone like to open a PR?

jeh15, 2025-07-10:
@erikfrey That makes sense, thank you for the clarification!

I think @varshneydevansh has a PR with that implementation.

Looking forward to the fix, thanks, guys!

Srinivasoo7, 2026-09-14:
Hi @btaba @erikfrey 
Kindly check my PR #3583 and help me with your views/suggestions on the same. 

Thanks