google-deepmind/mujoco #2275: use the python version of mujoco3.2.6 under win11, an OSError: [WinError 1114] The dynamic link library (DLL) initialization routine failed.

LJoson, 2024-12-06:
### Intro

Hi!

I am a Robotics Engineer I use MuJoCo for my project.

### My setup

mujoco 3.2.6

### What's happening? What did you expect?

When I use the python version of mujoco3.2.6 under win11, an OSError: [WinError 1114] The dynamic link library (DLL) initialization routine failed. What puzzles me is that it works normally under Linux. In the end, I can only use the 2.3.7 version of mujoco under win.

### Steps for reproduction

Just use pip install mujoco on the win11 platform, and then execute python -m mujoco.viewer to reproduce

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

LJoson, 2024-12-06:
After many attempts, I finally found that only python of mujoco3.2.6 will cause this error on the win11 platform.The only solution is to lower the version. Are there any other solutions?

yuvaltassa, 2024-12-07:
To clarify, are you reporting a regression in Windows support between version <= 3.2.5 and 3.2.6 ?

LanYangya, 2024-12-14:
win10 also appear

LJoson, 2024-12-14:
> To clarify, are you reporting a regression in Windows support between version <= 3.2.5 and 3.2.6 ?

yes

Kakcalu13, 2024-12-27:
Yes, I faced the same issue. I was developing the work on a Mac M2 and Linux using mujoco 3.2.6. However, when it came to Windows 11, I encountered the exact same issue described in the title.  

I was forced to use version 3.2.3 on Windows 11 since 3.2.3 also worked on Mac and Linux

I created a simple pip that allows anyone to use their own XML or my default XML to enable FEAGI to take control. This means a lot of people (hopefully) will try it out, including non-tech-savvy users. So this is likely to be an issue for Windows users only

saran-t, 2025-01-01:
Could you please confirm your Python version?

LJoson, 2025-01-01:
> Could you please confirm your Python version?

3.9.19

saran-t, 2025-01-01:
Thanks. Does anyone have a stack trace?

Kakcalu13, 2025-01-01:
> Could you please confirm your Python version?

w11 in my office has version 3.12, I believe. I will post tomorrow for the accurate version using 3.2.6

> Thanks. Does anyone have a stack trace?

I couldn’t install mujoco on my w8 computer, so I will need to wait until I get back to my office. I only have a Mac and Linux atm

Kakcalu13, 2025-01-02:
Stack trace here:

```
C:\Users\station5\Desktop\godot-games>python -m feagi_connector_mujoco
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\station5\AppData\Local\Programs\Python\Python312\Lib\site-packages\feagi_connector_mujoco\__main__.py", line 24, in <module>
    from feagi_connector_mujoco import controller as feagi_controller_mujoco
  File "C:\Users\station5\AppData\Local\Programs\Python\Python312\Lib\site-packages\feagi_connector_mujoco\controller.py", line 26, in <module>
    import mujoco.viewer
  File "C:\Users\station5\AppData\Local\Programs\Python\Python312\Lib\site-packages\mujoco\__init__.py", line 76, in <module>
    _load_all_bundled_plugins()
  File "C:\Users\station5\AppData\Local\Programs\Python\Python312\Lib\site-packages\mujoco\__init__.py", line 69, in _load_all_bundled_plugins
    PLUGIN_HANDLES.append(ctypes.CDLL(os.path.join(directory, filename)))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\station5\AppData\Local\Programs\Python\Python312\Lib\ctypes\__init__.py", line 379, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [WinError 1114] A dynamic link library (DLL) initialization routine failed

C:\Users\station5\Desktop\godot-games>
```

ignore the feagi_connector_mujoco as it simply import mujoco.viewer.

python version here:
```
Python 3.12.5
```

OS: W11

saran-t, 2025-01-03:
OK, I suspect this is caused by our recent LLVM/Clang version upgrade. Needs further investigation.

saran-t, 2025-01-03:
Our CI runs on Windows Server 2022 (roughly equivelent to Windows 10) and we don't see this issue there. If anyone's able to verify this on Windows 10 as well that'd be appreciated.

saran-t, 2025-01-03:
Could you please try [installing the latest MSVC redistributable](https://aka.ms/vs/17/release/vc_redist.x64.exe) ?

Kakcalu13, 2025-01-03:
Thank you so much for your time again!  

I have a question, and pardon my ignorance, but what’s the difference between local native w11 and Windows Server 2022 (aside from it being based on w10)?  

Also, from my understanding, shouldn’t `pip` already be functional on every platform? Sure, I can install and make it work, but what about users without a tech-savvy background, unlike everyone involved in this issue? Wouldn’t that make it more challenging for them? I'm more worried about other non savvy users.

I’d really love for more people to try and explore mujoco, as it has become a passion of mine!

Kakcalu13, 2025-01-03:
> Could you please try [installing the latest MSVC redistributable](https://aka.ms/vs/17/release/vc_redist.x64.exe) ?

[image]

It doesn't work even after following these steps I did:  
1. Clicked the link you provided to download.  
2. Installed it.  
3. Verified that the dialogue of installation says its completed.  
4. Opened a new cmd.  
5. Reinstalled Mujoco to 3.6.3.  
6. Ran the program and encountered the error shown in the screenshot.

saran-t, 2025-01-03:
OK, I'm now very baffled. I've just tested this on a brand new Windows Server 2025 VM (on GCP) and it's working fine. The only thing that I installed on the VM are Python 3.12 and [MSVC Build Tools](https://aka.ms/vs/17/release/vs_BuildTools.exe). 

In the MSVC Build Tools installer I selected Windows SDK 10.0.26100.1742 and the C++ Build Tools. AFAICT the only possible difference between this and a typical Win 11 machine is the Windows SDK which shouldn't be required at runtime!

Just to rule out the issue, please also try installing [Windows SDK 10.0.26100.1742](https://go.microsoft.com/fwlink/?linkid=2286561) although I don't expect that this will solve your issue.

saran-t, 2025-01-03:
> Thank you so much for your time again!
>
> I have a question, and pardon my ignorance, but what’s the difference between local native w11 and Windows Server 2022 (aside from it being based on w10)? [...]

I'm not a Windows expert, but my understanding is that from the userland perspective Windows Server 2022 should look the same as Windows 10, and Windows Server 2025 should look the same as Windows 11.

The idea is for `pip install mujoco` to work out-of-the-box for as many users as possible. Generally speaking we try to keep it compatible with operating systems under ~5 years old. The fact that it doesn't work for you should be treated as a bug.

LJoson, 2025-01-03:
I am also very confused. I am not a win expert too. According to the query information, I feel that it is because mujoco.dll cannot be correctly recognized on win11.

Kakcalu13, 2025-01-03:
> Just to rule out the issue, please also try installing [Windows SDK 10.0.26100.1742](https://go.microsoft.com/fwlink/?linkid=2286561) although I don't expect that this will solve your issue.

[image]

no luck :(

saran-t, 2025-01-03:
The stack trace shows an error in loading one of the bundled plugin DLLs. As a sanity check, could you comment out line 76 in `__init__.py` and see if the actual library is loading correctly?

Kakcalu13, 2025-01-03:
> The stack trace shows an error in loading one of the bundled plugin DLLs. As a sanity check, could you comment out line 76 in `__init__.py` and see if the actual library is loading correctly?

[image]

no error, simply executed and thats it. Nothing happens. Did I do it correctly?

Kakcalu13, 2025-01-03:
> I am also very confused. I am not a win expert too. According to the query information, I feel that it is because mujoco.dll cannot be correctly recognized on win11.

I think this is the case, but someone else with Windows 10 has that issue too.

saran-t, 2025-01-03:
> no error, simply executed and thats it. Nothing happens. Did I do it correctly?

OK this is a good sign, try making some MuJoCo function calls and see if the library is working. Next step in troubleshooting is to add a `print(filename)` before line 69, that way we'll know which DLL is causing the issue.

LJoson, 2025-01-03:
> > 我也很疑惑，我也不是win高手，根据查询的信息，感觉是因为win11上无法正确识别mujoco.dll导致的。
> 
> 我认为情况确实如此，但其他使用 Windows 10 的人也遇到了同样的问题。

my mistake，It seems that the error occurs when loading the dll of the plugin.

LJoson, 2025-01-03:
I have no idea what's going on，i，comment out line 76 in __init__.py
# _load_all_bundled_plugins()

Something went wrong

log:
`(work) E:\BaiduNetdiskDownload\dev\desktop\learningsource\GlimmerLab\idto>python -m mujoco.viewer
Windows fatal exception: access violation

Thread 0x00021de8 (most recent call first):
  File "D:\Miniconda\envs\work\lib\site-packages\mujoco\viewer.py", line 262 in _physics_loop
  File "D:\Miniconda\envs\work\lib\threading.py", line 917 in run
  File "D:\Miniconda\envs\work\lib\threading.py", line 980 in _bootstrap_inner
  File "D:\Miniconda\envs\work\lib\threading.py", line 937 in _bootstrap

Current thread 0x00023a44 (most recent call first):
  File "D:\Miniconda\envs\work\lib\site-packages\mujoco\viewer.py", line 415 in _launch_internal
  File "D:\Miniconda\envs\work\lib\site-packages\mujoco\viewer.py", line 430 in launch
  File "D:\Miniconda\envs\work\lib\site-packages\mujoco\viewer.py", line 508 in main
  File "D:\Miniconda\envs\work\lib\site-packages\absl\app.py", line 254 in _run_main
  File "D:\Miniconda\envs\work\lib\site-packages\absl\app.py", line 308 in run
  File "D:\Miniconda\envs\work\lib\site-packages\mujoco\viewer.py", line 510 in <module>
  File "D:\Miniconda\envs\work\lib\runpy.py", line 87 in _run_code
  File "D:\Miniconda\envs\work\lib\runpy.py", line 197 in _run_module_as_main
Windows fatal exception: access violation

Current thread 0x00021de8 (most recent call first):
  File "D:\Miniconda\envs\work\lib\site-packages\mujoco\viewer.py", line 264 in _physics_loop
  File "D:\Miniconda\envs\work\lib\threading.py", line 917 in run
  File "D:\Miniconda\envs\work\lib\threading.py", line 980 in _bootstrap_inner
  File "D:\Miniconda\envs\work\lib\threading.py", line 937 in _bootstrap

Thread 0x00023a44 (most recent call first):
  File "D:\Miniconda\envs\work\lib\site-packages\mujoco\viewer.py", line 415 in _launch_internal
  File "D:\Miniconda\envs\work\lib\site-packages\mujoco\viewer.py", line 430 in launch
  File "D:\Miniconda\envs\work\lib\site-packages\mujoco\viewer.py", line 508 in main
  File "D:\Miniconda\envs\work\lib\site-packages\absl\app.py", line 254 in _run_main
  File "D:\Miniconda\envs\work\lib\site-packages\absl\app.py", line 308 in run
  File "D:\Miniconda\envs\work\lib\site-packages\mujoco\viewer.py", line 510 in <module>
  File "D:\Miniconda\envs\work\lib\runpy.py", line 87 in _run_code
  File "D:\Miniconda\envs\work\lib\runpy.py", line 197 in _run_module_as_main`

I'm pretty sure it doesn't happen in versions lower than 3.2.5

LJoson, 2025-01-03:
by the way,this is my mini test python code ,Test load dll
`import ctypes
import os

mujoco_path = r"D:\Miniconda\envs\work\lib\site-packages\mujoco\mujoco.dll"
ctypes.WinDLL(mujoco_path)

PLUGIN_HANDLES = []
PLUGINS_DIR = r"D:\Miniconda\envs\work\lib\site-packages\mujoco\plugin"

def _load_all_bundled_plugins():
    for directory, _, filenames in os.walk(PLUGINS_DIR):
        for filename in filenames:
            dll_path = os.path.join(directory, filename)
            if os.path.splitext(filename)[-1] in [".dll", ".dylib", ".so"]:
                try:
                    print(f"Loading DLL: {dll_path}")
                    handle = ctypes.CDLL(dll_path)
                    PLUGIN_HANDLES.append(handle)
                    print(f"Successfully loaded: {filename}")
                except OSError as e:
                    print(f"Failed to load {filename}: {e}")
            else:
                warnings.warn(f"Ignoring non-library in plugin directory: {dll_path}")

_load_all_bundled_plugins()
`

some wrong log
python.exe e:/BaiduNetdiskDownload/dev/desktop/learningsource/GlimmerLab/idto/example.py
Loading DLL: D:\Miniconda\envs\work\lib\site-packages\mujoco\plugin\actuator.dll
Failed to load actuator.dll: [WinError 1114] 动态链接库(DLL)初始化例程失败。
Loading DLL: D:\Miniconda\envs\work\lib\site-packages\mujoco\plugin\elasticity.dll
Failed to load elasticity.dll: [WinError 1114] 动态链接库(DLL)初始化例程失败。
Loading DLL: D:\Miniconda\envs\work\lib\site-packages\mujoco\plugin\sdf.dll
Failed to load sdf.dll: [WinError 1114] 动态链接库(DLL)初始化例程失败。
Loading DLL: D:\Miniconda\envs\work\lib\site-packages\mujoco\plugin\sensor.dll
Failed to load sensor.dll: [WinError 1114] 动态链接库(DLL)初始化例程失败。
,

In fact, what I find very strange is that it only appears in version 3.2.6. In order to figure out the reason, I even checked all PR changes compared to 3.2.5, but still found nothing (sorry, I am a technical person rookie

saran-t, 2025-01-03:
It's likely caused by a change in our CI configuration (which we refresh roughly annually) that we use to build our binaries.

Kakcalu13, 2025-01-03:
> some wrong log python.exe e:/BaiduNetdiskDownload/dev/desktop/learningsource/GlimmerLab/idto/example.py Loading DLL: D:\Miniconda\envs\work\lib\site-packages\mujoco\plugin\actuator.dll Failed to [...]

This is exactly same as my output on my end!

Fifirex, 2025-01-20:
> This still exists with 3.2.7 Are there any fixes that people found other than downgrading?

I have no idea how it did, but updating my graphic driver fixes the issue

robertsulej, 2025-02-28:
Win11, python 3.10.6, NVIDIA driver 572.60 - still the same problem in mujoco 3.3.0, 3.2.6, caused by `mujoco\plugin\actuator.dll`.
Downgrading to mujoco 3.2.5 works.

robertsulej, 2025-03-12:
Here is what I've found:
- actually, for my case (python 3.10.6), mujoco 3.3.0 fails on loading dll only when imported from jupyter notebook; when I import it from python started in terminal, then all works fine;
- so I tried python 3.13.2: works fine started from terminal and from notebook (but I need some packages not available in python 3.13...)
- so I tried python 3.11.9: works fine, terminal and notebook

saran-t, 2025-08-26:
Would someone here be able to help verify if https://github.com/google-deepmind/mujoco/pull/2819 resolves this issue?