google-deepmind/mujoco #2288: Bug in mjx.get_data() for batched data

branvu, 2024-12-13:
### Intro

Hi!

I am a graduate student, I starting to use MuJoCo for my research.

### My setup

mujoco==3.2.6
mujoco-mjx==3.2.6

### What's happening? What did you expect?

I'm trying to use get_data to transfer mjx data to mujoco data for rendering, but I'm getting an error. I think it's related to prior issue #2212 @btaba ?

### Steps for reproduction

1. Run the code below.
2. See error.

### Code required for reproduction

```python 
xml = """
  <mujoco>
    <compiler autolimits="true"/>
    <worldbody>
      <body name="box">
        <joint name="slide1" type="slide" axis="1 0 0" />
        <geom type="box" size=".05 .05 .05" mass="1"/>
      </body>
    </worldbody>
    <actuator>
      <motor joint="slide1"/>
    </actuator>
  </mujoco>
"""
m = mujoco.MjModel.from_xml_string(xml)
d = mujoco.MjData(m)
dx = mjx.put_data(m, d)
mx = mjx.put_model(m, device=jax.devices()[0])
mjx.get_data(m, dx)  # From the added test from #2212 this works!
rng = jax.random.PRNGKey(0)
rng = jax.random.split(rng, 100)
batch_dx = jax.vmap(lambda rng: dx.replace(ctrl=0*jax.random.uniform(rng, (d.ctrl.shape[0]))))(rng)
single_jit_step = jax.jit(jax.vmap(mjx.step, in_axes=(None, 0)))
batch_dx = single_jit_step(mx, batch_dx)
mx = mjx.get_data(mx, batch_dx)  # This does not work
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

thowell, 2024-12-13:
`mjx.get_data` should be called with `mujoco.MjModel`. For reference, please see [mjx.get_data](https://github.com/google-deepmind/mujoco/blob/ec230fa4b622a387cd7e0d2bee491c974b2a3674/mjx/mujoco/mjx/_src/io.py#L403).

closed 2024-12-13 by thowell

hartikainen, 2024-12-21:
I'm having similar issue with `mujoco==3.2.6`, although I'm using `datas = mjx.get_data(m, batch_dx)` instead of `mx = mjx.get_data(mx, batch_dx)` as in the reproduction code above.

(Note that `mx = mjx.get_data(mx, batch_dx)` errors for other reasons, because `mujoco.MjData` cannot be created from `mjx.Model`.)

thowell, 2024-12-27:
There was a bug in `mjx.get_data_into` that was fixed [here](https://github.com/google-deepmind/mujoco/blob/8dd360a2c087558f96b33b6341605783d0e2aa35/mjx/mujoco/mjx/_src/io.py#L478) after release 3.2.6.