google-deepmind/mujoco #2301: MuJoCo XLA (MJX) support in Unity

Norite-dev, 2024-12-19:
### The feature, motivation and pitch

Hi guys,
Is there any chance to have MuJoCo XLA (MJX) support in Unity for GPU-based physics calculation, assuming it will increase the training speed significantly. What do you think? What do you @Balint-H personally think?)

Thanks! Love your product

Balint-H, 2024-12-19:
Hello, this is already partially possible. The process involves:

1. Train with MJX outside Unity. This should incorporate at least moderate domain randomization for robustness.
2. **Confirm the policy works to a satisfactory level in CPU MuJoCo outside Unity.**
3. Converting the policy to an ONNX format.
5. Install the ML-Agents package to perform inference in Unity with the ONNX policy
4. Import the scene in Unity, confirming scene settings are well replicated from the MJX/MuJoCo scene (e.g. timestep, solver settings). Recreate the observation and action space in Unity as ML-Agents observations and action components.
5. Test performance of your policy trained with MJX in Unity!

@joanllobera and I have been experimenting with this process, with no full end-to-end solution demonstrated yet. You can check out our WIP brax-to-ML-Agents converter scripts here: https://github.com/Balint-H/modular-agents/tree/fix/deepmimic-training/mjx-to-unity , we welcome feedback and contributions. You can also find a lot of other MuJoCo in Unity with RL tools in that repo.

Balint-H, 2024-12-19:
Or did you mean to perform the learning and inference directly in a Unity scene?