google-deepmind/mujoco #2337: Memory allocation error using Windows OS

andreh1111, 2025-01-08:
### Intro

Hi!

I am a PhD student at the Polytechnic University of Turin, I use MuJoCo for my research on muscle activity estimation.

I encountered a memory allocation error using Windows that is not present when running the same code with the same PC using WSL. Below is a minimal script to reproduce the error with the arm26.xml model.

CC involved collaborators: @vikashplus @vittorione94

### My setup

Python: 3.11.7
Mujoco: 3.2.6
OS: Windows 10
PC: Dell XPS 13 9310 (16GB RAM)

### What's happening? What did you expect?

When running the code attached below using Windows, I get `mujoco.FatalError: Could not allocate memory` at the i-th instance of MjData, despite the RAM being used by the process is lower than the available (<2GB used, >4GB available). When I run the same code using WSL on the same PC there are no errors if I have RAM available (I also report that the code is about 10 times faster in this case).

### Steps for reproduction

Run the code below using Windows OS.

### Minimal model for reproduction

arm26.xml

<details>
<summary>minimal XML</summary>

```XML

<!-- Copyright 2021 DeepMind Technologies Limited

     Licensed under the Apache License, Version 2.0 (the "License");
     you may not use this file except in compliance with the License.
     You may obtain a copy of the License at

         http://www.apache.org/licenses/LICENSE-2.0

     Unless required by applicable law or agreed to in writing, software
     distributed under the License is distributed on an "AS IS" BASIS,
     WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
     See the License for the specific language governing permissions and
     limitations under the License.
-->

<mujoco model="2-link 6-muscle arm">
  <option timestep="0.005" iterations="50" solver="Newton" tolerance="1e-10"/>

  <visual>
    <rgba haze=".3 .3 .3 1"/>
  </visual>

  <default>
    <joint type="hinge" pos="0 0 0" axis="0 0 1" limited="true" range="0 120" damping="0.1"/>
    <muscle ctrllimited="true" ctrlrange="0 1"/>
  </default>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/>

    <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>

    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
  </asset>

  <worldbody>
    <geom name="floor" pos="0 0 -0.5" size="0 0 1" type="plane" material="matplane"/>

    <light directional="true" diffuse=".8 .8 .8" specular=".2 .2 .2" pos="0 0 5" dir="0 0 -1"/>

    <site name="s0" pos="-0.15 0 0" size="0.02"/>
    <site name="x0" pos="0 -0.15 0" size="0.02" rgba="0 .7 0 1" group="1"/>

    <body pos="0 0 0">
      <geom name="upper arm" type="capsule" size="0.045" fromto="0 0 0  0.5 0 0" rgba=".5 .1 .1 1"/>
      <joint name="shoulder"/>
      <geom name="shoulder" type="cylinder" pos="0 0 0" size=".1 .05" rgba=".5 .1 .8 .5" mass="0" group="1"/>

      <site name="s1" pos="0.15 0.06 0" size="0.02"/>
      <site name="s2" pos="0.15 -0.06 0" size="0.02"/>
      <site name="s3" pos="0.4 0.06 0" size="0.02"/>
      <site name="s4" pos="0.4 -0.06 0" size="0.02"/>
      <site name="s5" pos="0.25 0.1 0" size="0.02"/>
      <site name="s6" pos="0.25 -0.1 0" size="0.02"/>
      <site name="x1" pos="0.5 -0.15 0" size="0.02" rgba="0 .7 0 1" group="1"/>

      <body pos="0.5 0 0">
        <geom name="forearm" type="capsule" size="0.035" fromto="0 0 0  0.5 0 0" rgba=".5 .1 .1 1"/>
        <joint name="elbow"/>
        <geom name="elbow" type="cylinder" pos="0 0 0" size=".08 .05" rgba=".5 .1 .8 .5" mass="0" group="1"/>

        <site name="s7" pos="0.11 0.05 0" size="0.02"/>
        <site name="s8" pos="0.11 -0.05 0" size="0.02"/>
      </body>
    </body>
  </worldbody>

  <tendon>
    <spatial name="SF" width="0.01">
      <site site="s0"/>
      <geom geom="shoulder"/>
      <site site="s1"/>
    </spatial>

    <spatial name="SE" width="0.01">
      <site site="s0"/>
      <geom geom="shoulder" sidesite="x0"/>
      <site site="s2"/>
    </spatial>

    <spatial name="EF" width="0.01">
      <site site="s3"/>
      <geom geom="elbow"/>
      <site site="s7"/>
    </spatial>

    <spatial name="EE" width="0.01">
      <site site="s4"/>
      <geom geom="elbow" sidesite="x1"/>
      <site site="s8"/>
    </spatial>

    <spatial name="BF" width="0.009" rgba=".4 .6 .4 1">
      <site site="s0"/>
      <geom geom="shoulder"/>
      <site site="s5"/>
      <geom geom="elbow"/>
      <site site="s7"/>
    </spatial>

    <spatial name="BE" width="0.009" rgba=".4 .6 .4 1">
      <site site="s0"/>
      <geom geom="shoulder" sidesite="x0"/>
      <site site="s6"/>
      <geom geom="elbow" sidesite="x1"/>
      <site site="s8"/>
    </spatial>
  </tendon>

  <actuator>
    <muscle name="SF" tendon="SF"/>
    <muscle name="SE" tendon="SE"/>
    <muscle name="EF" tendon="EF"/>
    <muscle name="EE" tendon="EE"/>
    <muscle name="BF" tendon="BF"/>
    <muscle name="BE" tendon="BE"/>
  </actuator>
</mujoco>

```
</details>

### Code required for reproduction

```python
import mujoco

N = 10000

mj_model = mujoco.MjModel.from_xml_path('arm26.xml')
mj_datas = [None] * N
for i in range(N):
  mj_datas[i] = mujoco.MjData(mj_model)
  print(i)
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

saran-t, 2025-01-08:
Could you please check if this happens with version 3.2.5 as well? (Both the RAM issue and the performance issue)

andreh1111, 2025-01-09:
Yes, with 3.2.5 I get the same behavior

yuvaltassa, 2025-01-09:
What happens if you reduce the allocation with <size memory="x"/> ?

As a side-note, I hope we will be able to soon improve and reduce the default allocation heuristic.

That said, this does not explain the Windows discrepancy or the performance issue. Note that there is already an open issue regarding Windows 11.

andreh1111, 2025-01-09:
That's a very interesting test, something happens between allocating 4MB and 3MB:
<size memory="4M"/> --> fails when 5.747 GB are used by the process, vs 2.197 GB still available
<size memory="3M"/> --> fails when 6.582 GB are used by the process, vs 0.453 GB still available

The test is consistent among different repetitions (of course with different exact values, but with the same order of magnitude, i.e., with >=4M I always still have a "huge" RAM availability when the code fails, while with <=3M the fail is more reasonable)

yuvaltassa, 2025-01-09:
Can you try 100k or something small like that?

Tip: when you load a model in simulate, the F2 panel tells you how much of the memory you gave it is actually used. You'll also see this at the top of the file if you do print data.