google-deepmind/mujoco #2338: Programmatically change GUI size in mujoco_viewer / mujoco_viewer_passive

pab47, 2025-01-08:
### The feature, motivation and pitch

The GUI that opens up when launching the viewer is set at about 960 x 540 (approx). I can manually resize it once it opens up. But this is getting tedious as I have to keep doing it every single time. 

Is it possible to set this up programmatically?

In the past, I could do with something like this: window = glfw.create_window(800, 600, "Demo", None, None), but it does not look like glfw's handle is exposed in the viewer.py which makes it impossible to use the functions, set_window_size and get_window_size.

### Alternatives

1) resizing the window manually (works but tedious)
2) Tried adding glfw.create_window in viewer.py (does not work)

yuvaltassa, 2025-01-10:
Good suggestion.
Do you want to try doing doing this yourself and sending a PR?

yuvaltassa, 2025-01-10:
steps:
1. build from source
2. find out where the hard-coding occurs
3. expose what you want to expose.

the place to look is not in the py file but in simulate.cc

vikashplus, 2025-02-03:
+1 on this feature. I'm opening multiple windows. Its hard to co-ordinate them without this feature

RealJasonHu, 2026-08-27:
Hi @yuvaltassa, I would like to pick this up if the API direction still makes sense on current main.

I traced the current path: `GlfwAdapter` still creates the window at two thirds of the primary monitor size, `PlatformUIAdapter` only exposes `GetWindowSize()`, and the Python `viewer.launch` / `launch_passive` APIs expose neither an initial size nor a setter.

For a focused first PR, I propose optional keyword-only `width` and `height` arguments on `viewer.launch`, `launch_from_path`, and `launch_passive`, threaded through the Python binding to `GlfwAdapter`. Omitting them would preserve the current default; non-positive dimensions would fail clearly. I would add C++ coverage for explicit/default sizing and Python tests for argument propagation, and leave raw GLFW-handle exposure and window positioning out of scope.

Would you prefer this initial-size API, or a runtime `Handle.set_window_size(width, height)` setter for `launch_passive` instead?

yuvaltassa, 2026-08-27:
This viewer is basically deprecated (though still supported), we are in the process of migrating to the filament + imgui stack.

It's all at head but needs some polish and better docs.

If you try using that and hit problems were much more likely to fix/help.

POC is @okmatija