google-deepmind/mujoco #2342: Unjustifid Error Thrown "XML Error: Error: Node tags must be sequential"

Nick20500, 2025-01-09:
### Intro

Hi!

I am a undergrad student at ETH Zurich, I use MuJoCo for my research on Multimaterial deformable objects.

### My setup

I use MuJoCo 3.2.5 with python bindings.

### What's happening? What did you expect?

When I try to load my "drop_1_axis-y_sim-0.xml" with ```mujoco.MjModel.from_xml_path(scene_path) ```, mujoco throws the error:
" Error: XML Error: Error: Node tags must be sequential
Element 'flexcomp', line 33". 
After inspecting the mesh manually and computationally, the mesh doesn't have discontinouties as mujoco claims. All 546 Nodes of  the "caorsened_bottle_neck.41.msh" file are numbered continous.

Therefore mujoco should not throw any error when loading this scene for simulation.
Here is the full error thrown:

```
Traceback (most recent call last):
  File "/mnt/c/Users/Nickt/OneDrive - ETH Zurich/Desktop/Private stuff/06_Schule/ETH/Bachelorarbeit/Repos/let-the-best-sim-win/src/pipeline/simulate.py", line 304, in simulate_scene_with_timestep
    mj_model = mujoco.MjModel.from_xml_path(scene_path)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: XML Error: Error: Node tags must be sequential
Element 'flexcomp', line 33

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "path/to/my/repo/src/pipeline/simulate.py", line 401, in simulate_scene
    sim_success = simulate_scene_with_timestep(scene_path, output_path,  timestep, sim_duration, fps, save_pixel_frames)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "path/to/my/repo/src/pipeline/simulate.py", line 307, in simulate_scene_with_timestep
    raise Exception(f"InitializationError: the mujoco model and data for {scene_path} couldn't be initialzed.\nError: {e}\n\n")
Exception: InitializationError: the mujoco model and data for bin/multi-mat_dataset/Bottle/1/drop/drop_1_axis-y_sim-0.xml couldn't be initialzed.
Error: XML Error: Error: Node tags must be sequential
Element 'flexcomp', line 33
```

### Steps for reproduction

Here is all the data needed: [example.zip](https://github.com/user-attachments/files/18359393/example.zip)

1. Load the "drop_1_axis-y_sim-0.xml" scene into mujoco. Either programmatically through ```mj_model = mujoco.MjModel.from_xml_path(scene_path)``` or in the interactive viewer.
2. See the error thrown

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

quagla, 2025-01-09:
@mohammad200h Do you know what could be the issue?

mohammad200h, 2025-01-13:
The issue was in the "coarsened_bottle_neck.41.msh"
```
$Nodes
1 546 2 831 <- max node should be equal to 546. There are 546 nodes
3 0 0 546

```

I used [GMSHConverter](https://github.com/mohammad200h/GMSHConverter) to clean the file. Here is the cleaned file:

[example_fixed.zip](https://github.com/user-attachments/files/18396284/example_fixed.zip)

Have a look at coarsened_bottle_neck.41_out_vol.msh 

```
1 546 1 546
3 1 0 546
```

commit 3429a488 (2025-04-10): Detect mismatch between maxNodes and Nodes in a gmsh block. Fixes #2342

PiperOrigin-RevId: 745967745
Change-Id: If1b5fcd71079a0e6a6def9b442758678e31a5828
closed 2025-04-10 by copybara-service[bot]