google-deepmind/mujoco #2345: Add data.joint(k).dof_index and data.joint(k).qpos_index

vikashplus, 2025-01-09:
### The feature, motivation and pitch

Significant quality-of-life improvement if all indices are returned for usecases involving
- subsampling large datasets with joint subsets
- resampling batches with joint subsets 
- building submatrices and subvectors of inertia, pos error, etc, for e.g. stable PD control (cc @Balint-H)
-  Optimizing for a trajectory with a subset of joints, systemID, etc. (cc @Balint-H)

Something similar already happens for 
- data.joint("my_ball_joint").qpos gives 4 numbers.
- data.joint("my_ball_joint").qvel will give 3

### Alternatives

There exist alternatives where one can check the type of joints and calculate the required indices.

quagla, 2025-07-14:
Closing since I believe that `bind` with `mjSpec` now solves this.

vikashplus, 2025-07-14:
@quagla : Im curious to understand why 
- data.joint("my_ball_joint").qpos gives 4 numbers. 
- data.joint("my_ball_joint").qpos_index returns just 1 number