google-deepmind/mujoco #2352: freejoint seemingly stuck at 0 0 0

cameron-crimins-pahl, 2025-01-12:
### Intro

Hello -

I am a researcher at OHSU, I use MuJoCo for my research on animals.  I got a bit stuck made a simple bot in Blender, and hopefully can get unstuck so thanks in advance for your help! I've scoured the documentation about freejoints and ChatGPT and Claude.ai were not able to help resolve this, I think I need another set of eyes because I bet this is a simple fix.

### My setup

mujoco-3.2.6
python 3.12.4
macOS 13.7.1

### What's happening? What did you expect?

I tried to set the freejoint to the center of mass in a simple bot I made in Blender. But I'm unable to set the freejoint to any position except 0 0 0, unless I configure the body/meshes to be at opposite locations. 

For example, this XML block:
[image]

Yields this result in the UI:
[image]

As I understand it, the freejoint should be in the COM of the mesh with this configuration. However, I'm only able to get the freejoint to the COM by changing the XML tags to be in very offset positions, like this:

[image]

which results:

[image]

I have pored over the documentation and asked ChatGPT but I must be missing something. As I understand it, the freejoint should be in the COM of the body it lives under, so setting it to 0 0 0 should ideally make it coincident with the origin of the mesh, which is correctly defined in Blender.  Or potentially I'm missing a very important detail somewhere to make this work strangely. 

### Steps for reproduction

1. Load the model + the .stl file in the .zip.
2. change the pos values as defined in the XML to/ 0 0 0 and/or pos="-.596 -0.15516 2.775" or pos="0.5 0 -2.775".
3. Try to position the freejoint within the COM of the mesh/body.

### Minimal model for reproduction

[HIPS.stl.zip](https://github.com/user-attachments/files/18391074/HIPS.stl.zip)

<mujoco model="t-rex">
  <visual>
    <global offwidth="1920" offheight="1080" />
  </visual>  
  <compiler inertiafromgeom="true" meshdir="/folder/stl" /> 

<option gravity="0 0 -9.81" timestep="0.005"/>

<default>
  <joint limited="true" damping="1" pos="0 0 0" armature="0" />
  <geom condim="1" type="mesh" />
  <motor ctrllimited="true" forcelimited="false" />
  <default class="collision">
  </default>
 
</default>
    
       <asset>
        

        <mesh name="HIPS" file="HIPS.stl"/>
     
  </asset>

  <worldbody>

    <light directional="true" pos="-0.5 0.5 3" dir="0 0 -1" />
    
    <geom name="floor" pos="0 0 0" size="50 50 .1" class="collision" type="plane" rgba="1 0.83 0.61 0.5" />

    
    <body name="HIPS" pos="-.596 -0.15516 2.775">

      <joint name="root" type="free"/>
      <inertial pos="0 0 0" mass="100" />
      <geom name="HIPS" mesh="HIPS" type="mesh" pos="0.5 0 -2.775"/>
    </body>

    </worldbody>

</mujoco>

### Code required for reproduction

This issue is in the UI of MuJoCo, or potentially in the XML I made, no running code of mine seems related to this.

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

Balint-H, 2025-01-13:
The freejoint coincides with the body frame centre. The body frame centre is not necessarily the center of mass however, for example, imagine a body with a single offset primitive geom. The CoM would be in the geom, but the body frame originates elsewhere.

One solution is to offset your mesh (with Blender), so that the mesh vertices are defined from the CoM. Let me know if this answer was incomplete for your issue, or if I misunderstood, and I'll reopen it.

closed 2025-01-13 by Balint-H

cameron-crimins-pahl, 2025-01-13:
Hey thanks for the response!

Just one question:

> so that the mesh vertices are defined from the CoM

Is this different than manually setting the Origin to CoM in Blender?  Or, do you mean to set the mesh to global 0 0 0 ?