google-deepmind/mujoco #2353: `MjSpec` python bindings

milutter, 2025-01-13:
### Intro

Hi!
I am a researcher using MuJoCo for my robot manipulation research.

### My setup

MuJoCo==3.2.6, Python, Ubuntu 22.04

### What's happening? What did you expect?

I started playing around with `MjSpec` and noticed some problems in the API. I commonly use the manual asset dict functionality to create a `MjSpec` or `MjModel`. For example `mj_spec = MjSpec.from_file(filename, assets=assets)` and  `mj_spec.compile(assets)`. 

There seems to be some issue when doing `mj_spec.compile(assets=assets)`, even though it looks like `assets` is a valid keyword. While doing `mj_spec.compile(assets)` seems to work just fine, providing the assets `keyword` the program dies and a lot of binary code is printed to the console.

When the assets are passed as positional argument, passing `None` via `mj_spec.compile(None)` does not work. Historically the `assets` argument has been `dict[str, byte] | None` but this does not seem to be true for `MjSpec`.

When previously working with `mjcf`, I could get the assets `mjcf_root.get_assets()`. `MjSpec` does not seem to provide this option. Would it be possible to add this feature?

Lastly, I am having problems with `mujoco.MjSpec` in a docker container. For example 
```
import mujoco

xml = """
 <mujoco>
   <worldbody>
     <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
     <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
   </worldbody>
</mujoco>
"""

spec = mujoco.MjSpec.from_string(xml)
print(spec.geoms[0].name)
```
yields the error `UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc2 in position 1: invalid continuation byte`. This error seems to be specific to my Docker setup as I do not observe the issues when running outside docker. Do you have any idea what could be causing this difference?

### Steps for reproduction

(see above)

### Minimal model for reproduction

If you encountered the issue in a complex model, please simplify it as much as possible (while still reproducing the issue).

<details>
<summary>minimal XML</summary>

```XML

<mujoco>
  <worldbody>
    <light pos="0 0 1"/>
    <geom type="sphere" size="1" rgba="1 0 0 1"/>
  </worldbody>
</mujoco>

```
</details>

### Code required for reproduction

```python
import mujoco
import mediapy as media
model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)

with mujoco.Renderer(model) as renderer:
  mujoco.mj_forward(model, data)
  renderer.update_scene(data)

  media.show_image(renderer.render())
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

quagla, 2025-01-13:
I have just submitted ba183adb6795f917f8e52252ce9475411805f8d0 which adds the attribute `assets` so you can do

```
specs.assets = asset_dict
model = spec.compile()
```

I have removed any keywords from `compile` so you also don't have the other issues and there is no ambiguity concerning the assets used for compilation vs those that belong to the spec. If you want to retrieve the assets associated to the spec,  you can do `spec.assets`, so for example after attaching a spec you can do

```
parent.assets.update(child.assets)
```
In order to add the child assets to the parent assets. I am not sure about the Docker issue. Maybe you can check if you still have this issue now?

yuvaltassa, 2025-01-16:
@quagla should we add a line about this here?
https://mujoco.readthedocs.io/en/latest/python.html#construction

milutter, 2025-01-21:
Thanks for fixing the interface so fast. I can confirm this issue is resolved in `3.2.7`. 

+1 to @yuvaltassa point of adding it to the documentation. I think it is quite undocumented that one can pass the meshes explicitly which I prefer as this enables storing the meshes and models at different locations. 

Don't worry about the encoding issue. I narrowed it down to how we build the `mujoco` python package in the docker.

yuvaltassa, 2025-01-21:
@milutter do you want to update here and send a PR?
https://github.com/google-deepmind/mujoco/blob/a5ab7a95151617ae9d8ed1a3476202c95b476161/doc/python.rst?plain=1#L528

milutter, 2025-01-21:
Sure I can do that later this week

quagla, 2025-02-13:
Fixed in f61973601ee9c7e7f1f6b8fbc5c7ba24b111195d