google-deepmind/mujoco #2357: Function to calculate CRB inertia/locked inertia

v-r-a, 2025-01-14:
### The feature, motivation and pitch

A function to compute the composite rigid body (CRB) inertia/locked inertia, i.e., instantaneous inertia of an articulated body (or subtree), assuming all the joints are frozen. 

Applications:

1. Testing validity of simplified approaches, e.g., single rigid body (SRBD) assumption
2. Inertia shaping. For example, see the [paper](https://ieeexplore.ieee.org/document/4209816) by Lee & Goswami 2007. Open-access [link](http://www.ambarish.com/paper/Lee_Goswami_intech2009.pdf).

The function might be a good addition to the support functions; many would have written this function already! The C code and a demo video are attached. The CRB inertia is visualised through an equimomental ellipse centred at the COM of the articulated body.

https://github.com/user-attachments/assets/e5e6a9b9-62f1-4823-a493-a2e6a90b6d0c

[crb_inertia_demo.zip](https://github.com/user-attachments/files/18411230/crb_inertia_demo.zip)

yuvaltassa, 2025-01-14:
Can you explain in which ways [mj_crb](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-crb) is not what you want?

v-r-a, 2025-01-14:
Thanks, got it, should have looked in deeper. Couldn't follow what is happening in `mj_crb`. 

Looks like these computations are happening in [mj_comPos](https://github.com/google-deepmind/mujoco/blob/ba183adb6795f917f8e52252ce9475411805f8d0/src/engine/engine_core_smooth.c#L183). First six numbers of `d->cinert `or `d->crb `should be matching with what I want.

closed 2025-01-14 by v-r-a