google-deepmind/mujoco #2361: MJX step gives f

alaurenzi, 2025-01-15:
### Intro

Hi!

I am a software engineer at Istituto Italiano di Tecnologia (IIT), we use MuJoCo for RL.

### My setup

Mujoco version
```
(mjx1) alaurenzi@alaurenzi-iit-desktop:~/code/mjx_playground/playground/mre_bug_noupdate $ pip show mujoco-mjx
WARNING: Skipping /usr/lib/python3/dist-packages/pytest_repeat.egg-info due to invalid metadata entry 'name'
Name: mujoco-mjx
Version: 3.2.7
Summary: MuJoCo XLA (MJX)
[...]
```
OS
```
(mjx1) alaurenzi@alaurenzi-iit-desktop:~/code/mjx_playground/playground/mre_bug_noupdate $ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.1 LTS
Release:	24.04
Codename:	noble
```
Kernel / arch
```
(mjx1) alaurenzi@alaurenzi-iit-desktop:~/code/mjx_playground/playground/mre_bug_noupdate $ uname -a
Linux alaurenzi-iit-desktop 6.8.0-51-generic #52-Ubuntu SMP PREEMPT_DYNAMIC Thu Dec  5 13:09:44 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

```

<details>

<summary>Output of `pip freeze`</summary>
absl-py==2.1.0
alabaster==0.7.12
appdirs==1.4.4
argcomplete==3.1.4
asttokens==2.4.1
attrs==23.2.0
Babel==2.10.3
bcc==0.29.1
beautifulsoup4==4.12.3
blinker==1.7.0
bloom==0.12.0
breezy==3.3.5
Brlapi==0.8.5
Brotli==1.1.0
catkin-pkg==1.0.0
catkin-pkg-modules==1.0.0
certifi==2023.11.17
chardet==5.2.0
click==8.1.6
cloud-init==24.4
colcon-argcomplete==0.3.3
colcon-bash==0.5.0
colcon-cd==0.2.1
colcon-cmake==0.2.29
colcon-common-extensions==0.3.0
colcon-core==0.18.4
colcon-defaults==0.2.9
colcon-devtools==0.3.0
colcon-installed-package-information==0.2.1
colcon-library-path==0.2.1
colcon-metadata==0.2.5
colcon-mixin==0.2.3
colcon-notification==0.3.0
colcon-output==0.2.13
[... 564 lines cut ...]
17  mjx [[ 0.20002352 -0.3071404 ]]
17  mj [ 0.2284091  -0.35195582]
17  err [ 0.02838558 -0.04481542]
---
18  mjx [[ 0.2000262  -0.31366572]]
18  mj [ 0.23007588 -0.3618566 ]
18  err [ 0.03004968 -0.04819089]
---
19  mjx [[ 0.20002903 -0.32054374]]
19  mj [ 0.23174266 -0.37226185]
19  err [ 0.03171363 -0.05171812]
---
```

### Steps for reproduction

Run the `run.py` script in the ZIP folder below. It also contains a `requirements.txt` file for reproducing my pip env.
Then, uncomment line 21 and run again.

[mre_bug_noupdate.zip](https://github.com/user-attachments/files/18423727/mre_bug_noupdate.zip)

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

alaurenzi, 2025-01-17:
From further investigation, it could be that the problem arises from some internal collisions that are not appropriately filtered
This may explain the different behavior between MJ and MJX which IIRC use different collision check pipelines (?)

btaba, 2025-04-07:
Hi @alaurenzi please use the XML [here ](https://github.com/google-deepmind/mujoco_playground/blob/main/mujoco_playground/_src/dm_control_suite/xmls/cartpole.xml)

Namely contacts should be disabled. In MJX we get these contacts:

[image]

In MJ-C we get these contacts:

[image]

Contacts with the railings should be turned off.

Interestingly MJ-C uses a  capsule-box collision function that does not pick the "correct" normal (the contact normals wedge a single capsule inside the box). MJX gives a normal s.t. a single capsule-box collision will be correctly resolved, but due to there being two railings the normals oppose each other and the box is wedged between two railings. @yuvaltassa @kbayes

closed 2025-04-07 by btaba

kbayes, 2025-04-07:
@btaba both nativeccd and libccd agree with MJX so maybe there's a bug in the MJ-C capsule-box collision function.