google-deepmind/mujoco #2362: API function on Python Viewer to inject textures independently from mjModel

vikashplus, 2025-01-15:
### The feature, motivation and pitch

The ability to add a new geom to the user_scn in the Python viewer is quite powerful in building and updating all sort of visualization needs -- from annotation to sensor data viz.  We have been using this functionality extensively. 

For a newly added geom, we also want to add and use a new user texture & material. This feature is currently not supported. This is a missed opportunity as it can unlock a wide variety of cool visualization features on top on simulation which should be lean and thin.

After discussion with @yuvaltassa  and @saran-t, it was proposed that we use "a new API function to inject textures and materials independently from mjModel". 

Starting this request to track the developments.

vikashplus, 2025-03-12:
@yuvaltassa - Im curious about your thoughts on this proposal.

yuvaltassa, 2025-03-15:
Obviously this is much harder since textures and meshes live on the GPU rather than `mjvScene`. 

This is certainly possible, but doing it with the current backend would probably be a wasted effort, since we are hoping to revise out entire visualization/rendering solution. I'd love to give you a timeline for this, but we're still in the early planning stages.

yuvaltassa, 2026-09-18:
This is now possible with the new `mjrf` (Filament) API

closed 2026-09-18 by yuvaltassa