google-deepmind/mujoco #2372: Unity plugin cannot load obj mesh files

awmartin, 2025-01-18:
### Intro

I'm a professor exploring MuJuCo for a course.

### My setup

macOS 15.2.
Unity plugin, MuJuCo 3.2.6, 3.2.7.
Unity 6, build 6000.0.34f1. latest version as of 2024 Jan 17.

### What's happening? What did you expect?

After having installed MuJuCo unity plugin and attempting to load the mujuco_menagerie `scene.xml` file for the unitree_go2 model, I got this error:

> NotImplementedException: OBJ mesh file loading is not yet implemented. Please convert to binary STL.

Since this scene loads as expected in the MuJoCo Simulate app, I expected it to load here in the Unity plugin as well. It seems that there was a previous discussion in 2022 regarding STL files not loading with the Unity plugin, but apparently it was fixed at the time.

Version 3.2.5 of MuJoCo results in an OverflowException:

> OverflowException: Mujoco.StlMeshParser.ParseBinary (System.Byte[] stlFileContents, UnityEngine.Vector3 scale)

...which is odd since the model contains OBJ files.

### Steps for reproduction

1. Assuming Unity 6 and MuJoCo 3.2.7 are installed. Source repo checked out at tag 3.2.7. dylib copied into the unity folder, renamed to `mujoco.dylib` as per installation instructions.
2. Start a new Unity project.
3. Load the Unity package with Window → Package Manager → Plus menu (+) → Install package from disk. Target the `unity` subfolder.
4. Clone or download the `mujoco_menagerie` repo.
5. Open the project. Select Assets → Import MuJoCo Scene. Target the `mujoco_menagerie/unitree_go2/scene.xml` file.
6. Error occurs.

### Minimal model for reproduction

Model found here: https://github.com/google-deepmind/mujoco_menagerie/tree/main/unitree_go2

### Code required for reproduction

No code required.

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

Balint-H, 2025-01-19:
Hello, 

Pre 3.2.7 versions treated and parsed all mesh files as STL files... even if they were obviously not STL. This is of course really bad, and lead to exceptions unsurprisingly.

Ironically, OBJ files have way more native support in Unity's asset manager than STL files. I will look into it now to see how hard it would be to add support for them today.

Balint-H, 2025-01-21:
I sent in the PR. If you have the time, could you check out its branch to see if it works for you?

awmartin, 2025-01-28:
It worked! Many thanks for the fix.

[image]

closed 2025-01-28 by Balint-H

Andorexad, 2025-04-11:
Hello! I also tried out this branch and it can work with my obj files in unity. But it caused new issues with my binary stl file. 
At line 156 of unity/Editor/Importer/MjImporterWithAssets.cs,
`if (Path.GetExtension(sourceFilePath) != ".obj" && Path.GetExtension(sourceFilePath) != ".stl") `

For some reasons Path.GetExtension(sourceFilePath) returns ".STL" for my file, although my filename ends with ".stl". And because of the captial letters, it can't pass this if statement and throw exception.

Same problem with line 197
`if (Path.GetExtension(sourceFilePath) == ".stl") `.

I was able to fix the bug by adding STL case to the above two lines. But I couldn't figure out why my file extension got parsed as ".STL".