google-deepmind/mujoco #2393: GLFW Issue on Ubuntu 24.04

david-dorf, 2025-01-28:
### Intro

Hi!

I am a robotics engineer at Triton Systems and I use MuJoCo for simulation.

### My setup

MuJoCo 3.2.7 on PyPi (Python), AMD64, Ubuntu 24.04

### What's happening? What did you expect?

Running within a Bazel module:

INFO: Running command line: bazel-bin/mujoco_py
/home/david/.cache/bazel/_bazel_david/f695e6f26414630bc7daef67dcbe0426/execroot/_main/bazel-out/k8-fastbuild/bin/mujoco_py.runfiles/.mujoco_py.venv/lib/python3.12/site-packages/glfw/__init__.py:917: GLFWError: (65548) b'Wayland: The platform does not provide the window position'
  warnings.warn(message, GLFWError)

Attempts to import GLFW directly haven't worked to eliminate this from popping up every time the viewer is opened.

### Steps for reproduction

1. Run the code below.
2. Get the warning message.

### Minimal model for reproduction

Shouldn't need this if you import the robot_descriptions package as follows

### Code required for reproduction

```python

import time
import mujoco
import mujoco.viewer
from robot_descriptions.loaders.mujoco import load_robot_description

model = load_robot_description("panda_mj_description")
data = mujoco.MjData(model)

with mujoco.viewer.launch_passive(model, data) as viewer:
    try:
        while viewer.is_running():
            step_start = time.time()
            mujoco.mj_step(model, data)
            with viewer.lock():
                viewer.opt.flags[mujoco.mjtVisFlag.mjVIS_CONTACTPOINT] = int(data.time % 2)
            viewer.sync()
            time_until_next_step = model.opt.timestep - (time.time() - step_start)
            if time_until_next_step > 0:
                time.sleep(time_until_next_step)
    except KeyboardInterrupt:
        pass

```

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

wpumacay, 2025-02-02:
Hi @david-dorf . I think I ran into a similar issue the other day when using wayland. Could you check which version of `glfw` are you using?

```bash
pip freeze | grep glfw
```

If you're using version `1.12.0` then you'll just have to update to the latest version (current is `2.8.0`). I get the same warning, but the visualizer opens normally. I got this issue bc it seems that the package from `dm_control` uses an old version of `glfw` in their requirements.txt [file](https://github.com/google-deepmind/dm_control/blob/a4b9b312c4ae82dded5d2a331d0b5904bf33b869/requirements.txt#L4).

Hope this solves your issue.

david-dorf, 2025-02-05:
It's version 2.8.0, but it's just that the warning is a little annoying - not really a major issue stopping me from doing anything. If this is already expected then I can close this issue.

LaVieEstDure, 2025-02-06:
I have glfw version 2.7.0 and I get a `GLFWError: (65544) b'Wayland: Window position retrieval not supported' warnings.warn(message, GLFWError)` and it actually makes it so that none of the buttons on the MJ visualiser work.

Adding an environment variable to force GLFW to use x11 instead works in fixing both the warning and the issues with the buttons: `export PYGLFW_LIBRARY_VARIANT=x11`. If it's just an annoying warning and you have XWayland, maybe just use that.

davidhozic, 2025-10-28:
Solution until Wayland itself adds support for the missing function: `WAYLAND_DISPLAY= XDG_SESSION_TYPE=x11 python -m mujoco.viewer`

This will force X11 to be used for MuJoCo.

davidhozic, 2025-12-30:
Note that when using Wayland, I also get OpenGL errors during calls to mjr_render (through [`glDebugMessageCallback`](https://registry.khronos.org/OpenGL-Refpages/gl4/html/glDebugMessageCallback.xhtml)):

Severity: HIGH, Message: "GL_INVALID_OPERATION error generated. The required buffer is missing."
Severity: HIGH, Message: "GL_INVALID_OPERATION error generated."

alexkrz, 2026-09-08:
I am not seeing this issue on Ubuntu 24.04 with mujoco 3.12.0 but I am seeing the problem on Ubuntu 26.04.
Also the window does not render correctly on Ubuntu 26.04

closed 2026-09-13 by copybara-service[bot]