google-deepmind/mujoco #2405: Importing URDF works but export to MJCF XML is missing inertial parameters

LaVieEstDure, 2025-02-06:
### Intro

Hello!

I am a PhD student at the Australian Centre for Robotics @ the University of Sydney, I use MuJoCo for my research on trajectory optimisation and machine learning. 

### My setup

MuJoCo version: 3.2.4
API: Python
Architecture: x86
OS: Linux (Fedora 40)

### What's happening? What did you expect?

### Background
I have a URDF for a pre-existing manipulator that I've prepared to work with Mujoco. I added mass parameters to the URDF, and let mujoco infer `inertiafromgeom=true`. It works fine in simulation with the visualiser, and visualisation shows it has reasonable inertia ellipsoids around the links.

I found out that MJX doesn't yet support CYLINDER and BOX collisions and decided to change some collision bodies from cylinder to capsules. Since URDF doesn't support capsules, I tried to export to the mujoco XML format.

### Problem 
After exporting to XML, the XML is missing inertial parameters on all of the rigid bodies. I've tried both using the export button in the default visualiser, and exporting after compiling to MjModel. For context, the URDF is pretty standard with some dummy links that get coalesced into single bodies. 

I can confirm that inertial parameters definitely show up when looking at the MjModel
```
>>> x.body('alpha_m2_1_1_link')
<_MjModelBodyViews
  dofadr: array([0], dtype=int32)
  dofnum: array([1], dtype=int32)
  geomadr: array([2], dtype=int32)
  geomnum: array([2], dtype=int32)
  id: 1
  inertia: array([3.84003153e-05, 3.84003153e-05, 1.65876092e-05])
  invweight0: array([  0.        , 148.05683347])
  ipos: array([0., 0., 0.])
  iquat: array([1., 0., 0., 0.])
  jntadr: array([0], dtype=int32)
  jntnum: array([1], dtype=int32)
  mass: array([0.08293805])
```

### Extract from URDF
There should be mass values for every "super link" (at least one mass between every fixed-connected rigid bodies). There's inertias for some but they are probably not accurate, hence inferring from geometry. 
```xml
...
  <link name="alpha_base_link" />
  <link name="alpha_m3_inline_link">
    <inertial>
      <mass value="0.341" />
      <inertia ixx="0.0013914" ixy="-0.00005970" ixz="0.0003848" iyy="0.00158490" iyz="0.0000614"
        izz="0.006196" />
    </inertial>
    <visual>
      <geometry>
        <mesh filename="meshes/M3-INLINE.stl" />
      </geometry>
      <origin rpy="0 0 0" xyz="0 0 0" />
      <material name="alpha_black" />
    </visual>
    <collision>
      <geometry>
        <cylinder length="0.150" radius="0.020" />
      </geometry>
      <origin rpy="0 1.57075 0" xyz="-0.075 0 0" />
    </collision>
  </link>
  <joint name="alpha_base_link_joint" type="fixed">
    <parent link="alpha_base_link" />
    <child link="alpha_m3_inline_link" />
    <origin rpy="0 0 0" />
  </joint>
...
```

##### Output XML
Note missing mass and inertia values
```xml
...
  <worldbody>
    <geom type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.3 0.3 0.3 1" mesh="M3-INLINE"/>
    <geom size="0.02 0.075" pos="-0.075 0 0" quat="0.707123 0 0.70709 0" type="cylinder" rgba="0.3 0.3 0.3 1"/>
    <body name="alpha_m2_1_1_link" pos="0 0 0.014">
      <joint name="alpha_axis_e" pos="0 0 0" axis="0 0 1" range="0 6.3" actuatorfrcrange="-9 9"/>
      <geom type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.3 0.3 0.3 1" mesh="M2-1-1"/>
      <geom size="0.02 0.033" type="cylinder" rgba="0.3 0.3 0.3 1"/>
      <body name="alpha_m2_joint_link" pos="-0.02 0 0.033">
        <joint name="alpha_axis_d" pos="0 0 0" axis="0 1 0" range="0 3.8" actuatorfrcrange="-9 9"/>
        <geom type="mesh" contype="0" conaffinity="0" group="1" density="0" rgba="0.3 0.3 0.3 1" mesh="M2"/>
        <geom size="0.02 0.0725" pos="-0.02 0 -0.0725" type="cylinder" rgba="0.3 0.3 0.3 1"/>
        <geom size="0.02 0.02" quat="0.707141 0.707073 0 0" type="cylinder" rgba="0.3 0.3 0.3 1"/>
        <geom size="0.02 0.02" pos="-0.04 0 -0.1453" quat="0.707141 0.707073 0 0" type="cylinder" rgba="0.3 0.3 0.3 1"/>
...
```
Both of these are visible at this repo https://github.com/LaVieEstDure/reach-mjx/

### Steps for reproduction

1. Load the model at this repo https://github.com/LaVieEstDure/reach-mjx/ (contains all binary assets too)
2. i) Run mjviz_alpha.py for the visualisation GUI or ii) load it using from_xml() directly 
3. i) Export using the visualisation GUI directly or ii) export using the save XML function directly
4. Notice lack of inertial parameters in exported mjmodel.xml

### Confirmations

- [x] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [x] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2025-02-06:
We need a compiler flag that tells MuJoCo to save explicit inertias... I think the issue is the mixing of explicit mass from URDF and inferred inertia. Can you try to load and save using `mjSpec` and setting the [explicitinertial](https://github.com/google-deepmind/mujoco/blob/dec7b661c14074768259e770a8425bf84c93eb1b/include/mujoco/mjspec.h#L224) flag to 1?

See also https://github.com/google-deepmind/mujoco/issues/2370

LaVieEstDure, 2025-02-06:
For clarity, preferably when I write the inertial values to XML, I would prefer to do it with the Mujoco's inferred inertias rather than the ones already in the URDF.

yuvaltassa, 2025-02-07:
@quagla can you help?

quagla, 2025-02-07:
We have the following condition in the writer 

```
if (body->explicitinertial && model->compiler.inertiafromgeom!=mjINERTIAFROMGEOM_TRUE) {
    XMLElement* inertial = InsertEnd(elem, "inertial");
    WriteAttr(inertial, "pos", 3, body->ipos);
    WriteAttr(inertial, "quat", 4, body->iquat, unitq);
    WriteAttr(inertial, "mass", 1, &body->mass);
    WriteAttr(inertial, "diaginertia", 3, body->inertia);
}
```

Do we want remove the second condition and always write the inertia if `explicitinertial` is true?

yuvaltassa, 2025-02-07:
Great catch!

I think the answer is yes? I'll do it locally and see if anything breaks...

yuvaltassa, 2025-02-07:
Mmm, I now think this is not the right solution. Looking at the relevant [docs](https://mujoco.readthedocs.io/en/latest/XMLreference.html#compiler-inertiafromgeom): "If this setting is “true”, the mass and inertia of each body will be inferred from the geoms attached to it, overriding any values specified with the inertial element.". So if we make this change then saving and loading such a model will break this override property...

@LaVieEstDure can you try

```py
import mujoco
x = mujoco.MjSpec.from_file("alpha_urdf/alpha.urdf")
x.compile()
for body in x.bodies:
    body.explicitinertial = True
x.compiler.inertiafromgeom = 2
mjmodel = x.compile()
print(x.to_xml())
```

LaVieEstDure, 2025-02-09:
Good and not as good: The script outputs mass and inertia to the XML! But it does not save the inferred inertias, only the URDF inertias. At the end, the mjmodel object only has inerties from the URDF. 
Output after directly loading mjModel.from_xml_path:
```
>>> x.body("alpha_m2_joint_link")
<_MjModelBodyViews
  dofadr: array([1], dtype=int32)
  dofnum: array([1], dtype=int32)
  geomadr: array([4], dtype=int32)
  geomnum: array([4], dtype=int32)
  id: 2
  inertia: array([9.28397232e-04, 9.17635297e-04, 7.40126669e-05]) # this is inferred!
```
Inertia after the end of the provided script:
```
>>> mjmodel.body("alpha_m2_joint_link")
<_MjModelBodyViews
  dofadr: array([1], dtype=int32)
  dofnum: array([1], dtype=int32)
  geomadr: array([4], dtype=int32)
  geomnum: array([4], dtype=int32)
  id: 2
  inertia: array([0.00891764, 0.00874922, 0.00079774]) # directly read from the URDF!
```
That's the same values now being stored in the XML
```
      <body name="alpha_m2_joint_link" pos="-0.02 0 0.033">
        <inertial pos="0 0 0" quat="-0.0583938 0.71276 0.00651001 0.698943" mass="0.429" diaginertia="0.00891764 0.00874922 0.000797739"/>
```

Without re-compiling the mjspec, to_xml() doesn't have the inertial tags. Setting compiler.inertialfromgeom=1 again gets rid of the inertial tags too. 

There's also another bug though: saving this XML seems to break the model somehow, and the gripper seems to be inside another link now:

[image]

This is different from the URDF that gets loaded:

[image]

Both directly loaded and visualised in the visualiser. Should I open a second issue for this?

LaVieEstDure, 2025-02-09:
If you save the URDF as MJCF and reload it, both inferring geometries when compiling, and you look at the diff between the parsed MjModels by doing
```
for idx in range(0, 7):
    print(x.body(idx))
```
you see that the models have slightly differing values of inertia and mass for most/all bodies, but also three bodies have very significant changes on their z axis `pos`. 
[image] 

One body goes `pos: array([0.     , 0.0155 , 0.10665])` -> `pos: array([0.    , 0.0155, 0.0069])` and two bodies go from `pos: array([0.     , 0.     , 0.10875])` -> `pos: array([0.   , 0.   , 0.009])`. I can't explain what happened here. Full diff [here](https://www.diffchecker.com/NWXJWi8d/)

yuvaltassa, 2025-02-09:
@LaVieEstDure can you try to reduce the issue to a minimal example? It's not clear to me if this is URDF-specific or not...

Also not clear if there is one or multiple issues.

LaVieEstDure, 2025-02-11:
```py
XML = """
<robot name="test">
  <mujoco>
    <compiler balanceinertia="true" discardvisual="false" inertiafromgeom="true" />
  </mujoco>
  <link name="base_link" />
  <link name="second_link">
    <inertial>
      <mass value="0.341" />
      <inertia ixx="0.0013914" ixy="-0.00005970" ixz="0.0003848" iyy="0.00158490" iyz="0.0000614"
        izz="0.006196" />
    </inertial>
    <collision>
      <geometry>
        <cylinder length="0.150" radius="0.020" />
      </geometry>
      <origin rpy="0 1.57075 0" xyz="-0.075 0 0" />
    </collision>
  </link>
  <joint name="test_joint" type="revolute">
    <parent link="base_link" />
    <child link="second_link" />
    <origin rpy="0 0 0" />
  </joint>
</robot>
"""

import mujoco
x = mujoco.MjSpec.from_string(XML)

# This saves an XML WITHOUT any inertia tags
mjmodel = x.compile() 
print("DEFAULT SAVED MJCF:")
print(x.to_xml(), "\n")

# These are the inferred inertias, different from the ones in the URDF
print("INFERRED INERTIAS:")
print(mjmodel.body(1).inertia, "\n")

# This produces XMLs that have inertias but same as URDF
# Same output as inferfromgeom="false" in compiler tags
for body in x.bodies:
    body.explicitinertial = True
x.compiler.inertiafromgeom = 2
mjmodel = x.compile() 
print("SAVED MJCF AFTER explicitinertial=True:")
print(x.to_xml())
```

This is the minimal URDF + script that shows the inertia saving bug. Frankly, I do not know what caused the wrong kinematic tree output for my current full URDF and do not have time to recreate it

yuvaltassa, 2025-02-13:
The issue is that they want explicit inertias saved for all bodies and can't figure out how to do it. I thought 
```py
for body in x.bodies:
    body.explicitinertial = True
x.compiler.inertiafromgeom = 2
```
would do it, given the lines you posted above, but it doesn't. I think it gets overwritten. 

@LaVieEstDure did you try commenting out the third line before the end in your MWE? (the last recompile)

aftersomemath, 2025-02-27:
The inertia inferred from geoms is contained in the compiled MjModel. So, a workaround is to compile the spec to an MjModel, overwrite the inertia in the spec with that in the MjModel, and then save the spec to xml. @yuvaltassa 's suggestion of a compiler flag to modify the behavior of "to_xml" would basically do the same thing.

Here is a version of the example above with this workaround:

```python
XML = """
<robot name="test">
  <mujoco>
    <compiler balanceinertia="true" discardvisual="false" inertiafromgeom="true" />
  </mujoco>
  <link name="base_link" />
  <link name="second_link">
    <inertial>
      <mass value="0.341" />
      <inertia ixx="0.0013914" ixy="-0.00005970" ixz="0.0003848" iyy="0.00158490" iyz="0.0000614"
        izz="0.006196" />
    </inertial>
    <collision>
      <geometry>
        <cylinder length="0.150" radius="0.020" />
      </geometry>
      <origin rpy="0 1.57075 0" xyz="-0.075 0 0" />
    </collision>
  </link>
  <joint name="test_joint" type="revolute">
    <parent link="base_link" />
    <child link="second_link" />
    <origin rpy="0 0 0" />
  </joint>
</robot>
"""

import mujoco
import numpy as np

# This XML forces inertia to be inferred from geoms using the inertiafromgeom="true"
x = mujoco.MjSpec.from_string(XML)
mjmodel = x.compile() # mjmodel contains the inferred inertia

# Modify spec to allow setting explicit inertia's
x.compiler.inertiafromgeom = 2 # inertiafromgeom="auto"

# Overwrite the spec's inertia parameters with the inertia inferred inertia in the MjModel
x.bodies[1].explicitinertial = True
x.bodies[1].fullinertia = np.full((6, 1), np.nan) # Discard explicit inertia from original XML
x.bodies[1].mass = mjmodel.body(1).mass[0]
x.bodies[1].inertia = mjmodel.body(1).inertia
x.bodies[1].ipos = mjmodel.body(1).ipos
x.bodies[1].iquat = mjmodel.body(1).iquat

# Print the XML which now contains inertia tags with inertia inferred from the geoms
mjmodel = x.compile()
print(x.to_xml())
```

Which produces:
```xml
<mujoco model="test">
  <compiler angle="radian"/>

  <worldbody>
    <body name="second_link">
      <inertial pos="-0.075 0 0" quat="0.707123 0 0.70709 0" mass="0.188496" diaginertia="0.000372279 0.000372279 3.76991e-05"/>
      <joint name="test_joint" pos="0 0 0" axis="1 0 0"/>
      <geom size="0.02 0.075" pos="-0.075 0 0" quat="0.707123 0 0.70709 0" type="cylinder"/>
    </body>
  </worldbody>
</mujoco>
```

LaVieEstDure, 2025-03-02:
@aftersomemath That makes perfect sense and achieves my goal. The compilation process makes a disconnect in the behaviour between the MjSpec and the MjModel. 

Since `.to_xml()` is called on the `MjModel` rather than on the `MjSpec`, does it semantically make more sense for the behaviour to be to save a cloned and then updated MjSpec to XML? It does mean it'll need to invert the compilation or keep track of the changes in the object, which is unideal. Alternatively, `to_xml()` shouldn't exist on the MjModel and there should be a function to get the original MjSpec and then users can choose to save that. 

On a different note, in your example, the second_link mass seems to have gone from 0.341kg to 0.188496, which is approaching half. What happened there? I don't expect link mass changing beyond floating point error.

yuvaltassa, 2025-03-02:
> Since .to_xml() is called on the MjModel rather than on the MjSpec

What do you mean? `to_xml()` is a method of `MjSpec`. This is what happens when you name you variables `x`... 😝

aftersomemath, 2025-03-02:
The mass changes because geoms mass/density were not set, so a default density is being used to infer both mass and inertia. So, to get the desired results, we just need to set the geoms mass to the original bodies mass.

To do this, insert the following lines just after creating the spec object with `from_string`: 
```
# Copy the mass from the body in the XML to the geom
x.compile()
x.geoms[0].density = np.nan # Clear default density
x.geoms[0].mass = x.bodies[1].mass # Copy original mass, the density will be recalculated accordingly

mjmodel = x.compile() # mjmodel contains the inferred inertia with the desired mass
```

Note that, the `x.bodies[1].mass` field does not seem to be populated by `MjSpec.from_string`, so I had to insert an additional `MjSpec.compile()` call at the beginning.

LaVieEstDure, 2025-03-02:
> What do you mean? `to_xml()` is a method of `MjSpec`. This is what happens when you name you variables `x`... 😝

Touché, how quickly you lose your good practices when you start doing research :P I was misremembering because MjData has a `MjModel.from_xml_path`, but yeah there's no save XML, only saveLastXML from MjModel so that's not semantically wrong. There's probably still a missing space for saving an MjSpec that covers all the transformations performed by the compiler but that's a larger decision.

> The mass changes because geoms mass/density were not set, so a default density is being used to infer both mass and inertia

Oh I see, inertiafromgeom infers both moment of inertia and mass. Makes sense. Thank you! I think this just about covers everything I was having issues with. I still dont know how the URDF -> XML conversion broke the link position values for my robot, but I also dont have the time to recreate and debug that. I'll open it if it happens again. Thank you, I'll close the issue now!

commit f96f3e1c (2025-04-03): Add XML `compiler/saveinertial` flag. Fixes #2405

PiperOrigin-RevId: 743607654
Change-Id: I1f2e61be89b79c1995e079f36aa285096f9fb441
closed 2025-03-02 by LaVieEstDure