google-deepmind/mujoco #2414: MjSpec find support

vittorione94, 2025-02-09:
### The feature, motivation and pitch

Hello, 

In MjSpec so far there is only the 'find_all' option to recursively find a list of all elements in the model: [mujoco docs](https://github.com/google-deepmind/mujoco/blob/694eb6bb1d066896ab116ea14379e2cf2590de2e/doc/python.rst#L609)

I remember in mjcf there was the option to 'find('body', <body_name>)', would it be possible to port it to mjSpec?

Thanks for your work!

quagla, 2025-02-10:
Hi, it this https://github.com/google-deepmind/mujoco/blob/main/python/mujoco/specs_test.py#L806 what you are looking for?

vittorione94, 2025-02-10:
Yes, that’s perfect! Thanks @quagla 👍

closed 2025-02-12 by vittorione94